PX4/PX4-Autopilot #5654: Px4fmu-v1 out of flash space on mast3r

jgoppert, 2016-10-12:


jgoppert, 2016-10-12:
Are there any apps we can drop, seems to be 160 bytes over the limit after the latest ecl update.

davids5, 2016-10-12:
FYI px4fmu-v2 also

dagar, 2016-10-12:
GCC 4.9 buys us a small amount. Bottle drop and rover_steering_control can go on px4fmu-v1.

julianoes, 2016-10-12:
@dagar: I think we should go to GCC 5.4.1 because the gdb of 4.9 segfaults trying to read the elf files.
See #5650.

davids5, 2016-10-12:
Just some added data point:

I have not vetted GCC 5.4.1 on all SoC.

If you use gdb from 4.8.4 with 4.9.3 it works.

davids5, 2016-10-12:
@julianoes 

Have you ever found the root cause of the 4.9.3 gdb seg fault? I wonder if we have an invalid debug config flag set?

julianoes, 2016-10-13:
> Have you ever found the root cause of the 4.9.3 gdb seg fault? I wonder if we have an invalid debug config flag set?

I have no idea, sorry.

julianoes, 2016-10-22:
This is resolved I'd say.

closed 2016-10-22 by dagar