PX4/PX4-Autopilot #5657: PX4 1.5RC1 and master - no RTK possible

lvale, 2016-10-12:
Hi. We were following the wiki page for the M8P RTK’s http://px4.io/docs/rtk-gps/ and we don’t even get a simple GPS signal (no sats). We’re using the Drotek Tiny RTK kit https://drotek.com/en/tiny-rtk-documentation/ which works perfectly with ArduPilot on the same conditions (location). We’ve tested PX4 master and PX4 1.5RC1 but the results are the same. QGC can see the USB connected (Base) and we see that it gets a Fix for the GCS. Any hints/tips? I suspect that it might be some parameter on the GPS itself that has to be specifically set on PX4 that either Ardupilot doesn’t require or auto configures.
From the demo on the wiki, are those GPS config files available ?

LorenzMeier, 2016-10-12:
@lvale Are you using Wifi? We are not trying to support 900 MHz modems with RTK at this point. @bkueng Can you comment on that?

lvale, 2016-10-12:
@LorenzMeier Not using WiFi, because we already have other radio links going and WiFi is being used for ROS. Even getting another specific link to RTK is unfeasible.
But the problem is that the Rover unit doesn't even get a normal lock (either 2D or 3D) that's why I'm asking for a config file.

bkueng, 2016-10-13:
@lvale there's no additional config necessary. I don't have the Tiny RTK, so I'm sorry I can't help you with that. But drotek might be able to help. @jperin, any hints?

klopezal, 2016-10-13:
@lvale do you have the soldering bridge activated on rover side? If not, there is no connection between autopilot's TX and GPS RX, so it cannot be properly configured at start-up.

Note : even if M8P is properly configured to output NAV-PVT and other messages at 5 Hz it will not work because the driver waits for an acknowledge message from the GPS.

lvale, 2016-10-13:
@klopezal Thanks. I suppose the bridge activated would also be a problem with ArduPilot, but I'll recheck again.

lvale, 2016-10-19:
Managed to sort by rechecking the closed bridge on the rover side. Going out to test fly.

closed 2016-10-19 by lvale

LorenzMeier, 2016-10-19:
Hey Kevin - the solder bridge is horrible from an usability point of way - is there an alternate better method?

klopezal, 2016-10-19:
You're right, we added it because we needed something to switch between the configuration with RTCM messages coming directly from flight controller's serial port or the configuration with a separate dedicated telemetry link (for Network RTK applications over LTE without companion computer for instance).

The problem is that we cannot directly connect both FC and second datalink TX lines together otherwise GPS initial configuration will be polluted by second datalink's RTCM messages and will not work. On the other side, if you don't connect GPS RX to FC, even if it is manually correctly configured (NAV-PVT and MON-HW messages at 5 Hz), it does not work because FC waits for an acknowledge message during configuration, so it is also horrible from a usability point of view to have to connect/disconnect a wire.

Maybe a solution would be to allow a free TELEM port to forward messages to GPS port, what do you think? That way we could completely remove the solder bridge.

LorenzMeier, 2016-10-19:
I think a physical switch with clear marking and leds to indicate which position is active would do it. I haven't looked at the module, but I also would not want the BT module to be active / powered if its not needed.

klopezal, 2016-10-19:
Ok we'll take this into consideration then!

mzahana, 2016-11-21:
I am facing the same problem. Where is this bridge that needs to be soldered to make the rover module work with Pixhawk?
Thanks.

klopezal, 2016-11-21:
There it is : 

[image]

mzahana, 2016-11-21:
Thanks. In fact, I am using WiFi to connect to Pixhawk. The pixhawk is connected to Odroid which runs mavros that connects to QGC. So , I am not using a dedicated telemetry between the rover and the base. I just connected the rover's serial port to GPS port on Pixhawk. Connected the base via usb to QGC. Is that enough, or there is something missing. For that setup, do I still need to solder the bridge?
Thanks.

klopezal, 2016-11-21:
Your setup looks right,  you need to solder the bridge in order to have the autoconfiguration + RTCM packets going directly to the GPS.

mzahana, 2016-11-21:
Do I need to do any more configurations to the rover module using U-Center?

I played a lot with it and changed configuration as explained on the website:
https://drotek.com/en/tiny-rtk-documentation/

How to load the default configs?

Thanks.

mzahana, 2016-11-21:
I guess Pixhawk is still not able to talk to Pixhawk, even after the soldered bridge. I logged into the pixhawk shell, and ran the gps commands. Here is what I get,

```sh
nsh> gps status
WARN  [gps] GPS 1:
WARN  [gps] protocol: UBX
WARN  [gps] port: /dev/ttyS3, baudrate: 9600, status: NOT OK
WARN  [gps] sat info: disabled, noise: 0, jamming detected: NO
nsh> gps stop
nsh> gps start -d /dev/ttyS3 -m ubx -b 38400
nsh> gps status
WARN  [gps] GPS 1:
WARN  [gps] protocol: ASHTECH
WARN  [gps] port: /dev/ttyS3, baudrate: 115200, status: NOT OK
WARN  [gps] sat info: disabled, noise: 0, jamming detected: NO
```

klopezal, 2016-11-21:
In order to load default config, on U-Center go under View -> Configuration View -> CFG and select Revert to default configuration : 

[image]

and click on Send. 

Have you checked that TX and RX are correctly connected and not inverted?

mzahana, 2016-11-21:
I think the issue is solved. I just had to wait for it to communicate. Now I get,

```sh
gps status
WARN  [gps] GPS 1:
WARN  [gps] protocol: UBX
WARN  [gps] port: /dev/ttyS3, baudrate: 38400, status: OK
WARN  [gps] sat info: disabled, noise: 96, jamming detected: NO
WARN  [gps] position lock: 0, satellites: 0, last update:  39.3450ms ago
WARN  [gps] lat: 0, lon: 0, alt: -17000
WARN  [gps] vel: 0.00m/s, 0.00m/s, 0.00m/s
WARN  [gps] hdop: 99.99, vdop: 99.99
WARN  [gps] eph: 4294967.50m, epv: 3750091.75m
WARN  [gps] rate position:                5.00 Hz
WARN  [gps] rate velocity:                5.00 Hz
WARN  [gps] rate publication:             5.00 Hz
WARN  [gps] rate RTCM injection:          0.00 Hz
nsh> 

```

Thanks.

klopezal, 2016-11-21:
Perfect!

mzahana, 2016-11-22:
Now, If I'd like to add a dedicated telemetry link e.g.
1-  one telemetry radio connected to the telemetry port of rover module,
2- one telemetry radio to the telemetry port on the base module.
Do I keep the soldered bridge, or I have to unsolder it?

Thanks.

klopezal, 2016-11-24:
If you let it soldered, that means that autoconfiguration and RTCM messages will share the same port, so it can be problematic if differential data is present during FC startup.

If you unsolder it, there will be no connection between FC GPS TX port and GPS RX, so there will be no autoconfiguration nor ACK messages.

mzahana, 2016-11-24:
I think there should be a better way to deal with this. The acknowledge is needed, for px4 to pick it up, but also I need to have dedicated link. The way it works for me  now is keep it soldered, unplug the telemetry, power on, get the lock, plug telenetry. That way, it works fine.

klopezal, 2016-11-24:
You are right, but it is quite inconvenient from the user perspective. Maybe a solution would be to use a free Telem port in order to connect the other separate datalink and send its data to GPS only once autoconfiguration is ok.

mzahana, 2016-11-24:
I'd love to try that. Soldering and unsoldering is not good for usability.