PX4/PX4-Autopilot #5658: LPE (AGL as Z): Massive althold 'jump'.

ecmnet, 2016-10-13:
@jgoppert Using LPE_PUB_AGL_Z enabled, I had a serious altitude jump indoors:

[image]

In this case LPOS.Z had been about 3.5m at arming - I assume, this came from baro. In some cases LPOS.Z with LPE_PUB_AGL_Z is set to ground distance when LIDAR initializes. Here, this was not the case. In the messages i noticed 

2000_01_01_05_51_05: [lpe] lidar fault,  beta  9.21
2000_01_01_05_51_06: [lpe] lidar fault,  beta 31.74

Any position estimation (GPS,Flow,Vision) was disabled. LIDAR seems working fine all the time. I also noticed, that in the beginning Althold was not working very well.

My assumption is, that with AGL_Z enabled, LPOS.Z was set to ground distance at that point in time. This let the position controller assume that the vehicle was far to low.

The log: http://logs.uaventure.com/view/iNiUzgu5dZpiBQNGc5fyVd

jgoppert, 2016-10-13:
When it is 3.5 m at arming that is a problem. We should add back the altitude reset at arming based on a param. What happened is that initially lidar was faulted and ignored,  but eventually the covariance grew enough to treat the lidar agl as true.

LorenzMeier, 2016-10-13:
You shouldn't reset the altitude at arming, that's just wrong. The system needs to track the AGL consistently. By resetting all to zero you are fixing assumptions of values being at zero which are. It true.

For any real deployment resetting the physical position of the control system makes it absolutely impossible to do any sort of autonomous navigation. It messes with controllers, waypoint navigation, path planning and obstacle avoidance.

jgoppert, 2016-10-13:
Ok, then when landed we need to do a fake measurement to keep baro drift out of agl.

jgoppert, 2016-10-13:
While landed a fake lidar-like distance measurement of zero should do it.

ecmnet, 2016-10-13:
@jgoppert @LorenzMeier We had that discussion several times already and basically I agree with Lorenz. Nevertheless there are corner cases like this ( e.g. baro drift during a certain time on ground) that need to be covered to avoid such effects. Baro is simply too inaccurate to rely on indoors.

PR #5662 (2016-10-14): Added landed agl correction for lpe.
This simulates a lidar when you are landed, or disarmed and not in freefall and you don't have a sonar/ mocap/ vision/ or lidar measurement initialized. This should prevent agl drift from baro while on ground. I've tried it in simulation, but need to test in real life. @ecmnet Hopefully this fixes #5658 for you.
closed 2016-10-14 by jgoppert