PX4/PX4-Autopilot #5659: SITL/ROS - No position lock.

TobiMiller, 2016-10-13:
Hey guys,

i want to use the firmware as SITL with ROS. Gazebo is running, the Quad is spawning. But i cannot takeoff. arming is working and rotors are spinning.
`WARN  [commander] rejecting takeoff, no position lock yet. Please retry..`

This is happening with:
`make posix gazebo_iris_opt_flow`
`roslaunch px4 mavros_posix_sitl.launch` -> vehicle: "iris" and "iris_opt_flow"

hope anybody can explain this behaviour to me

jgoppert, 2016-10-13:
The camera is out of focus on the ground so you can't get a position lock and takeoff. Please takeoff in manual first then switch to auto.

jgoppert, 2016-10-14:
With iris it should be working since that uses gps and that is constantly tested in CI.

TobiMiller, 2016-10-17:
Ok. Thank you. i will try to start it manually with a gamecontroller and mavros_extras package. 
The iris is starting without gps. 
https://github.com/PX4/Firmware/blob/master/posix-configs/SITL/init/lpe/iris_opt_flow#L45

jgoppert, 2016-12-23:
Auto takeoff now supported with flow.

closed 2016-12-23 by jgoppert

alirezaka, 2017-10-28:
Hi TobiMiller
have yo solved your problem?
I am encountering your problem too.
if you have solved it, please explain how.
you can.
thanks