PX4/PX4-Autopilot #5660: Hard fault during flight

vvranjek, 2016-10-13:
Hi,

So I had the second crash now because of a hard fault. I am running a modified version of v1.5.0 so this error might be my mess or not, but since I am unable to fully debug the issue and it is serious I am filing a report.  This time I captured the hard fault dump but I am still unable to pinpoint the error. I followed the instructions on gdb debugging and it points me to 2 lines.

`Assertion failed at file:armv7-m/up_hardfault.c line: 184 task: navigator
sp:     20005fe8
IRQ stack:
  base: 2000602c
  size: 000002e8
20005fe0: 2001d2c8 080a1781 080d5830 000000b8 0809bd85 00000010 20004054 00000003
20006000: 00000000 080a1a11 080a19fd 080a9f5d 00000000 2001ba7c 080c699a 2001a960
20006020: 2001b968 080a1981 2001d044 00000000 00000000 00000000 00000000 20006f20
sp:     2001d118
User stack:
  base: 2001d2c8
  size: 0000063c
2001d100: 3b20f2b0 3d38213e 3d38213e 3a83126f 20000010 00000031 20000dc8 b600ee13
2001d120: 2001ba7c 080c699a 2001a960 080740e5 3c6cb338 2001d18c 0e702b15 00000000
2001d140: 61205057 7469746c 20656475 7373696d 66206465 3520726f 3c006d33 2001ac00
2001d160: 00000000 2001a960 2001b968 080b7843 00000000 2001a960 00000000 2001b968
2001d180: 00000000 08033dd1 080c699a 00000035 0e702ad1 0803464d 00000000 00000000
2001d1a0: 00000001 2001b968 20002a78 2001a960 00000001 00000001 2001b0c8 00000000
2001d1c0: 00000000 080311c9 0000000c 0802f52d 2001b968 0802ecf7 08c415aa 00000000
2001d1e0: cec46a32 4047096d e4e6ab2f 402d3925 438c7a5c 3f1b8e07 3f57ede6 bfdbbb92
2001d200: 3f7f6d48 00000000 00000000 00000000 00000001 ffffffff 00ffffff ffffffff
2001d220: ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff 00000003 00000000
2001d240: 00000101 2001c7ec 0000000e 00000000 00000001 2001c89c ffffffff ffffffff
2001d260: 00000000 00000000 00000000 00000000 00000000 00000000 0809d238 01000000
2001d280: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
2001d2a0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 0809d257
2001d2c0: 00000000 00000000 ffffffff 542b5a4c f454f367 6c861480 000001a0 80000650
R0: 20000dc8 b600ee13 2001d138 2001d138 2001ba7c 080c699a 2001a960 2001b968
R8: 00000001 2001b0c8 00000000 00000000 00000000 2001d118 080740e5 08077e56
xPSR: 61000000 BASEPRI: 00000000 CONTROL: 00000000
EXC_RETURN: ffffffe9`

`(gdb) info line *0x080740e5
Line 82 of "/home/vid/C-Astral/src/AP-Firmware/src/modules/systemlib/mavlink_log.c" starts at address 0x80740de <mavlink_vasprintf+46> and ends at 0x80740e6 <mavlink_vasprintf+54>.

(gdb) info line *0x08077e56
Line 371 of "/home/vid/C-Astral/src/AP-Firmware/src/modules/uORB/uORBDevices_nuttx.cpp" starts at address 0x8077e56 <uORB::DeviceNode::publish(orb_metadata const*, void*, void const*)+2>
   and ends at 0x8077e5a <uORB::DeviceNode::publish(orb_metadata const*, void*, void const*)+6>.
`

I attached the dump and some source code (limited to 10MB so not all is there) if anyone is willing to take a look.  I would appreciate it if you could give me some directions on how to debug this further myself or you can have deeper look yourself.

Let me know if there is any other info you would like and how can I help.

P.S.
Where can I find some more information on what the compete hard fault dump means, what are the addresses and what do they mean to me?

Cheers

[src_and_dump.tar.gz](https://github.com/PX4/Firmware/files/527386/src_and_dump.tar.gz)

kd0aij, 2016-10-13:
can you share a flight log?
Is this a fixed wing or multirotor?

vvranjek, 2016-10-13:
Sure. It's a FW.

[2016-10-13.tar.gz](https://github.com/PX4/Firmware/files/527535/2016-10-13.tar.gz)

kd0aij, 2016-10-13:
A fixed wing plane should still be flyable in manual mode even after the FMU hardfaults.

vvranjek, 2016-10-13:
Really? I didn't know that. But I did switch to manual I switched to manual with the remote as the plane was already heading to the ground but I didn't notice any change. Do you know of any tricks to to trigger hard fault manually, so I can test this?

Also, why doesn't FMU reboot after hard fault?

kd0aij, 2016-10-13:
QGC should show that you're in MANUAL mode (not STABILIZED).  The first minute of your log is in stabilized mode with zero thrust, then you switch to mission mode. In stabilized mode, you won't have control if the FMU is hung. 

I don't know whether reboot is possible after a hard fault.

Your preflight messages show:
```
 PID COMMAND                   CPU(ms) CPU(%)  USED/STACK PRIO(BASE) STATE 
 188 parachute                   19793 12.950   804/ 3592 140 (140)  READY 
```
That's a lot of stack space and a lot of cpu; how much free RAM do you have after booting?

vvranjek, 2016-10-13:
Ahh, so manual, not stabilized.. check! I know and don't mind the parachute module, it's in development. Other then stack and cpu usage it shouldn't be causing any damage.

julianoes, 2016-10-13:
> src_and_dump.tar.gz

It is hard to dig into a source like this and would take up too much time, at least for me.
What would be more helpful is if you could push your changes to your fork on Github which allows to see the git history, so where you forked off and what you changed/added.

vvranjek, 2016-10-13:
Sure, this is the one: https://github.com/vvranjek/Firmware/tree/dev-masterpx4

Thanks for the effort!

vvranjek, 2016-10-15:
I stumbled upon this bugfix https://github.com/PX4/Firmware/commit/237bdfdb61ea335df7c2baca0588fd7fa3e26c6d and realized that I have potentially the same issue in my line of code:

`mavlink_and_console_log_critical(_navigator->get_mavlink_log_pub(),
										 "Geofence exceeded vertical by %.1f m",
									         (double)(dist_z - max_vertical_distance));
`

Could this have caused the crash?

LorenzMeier, 2016-10-15:
I couldn't immediately find the smoking gun but its quite likely you ran out of stack space for floating point printing. That's particularly expensive and we do not allocate extra space by default for console output.

vvranjek, 2016-10-15:
This mast have been the issue... I certainly hope so. I'll report back if the problem persists. Thank a lot!

julianoes, 2016-10-15:
@vvranjek I had a look through the changes but didn't find any other obvious smoking gun. Let's hope the fix above was what triggered it.

julianoes, 2016-10-21:
Closing for now, please re-open if the issue is back.

closed 2016-10-21 by julianoes