PX4/PX4-Autopilot #5663: can't run gazebo simulation on ubuntu

zhahaoyu, 2016-10-14:
My ubuntu version is 16.04, and I have followed all the setup steps on px4.io. When I run the simulation with gazebo,  `make posix_sitl_default gazebo`, I got fatal error.

Here is an excerpt of the console message, where it starts to report erros. 

`/home/haoyu/src/Firmware/Tools/sitl_gazebo/src/gazebo_gimbal_controller_plugin.cpp:198:6: error: prototype for ‘void gazebo::GimbalControllerPlugin::OnPitchStringMsg(ConstGzStringPtr&)’ does not match any in class ‘gazebo::GimbalControllerPlugin’
 void GimbalControllerPlugin::OnPitchStringMsg(ConstGzStringPtr &_msg)
      ^
In file included from /home/haoyu/src/Firmware/Tools/sitl_gazebo/src/gazebo_gimbal_controller_plugin.cpp:20:0:
/home/haoyu/src/Firmware/Tools/sitl_gazebo/include/gazebo_gimbal_controller_plugin.hh:49:19: error: candidate is: void gazebo::GimbalControllerPlugin::OnPitchStringMsg(ConstAnyPtr&)
     private: void OnPitchStringMsg(ConstAnyPtr &_msg);
                   ^
/home/haoyu/src/Firmware/Tools/sitl_gazebo/src/gazebo_gimbal_controller_plugin.cpp:205:6: error: prototype for ‘void gazebo::GimbalControllerPlugin::OnRollStringMsg(ConstGzStringPtr&)’ does not match any in class ‘gazebo::GimbalControllerPlugin’
 void GimbalControllerPlugin::OnRollStringMsg(ConstGzStringPtr &_msg)
      ^
In file included from /home/haoyu/src/Firmware/Tools/sitl_gazebo/src/gazebo_gimbal_controller_plugin.cpp:20:0:
/home/haoyu/src/Firmware/Tools/sitl_gazebo/include/gazebo_gimbal_controller_plugin.hh:50:19: error: candidate is: void gazebo::GimbalControllerPlugin::OnRollStringMsg(ConstAnyPtr&)
     private: void OnRollStringMsg(ConstAnyPtr &_msg);
                   ^
/home/haoyu/src/Firmware/Tools/sitl_gazebo/src/gazebo_gimbal_controller_plugin.cpp:212:6: error: prototype for ‘void gazebo::GimbalControllerPlugin::OnYawStringMsg(ConstGzStringPtr&)’ does not match any in class ‘gazebo::GimbalControllerPlugin’
 void GimbalControllerPlugin::OnYawStringMsg(ConstGzStringPtr &_msg)
      ^
In file included from /home/haoyu/src/Firmware/Tools/sitl_gazebo/src/gazebo_gimbal_controller_plugin.cpp:20:0:
/home/haoyu/src/Firmware/Tools/sitl_gazebo/include/gazebo_gimbal_controller_plugin.hh:51:19: error: candidate is: void gazebo::GimbalControllerPlugin::OnYawStringMsg(ConstAnyPtr&)
     private: void OnYawStringMsg(ConstAnyPtr &_msg);
                   ^
CMakeFiles/rotors_gazebo_gimbal_controller_plugin.dir/build.make:62: recipe for target 'CMakeFiles/rotors_gazebo_gimbal_controller_plugin.dir/src/gazebo_gimbal_controller_plugin.cpp.o' failed
make[7]: *** [CMakeFiles/rotors_gazebo_gimbal_controller_plugin.dir/src/gazebo_gimbal_controller_plugin.cpp.o] Error 1
CMakeFiles/Makefile2:178: recipe for target 'CMakeFiles/rotors_gazebo_gimbal_controller_plugin.dir/all' failed
make[6]: *** [CMakeFiles/rotors_gazebo_gimbal_controller_plugin.dir/all] Error 2
make[6]: *** Waiting for unfinished jobs....
[ 54%] Building CXX object CMakeFiles/rotors_gazebo_mavlink_interface.dir/src/geo_mag_declination.cpp.o
[ 57%] Linking CXX shared library libLiftDragPlugin.so
[ 57%] Built target LiftDragPlugin
[ 60%] Linking CXX shared library librotors_gazebo_controller_interface.so
[ 60%] Built target rotors_gazebo_controller_interface
[ 62%] Linking CXX shared library librotors_gazebo_mavlink_interface.so
[ 62%] Built target rotors_gazebo_mavlink_interface
Makefile:149: recipe for target 'all' failed
make[5]: *** [all] Error 2
src/firmware/posix/CMakeFiles/sitl_gazebo.dir/build.make:110: recipe for target 'external/Stamp/sitl_gazebo/sitl_gazebo-build' failed
make[4]: *** [external/Stamp/sitl_gazebo/sitl_gazebo-build] Error 2
CMakeFiles/Makefile2:9086: recipe for target 'src/firmware/posix/CMakeFiles/sitl_gazebo.dir/all' failed
make[3]: *** [src/firmware/posix/CMakeFiles/sitl_gazebo.dir/all] Error 2
CMakeFiles/Makefile2:9960: recipe for target 'src/firmware/posix/CMakeFiles/gazebo.dir/rule' failed
make[2]: *** [src/firmware/posix/CMakeFiles/gazebo.dir/rule] Error 2
Makefile:3497: recipe for target 'gazebo' failed
make[1]: *** [gazebo] Error 2
Makefile:160: recipe for target 'posix_sitl_default' failed
make: *** [posix_sitl_default] Error 2
`

julianoes, 2016-10-14:
Hi, what is the output of `gazebo --version`?

zhahaoyu, 2016-10-14:
Thanks @julianoes for your help!

Here is the output.

`Gazebo multi-robot simulator, version 7.4.0
Copyright (C) 2012-2016 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

Gazebo multi-robot simulator, version 7.4.0
Copyright (C) 2012-2016 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

`

julianoes, 2016-10-14:
@zhahaoyu ok I have the same. 

I'm assuming you have tried all these:

```
make clean
git clean -d -f -x (-n)
make submodulesclean
```

At what commit are you at?
```
git rev-parse HEAD
```

julianoes, 2016-10-14:
Nevermind, I can reproduce your issue. Looking into it.

julianoes, 2016-10-14:
This is currently broken again, working on it...

closed 2016-10-14 by julianoes

julianoes, 2016-10-14:
@zhahaoyu please cross test, it should work again.

zhahaoyu, 2016-10-14:
The previous error was resolved. However, as the script opens Gazebo, there is no world initialized. The error message from terminal is
 
[Msg] Waiting for master.
[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.1.70
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.1.70
[Wrn] [ModelDatabase.cc:340] Getting models from[http://gazebosim.org/models/]. This may take a few seconds.
[Err] [Node.cc:105] No namespace found
[Err] [Node.cc:105] No namespace found
[Err] [Node.cc:105] No namespace found
[Err] [Node.cc:105] No namespace found
[Err] [Node.cc:105] No namespace found
[Err] [Node.cc:105] No namespace found
[Err] [Node.cc:105] No namespace found
[Err] [Node.cc:105] No namespace found
[Err] [Node.cc:105] No namespace found

PR #5665 (2016-10-14): sitl_gazebo: update submodule
Fixes #5663.
PR #5668 (2016-10-14): sitl_gazebo: update submodule yet again
This fixes the build for Gazebo >= 7.4.

Closes #5663.