PX4/PX4-Autopilot #5667: Running the px4 app without usb

anujkumargupta, 2016-10-14:
Hi 
I am working on Snapdragon
Can i run the px4 app on snapdragon board using wifi (without USB)

Actually i am running this using usb on FC ./px4 mainapp.config but as soon as i remove my usb the app stop running and communication was lost between QGC and FC.(Fc is also connected with WIFI and can able to ping the device)

Thanks

ndepal, 2016-10-14:
You should be able to run it by going into the device via SSH. Have you tried that?

julianoes, 2016-10-14:
@anujkumargupta You can either try to use the autostart as described in the Devguide. Or alternatively, just open a screen session and start it there. It should then survive when you disconnect USB and you can re-attach it later if you need to.

narendrakc-path, 2016-10-15:
Thanks @julianoes and @ndepal , Your both solutions works perfectly but i tried to going into the device using ssh but for that i need the root id and password but i did not find it any where.

mhkabir, 2016-10-15:
IIRC "root" and "linaro" (password).

closed 2016-10-23 by julianoes