PX4/PX4-Autopilot #5669: Mixer not loading

dennisss, 2016-10-15:
When trying to load a mixer in hardware (tested Pixracer + CF2), the file is incorrectly parsed:
```
NuttShell (NSH)
nsh> mixer load /dev/pwm_output0 /etc/mixers/quad_x.main.mix
[mixer] remaining in buf: 138, first char: M
[mixer] unrecognised geometry '4x 1000'
ERROR [mixer] failed to load mixers from /etc/mixers/quad_x.main.mix
ERROR [mixer] failed to load mixer
nsh> 
```

This commit seems to be causing it: https://github.com/PX4/Firmware/commit/db44129ec099a05debf9187da2fd09035c9a67d7

Proposed fix: https://github.com/PX4/NuttX/pull/86

Would be nice if someone could test to validate this. Thanks!

tubeme, 2016-10-17:
I dont know how the mixer looks like but it is obvious to me that the line for the quad mixer starts with letter M in the mixer so after M it could not recognize the 4x geometry. The MC mixer should look like this:

R: 4x 10000 10000 10000 0

dennisss, 2016-10-17:
The "M" in the output is referring to the next unparsed character of the line after the "R: 4x 10000..." line failed to parse. The issue was that sscanf was not working correctly which is fixed by an open PR on this repo updating the NuttX libraries.

tubeme, 2016-10-17:
OK then it is something else. Sorry.

closed 2016-10-17 by dennisss