PX4/PX4-Autopilot #5670: When i compiler firmware by eclipse in the Linux ,could i set a breakpoint or compile a single file?

xuqintao, 2016-10-15:
If you can do this, it will be very efficient。i really need your help .

julianoes, 2016-10-15:
@xuqintao is this for SITL or debugging on a Pixhawk?

xuqintao, 2016-10-16:
@julianoes   yes， do you konw can do this?

mhkabir, 2016-10-16:
@xuqinato Please answer the question about what you're trying to do this for? - 
i. SITL
ii. Debugging on a Pixhawk

xuqintao, 2016-10-17:
@mhkabir  debugging on a Pixhawk.

julianoes, 2016-10-17:
There used to be instructions to use the black magic probe for breakpoint or stepping through debugging, see:
https://pixhawk.org/dev/jtag/black_magic_probe
https://pixhawk.org/dev/jtag/start?s[]=arm&s[]=none&s[]=eabi&s[]=gdb
https://pixhawk.org/dev/gdb_cheatsheet?s[]=arm&s[]=none&s[]=eabi&s[]=gdb

xuqintao, 2016-10-17:
@julianoes  thanks.  Can we open and compile project files by Eclipse in Linux? i can not do that ,and i want to make sure it .

julianoes, 2016-10-17:
Yes you can use eclipse, try:
https://github.com/PX4/Firmware/blob/master/eclipse.project
https://github.com/PX4/Firmware/blob/master/eclipse.cproject

xuqintao, 2016-10-18:
@julianoes OK，thank you very much.

julianoes, 2016-10-18:
@xuqintao no worries.

closed 2016-10-18 by julianoes