PX4/PX4-Autopilot #5671: simulation - optical flow mode no position lock obtained

zhahaoyu, 2016-10-15:
system: ubutnu 16.04
gazebo: 7.04
firmware: 2d039af7a4299c6565e0d47f333adad65fd0c0a0

When I tried the gazebo simulation with optical flow, there is no` [commander] home: ....` message output. I was not able to ask the quadrotor to take off. Here is the output:

____  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [dataman] Unkown restart, data manager file 'rootfs/fs/microsd/dataman' size is 103090 bytes
Gazebo multi-robot simulator, version 7.4.0
Copyright (C) 2012-2016 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
Gazebo multi-robot simulator, version 7.4.0
Copyright (C) 2012-2016 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.1.70
[Msg] Publicized address: 192.168.1.70
+ NAV_ACC_RAD: curr: 2.0000 -> new: 12.0000
+ RTL_DESCEND_ALT: curr: 5.0000 -> new: 10.0000
+ RTL_LAND_DELAY: curr: 5.0000 -> new: 0.0000
+ MC_ROLLRATE_P: curr: 0.2000 -> new: 0.3000
+ MC_PITCHRATE_P: curr: 0.2000 -> new: 0.3000
+ MC_PITCH_P: curr: 6.0000 -> new: 5.5000
+ MC_ROLL_P: curr: 6.0000 -> new: 5.5000
  MC_ROLLRATE_I: curr: 0.0500 -> new: 0.1000
  MC_PITCHRATE_I: curr: 0.0500 -> new: 0.1000
+ MPC_Z_VEL_P: curr: 0.6000 -> new: 0.8000
+ MPC_XY_VEL_P: curr: 0.1000 -> new: 0.1500
+ MPC_XY_VEL_I: curr: 0.1000 -> new: 0.2000
  LPE_GPS_ON: curr: 1 -> new: 0
  MPC_ALT_MODE: curr: 0 -> new: 1
  LPE_T_MAX_GRADE: curr: 1.0000 -> new: 0.0000
  MPC_XY_VEL_MAX: curr: 8.0000 -> new: 2.0000
+ MPC_Z_VEL_MAX: curr: 1.5000 -> new: 2.0000
+ MIS_TAKEOFF_ALT: curr: 2.5000 -> new: 2.0000

INFO  [platforms__posix__drivers__ledsim] LED::init
INFO  [platforms__posix__drivers__ledsim] LED::init
INFO  [simulator] Not using /dev/ttyACM0 for radio control input. Assuming joystick input via MAVLink.
INFO  [simulator] Waiting for initial data on UDP port 14560. Please start the flight simulator to proceed..
Warning [parser.cc:778] XML Element[robotNamespace], child of element[sensor] not defined in SDF. Ignoring[robotNamespace]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
[Wrn] [gazebo_lidar_plugin.cpp:96] [gazebo_lidar_plugin] Please specify a robotNamespace.
[Dbg] [gazebo_mavlink_interface.cpp:125] <joint_name> not found for channel[0] no joint control will be performed for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:125] <joint_name> not found for channel[1] no joint control will be performed for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:125] <joint_name> not found for channel[2] no joint control will be performed for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:125] <joint_name> not found for channel[3] no joint control will be performed for this channel.
[Wrn] [gazebo_mavlink_interface.cpp:114] joint [zephyr_delta_wing::propeller_joint] not found for channel[4] no joint control for this channel.
[Wrn] [gazebo_mavlink_interface.cpp:114] joint [zephyr_delta_wing::flap_left_joint] not found for channel[5] no joint control for this channel.
[Wrn] [gazebo_mavlink_interface.cpp:114] joint [zephyr_delta_wing::flap_right_joint] not found for channel[6] no joint control for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:125] <joint_name> not found for channel[7] no joint control will be performed for this channel.
**[Err] [Scene.cc:2928] Light [sun] not found. Use topic ~/factory/light to spawn a new light.**
[Wrn] [Publisher.cc:141] Queue limit reached for topic /gazebo/default/iris_opt_flow/motors, deleting message. This warning is printed only once.
[Wrn] [Publisher.cc:141] Queue limit reached for topic /gazebo/default/iris_opt_flow/motor_speed/0, deleting message. This warning is printed only once.
[Wrn] [Publisher.cc:141] Queue limit reached for topic /gazebo/default/iris_opt_flow/motor_speed/1, deleting message. This warning is printed only once.
[Wrn] [Publisher.cc:141] Queue limit reached for topic /gazebo/default/iris_opt_flow/motor_speed/2, deleting message. This warning is printed only once.
[Wrn] [Publisher.cc:141] Queue limit reached for topic /gazebo/default/iris_opt_flow/motor_speed/3, deleting message. This warning is printed only once.
[Wrn] [Publisher.cc:141] Queue limit reached for topic /gazebo/default/iris_opt_flow/imu, deleting message. This warning is printed only once.
[Wrn] [RaySensor.cc:359] ranges not constructed yet (zero sized)
INFO  [simulator] Got initial simuation data, running sim..
INFO  [pwm_out_sim] MODE_8PWM
INFO  [tone_alarm] startup
INFO  [mavlink] mode: Normal, data rate: 2000000 B/s on udp port 14556 remote port 14550
INFO  [mavlink] mode: Onboard, data rate: 2000000 B/s on udp port 14557 remote port 14540
INFO  [sdlog2] [blackbox] rootfs/fs/microsd/log/2016-10-15
WARN  [sdlog2] sdlog2: failed setting sched params
INFO  [sdlog2] [blackbox] recording: 05_55_04.px4log
INFO  [local_position_estimator] [lpe] Lidar detected with ID 0
INFO  [local_position_estimator] [lpe] land init
INFO  [mavlink] MAVLink only on localhost (set param MAV_BROADCAST = 1 to enable network)
pxh> INFO  [local_position_estimator] [lpe] baro init 0 m std -2147483648 cm
INFO  [local_position_estimator] [lpe] z resume 
INFO  [local_position_estimator] [lpe] tz resume 
INFO  [local_position_estimator] [lpe] lidar init: mean 10 cm stddev 1 cm
**commander takeoff**
**WARN  [commander] rejecting takeoff, no position lock yet. Please retry..**
pxh> INFO  [platforms__posix__px4_layer] simulator is slow. Delay added: 10503 us
INFO  [platforms__posix__px4_layer] simulator is slow. Delay added: 10715 us
INFO  [platforms__posix__px4_layer] simulator is slow. Delay added: 15125 us
INFO  [platforms__posix__px4_layer] simulator is slow. Delay added: 11521 us
INFO  [platforms__posix__px4_layer] simulator is slow. Delay added: 10046 us
INFO  [platforms__posix__px4_layer] simulator is slow. Delay added: 15196 us
INFO  [platforms__posix__px4_layer] simulator is slow. Delay added: 11106 us

After the building on optical flow mode, I went back to try the regular mode. Now the regular mode does set a position either. But if I clean up all the files, and rebuild the regular mode, it works fine. The problem is just, whenever I build with the optical mode, I can't run any mode unless I clean up everything.

mhkabir, 2016-10-15:
You need to take off in manual and then switch to position control, because the optical flow camera cannot work and be in focus until some distance off the ground.

zhahaoyu, 2016-10-15:
Thanks for your time @mhkabir !

I am struggling with the possible modes and its syntax. I don't know how to use manual mode, and how can I give more power on throttle? Is there any reference on possible commands?

mhkabir, 2016-10-15:
Use a joystick via QGC for manual control.

mhkabir, 2016-10-15:
http://px4.io/docs/flight-mode-selection/

closed 2016-10-15 by mhkabir

zhahaoyu, 2016-10-15:
Is there a way to directly control the vehicle directly through command line?