PX4/PX4-Autopilot #5675: Bebop ADB push fails with "no space left on device"

mhkabir, 2016-10-16:
Flashing PX4 onto a brand new Bebop, and the ADB uploader fails with a "no space left on device" when copying over the PX4 executable. Also tried pushing it manually with `adb push px4 /usr/bin`, and the same problem occurs.

This issue did not happen when I manually pushed over the mixer and config files to `/home/root`.

I verified using `df` that there is indeed enough space (there should be, considering that it's an untouched filesystem).

@eyeam3 @jberaud Any ideas as to why this is happening?

eyeam3, 2016-10-17:
I don't have the hardware with me right now, but it seems that the firmware-binary has increased in size recently. So far I had enough memory on the Bebop and had no issues when copying.

I will have a look at it tomorrow, but for now, you could run:
`${RPI_TOOLCHAIN_DIR}/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/arm-linux-gnueabihf-strip build_posix_bebop_default/src/firmware/posix/px4`
The binary will then be about 2MB in size and fit into memory. However, you have to copy it by hand to the Bebop.

mhkabir, 2016-10-18:
Okay, that's fine. Strip should work. We should track down why the binary recently increased in size, however. I'm also curious as to why I the root partition has just 9% free in the first place.

@jberaud Anything I can clear out safely? There might be leftover files from the auto update it did via the FreeFlight app when I first connected (?) 

@eyeam3 Can you please post the output of  `df` here when you have the hardware with you.

jberaud, 2016-10-18:
@mhkabir you wouldn't want to do that. There is nothing leftover as the update process wipes everything except the media directory.
If you want to have more room, you can execute everything from the emmc, which is mounted in
/data/ftp/internal_000/

eyeam3, 2016-10-18:
Okay, I looked into the issue. About three weeks ago there were two commits b1708f3 and 8ff237c which remove the size optimization from modules and add it back for the nuttx platform. These changes are the reason, why the binary is to large to fit into the system partition now. However, it was already tight-fitting and adding other modules eventually would have caused the same error. 

This is the memory usage on my Bebop:
```
Filesystem           1M-blocks      Used Available Use% Mounted on
ubi1:system                 42        34         6  86% /
udev                       145         0       145   0% /dev
shm                        145         0       145   0% /dev/shm
tmp                        145         0       145   0% /tmp
ubi0:factory                 5         0         4   2% /factory
ubi2:data                    9         0         8   2% /data
ubi2:update                 28         0        27   0% /update
/dev/mmcblk0              7339      1120      6219  15% /data/ftp/internal_000
```

I then tried to optimize for size during the build process and added `-Os` back. In addition, I tried various build flags which indeed reduce the binary's size but would not solve the problem in the long term. I would also prefer to have the executable in `/usr/bin` for ease of use and suggest to strip the binary before uploading. This could be done in `./Tools/adb_upload_to_bebop.sh` by adding the lines:
```
${RPI_TOOLCHAIN_DIR}/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/arm-linux-gnueabihf-strip \
    -R .comment -R .gnu.version \
    ../build_posix_bebop_default/src/firmware/posix/px4
```
The resulting 2MB executable would definitely fit into memory and if needed one could still compile the Firmware, upload it by hand to the emmc and run it from there.

LorenzMeier, 2016-10-18:
@eyeam3 Can we add the strip step to the default build steps for now?

eyeam3, 2016-10-18:
I just added a PR #5693. In that case it would be executed when uploading the default build, but the build itself would not be changed.
Is this also okay?

LorenzMeier, 2016-10-18:
Makes sense - I'll close this for now.

closed 2016-10-18 by LorenzMeier