PX4/PX4-Autopilot #5676: Default open loop loiter params don't seem right.

nanthony21, 2016-10-16:
The default roll in open loop loiter following a gps failure is 15deg, shouldn't it be 0?
Also 70% thrust seems a bit high to me, depends on the vehicle though.

julianoes, 2016-10-21:
0 degrees would go straight but loiter should really do a circle, so 15 degrees would make sense.

nanthony21, 2016-10-21:
[image]

Here's what I have under "GPS FAILURE NAVIGATION" the 70% thrust is at the bottom.

Circling makes sense for fixed wing, but for a multicopter it can cause the vehicle to randomly fly off during a hover or takeoff, which is exactly what I've experienced.

dagar, 2017-04-12:
The open loop loiter isn't enabled anymore. MC should land immediately with GPS failure.

closed 2017-04-12 by dagar