PX4/PX4-Autopilot #5677: External MPU9250 Not registering

JakeLeaf, 2016-10-17:
I have an external MPU9250 on a GPS chip connected via i2c and I cant get the MPU to work (I know for awhile the driver was only SPI based but that seems to be changed). Won't register or work at all. Also has a BMP280 onboard but can only get to it via i2c so one issue at a time.
Any ideas? I have a couple of these modules so it would be great to get it all in one. This is the pinout and GPS [http://www.reyax.com/Module/GPS/RY836AI/RY836AI.pdf](url). 

Thanks!

Jake

LorenzMeier, 2016-12-28:
You will need to look into the driver and see if it scans the external bus. You might also have more luck with recent Firmware versions.

closed 2016-12-28 by LorenzMeier

dagar, 2018-08-01:
@Mtbarros both of those drivers already have i2c support. They're likely only missing appropriate startup code. Please open a new issue if you'd like to continue discussing and can help with testing.

Mtbarros, 2018-08-01:
Just opened [a new issue.](https://github.com/PX4/Firmware/issues/10109) I can help some testing if needed. Thanks for the quick response.