PX4/PX4-Autopilot #5678: VTOL: Tailsitter: when i test in mission mode, the plane sometimes can land on the landing spot, sometimes it can not or just float away, Dose anybody know why?

applededipan, 2016-10-17:
VTOL: Tailsitter: I test the vtol in mission mode these days, the plane sometimes can land on the landing spot, sometimes it can not or just float away, Dose anybody know why?

julianoes, 2016-10-21:
Could you share a log? Can you comment @CarlOlsson?

applededipan, 2016-10-22:
ok, i will share logs later. i am wondering that if the position controller work during landing phase?

julianoes, 2016-10-22:
@applededipan I think the catch is that the roll/pitch is limited during the land phase in order to be level when landing. However, this can have the effect that it drifts off in wind and can't keep up.

applededipan, 2016-10-24:
I think so! and I find that the position controller and altitude controller are still working during landing, Am I right?

applededipan, 2016-10-24:
I think so! and I find that the position controller and altitude controller are still working during landing, Am I right? @julianoes

julianoes, 2016-10-24:
I think so, yes.

applededipan, 2016-11-07:
as i tune the roll/pitch limit a bit larger, it works better now! thanks for your help!!! @julianoes

closed 2016-11-24 by dagar