PX4/PX4-Autopilot #5680: HIL: disable lockdown not working

vvranjek, 2016-10-17:
When in HIL mode (SYS_AUTOSTART: 1000) I find it impossible to enable servos. I suppose that running the command `px4io lockdown disable` should do the trick but nothing happens. Also running `pwm arm` in HIL returns 
```
nsh> pwm arm
ERROR [pwm] PWM_SERVO_SET_ARM_OK
ERROR [pwm] Task exited with errno=22
```
Am I missing something?

Version:
```
nsh> ver all
HW arch: PX4FMU_V2
FW git-hash: 263b2a63ea6abfab48b00b713e4d3c77d20213e5
FW version: v1.5.0-1-g263b2a6 (1.5.0 0), 17104896
OS version: 6.27 (4294967295)
Build datetime: Oct 17 2016 13:39:11
Toolchain: 4.7.4 20140401 (release) [ARM/embedded-4_7-branch revision 209195]
MCU: STM32F42x, rev. A
```

vvranjek, 2016-10-17:
And the boot log...

```
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
WARN  [param] selected parameter default file /fs/mtd_params
rgbled on I2C bus 2 at 0x55 (bus: 100 KHz, max: 100 KHz)
WARN  [px4io] CRCs match
INFO  [dataman] Power on restart, data manager file '/fs/microsd/dataman' size is 103090 bytes
MS5611_SPI on SPI bus 1 at 3 (20000 KHz)
WARN  [bst] no devices found
INFO  [ver] match: PX4FMU_V2
WARN  [hmc5883] no device on bus 2
read_reg failWARN  [lis3mdl] no device on bus 2
WARN  [hmc5883] no device on bus 1
WARN  [mpu6000] no device on bus 3
WARN  [mpu6000] no device on bus 4
MPU6000 on SPI bus 1 at 4 (1000 KHz)
WARN  [mpu6000] no device on bus 4
nsh: mpu9250: command not found
L3GD20 on SPI bus 1 at 1 (11000 KHz)
LSM303D on SPI bus 1 at 2 (11000 KHz)
ERROR [meas_airspeed] no MS4525 airspeed sensor connected
ERROR [ets_airspeed] no ETS airspeed sensor connected
ERROR [ets_airspeed] no ETS airspeed sensor connected
nsh: sf1xx: command not found
INFO  [pwm_out_sim] MODE_8PWM
INFO  [mavlink] mode: Normal, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
INFO  [mavlink] mode: OSD, data rate: 1000 B/s on /dev/ttyS2 @ 57600B
INFO  [ver] match: PX4FMU_V2
nsh: px4flow: command not found
px4flow [176:100]
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
WARN  [sdlog2] [blackbox] no space on MicroSD: 28 MiB
WARN  [sdlog2] ERR: MicroSD almost full
INFO  [init] Fixedwing 
INFO  [init] Mixer: /etc/mixers/AERT.main.mix on /dev/pwm_output0 
INFO  [init] Mixer: /etc/mixers/pass.aux.mix on /dev/pwm_output1 

INFO  [init] Addons script: /fs/microsd/etc/extras.txt 
STOPPING MAVLINK 
Startin mavLink 
WARN  [mavlink] /dev/ttyS1 already running
Processes: 21 total, 3 running, 18 sleeping
CPU usage: 14.88% tasks, 0.50% sched, 84.62% idle
DMA Memory: 5120 total, 2560 used 2560 peak
Uptime: 8.048s total, 5.057s idle

 PID COMMAND                   CPU(ms) CPU(%)  USED/STACK PRIO(BASE) STATE 
   0 Idle Task                    5057 84.623     0/    0   0 (  0)  READY 
   1 hpwork                         67  1.785   668/ 1592 192 (192)  w:sig 
   2 lpwork                         30  0.496   580/ 1592  50 ( 50)  w:sig 
   3 init                         1504  0.000  1236/ 2496 100 (100)  w:sem 
 235 top                            10  0.000  1236/ 1696 100 (100)  RUN   
  93 dataman                        85  0.000   724/ 1192  90 ( 90)  w:sem 
 121 sensors                       255  3.869   948/ 1496 250 (250)  w:sem 
 123 commander                     119  1.785  2572/ 2992 140 (140)  w:sig 
 124 commander_low_prio              0  0.000   540/ 2992  50 ( 50)  w:sem 
 133 pwm_out_sim                     0  0.000   656/ 1192 100 (100)  w:sem 
 135 px4io                          98  0.992  1028/ 1392 240 (240)  w:sem 
 141 mavlink_if0                    66  0.892  1676/ 2392 100 (100)  w:sig 
 142 mavlink_rcv_if0                 0  0.000   940/ 2096 175 (175)  w:sem 
 149 mavlink_if1                    49  0.694  1508/ 2392 100 (100)  w:sig 
 150 mavlink_rcv_if1                 0  0.000   940/ 2096 175 (175)  w:sem 
 179 mavlink_if2                   262  4.365  1356/ 2392 100 (100)  w:sig 
 180 mavlink_rcv_if2                 0  0.000   956/ 2096 175 (175)  w:sem 
 215 ekf2                            1  0.000  2020/ 5992 250 (250)  w:sem 
 218 fw_att_control                  1  0.000  1060/ 1392 250 (250)  w:sem 
 223 fw_pos_ctrl_l1                 15  0.000   620/ 1392 250 (250)  w:sem 
 232 navigator                       1  0.000   804/ 1296 105 (105)  w:sem 

NuttShell (NSH)
nsh> WARN  [mavlink] Disabling hardware flow control
WARN  [mavlink] Disabling hardware flow control

nsh> px4io lockdown disable
WARN  [px4io] WARNING: ACTUATORS WILL BE LIVE IN HIL! PROCEED?
WARN  [px4io] Press 'y' to enable, any other key to abort.
WARN  [px4io] WARNING: ACTUATORS ARE NOW LIVE IN HIL!
nsh> pwm arm
ERROR [pwm] PWM_SERVO_SET_ARM_OK
ERROR [pwm] Task exited with errno=22

nsh> 

```

LorenzMeier, 2016-10-17:
Are you powering the servo rail externally? Because the autopilot is not powering it.

And you have the order wrong: Arm first, then disable the lockdown.

vvranjek, 2016-10-18:
I am definitely powering the servo rail. Arming and servos work just fine in regular mode. I've been trying to make it work for two days now and have probably tried every possible combination. Arming in QGC works fine while `px4io lockdown disable` does nothing. No matter what I do I get the errno=22 with `pwm arm` (only HIL mode). **What does error 22 mean?**

If it helps in any way, I have one pixhawk dedicated only to simulation thus having no receiver connected. Just power and a few servos. I am powering the servo rail from usb 5V, no power module.

LorenzMeier, 2016-10-18:
Are you really powering the servo rail (with a 3-pin connector)? Pixhawk does not power the servo rail.

vvranjek, 2016-10-18:
Yes, below is a picture of my workaround for powering servos. The servos were moving just fine with `px4io test' and `pwm test -a -p xxxx`, but in non-HIL mode only. I tried just now again and for some reason they don't move even in non-HIL (3032_skywalker_x5) but there was a lot of tinkering going on including the switch to stock v1.5.0 (to make sure it wasn't caused by my modifications). Let me figure this out on stock version and I'll get back to you. But still, I dont't get the error 22 in non-HIL mode, what does it mean?

[image]

vvranjek, 2016-10-18:
Ok, it didn't work because of IO safety. Forgot about the circuit breaker.

vvranjek, 2016-10-18:
Also in HIL mode I get:
```
nsh> px4io test
WARN  [px4io] Press CTRL-C or 'c' to abort.
ERROR [px4io] servo 0 readback error, got 0 expected 1000
```

vvranjek, 2016-10-18:
I've just tried a different pixhawk, connected the power module, just in case flashed the 1.4.4 version, still the same issues... I don't get it.

kd0aij, 2016-10-18:
How much current do you think is available from USB to drive servos?

vvranjek, 2016-10-19:
The whole system (pixhawk + 2x 9g servos) draw 270mA in idle and about 400mA when moving. Servos are always unloaded since I use this system for the simulation only. There must be at least 500mA available from the USB port.

kd0aij, 2016-10-19:
Still a very bad idea to be running motors off the avionics supply

vvranjek, 2016-10-19:
I am aware of that, and it's exclusively for the simulation, so I don't need external power, it's handy.

Anyway, I managed to get the servos running in HIL by not enforcing lockdown in state_machine_helper.cpp.

This is a temp solution for me since `px4io lockdown disable` isn't doing it for me.

closed 2018-02-21 by PX4BuildBot

SkyDeBaun, 2019-05-22:
Has anyone had success enabling servos in HIL?

hamishwillee, 2019-05-22:
@SkyDeBaun Please don't append to old and closed issues.

HashimAifi, 2020-11-05:
> Has anyone had success enabling servos in HIL?

I got it working. This is what I did:

I used the latest px4 firmware (v1.11.1 - also works on current master branch)
I only changed one file (state_machine_helper.cpp): Commented the line armed->lockdown = true; inside if (hil_enabled) { block
I compiled, built and then uploaded this firmware in QGC v3.5.6
In QGC I disabled HITL then rebooted vehicle and then selected Standard-Plane as the airframe
I pressed the safety switch to check if my servos are working as per the radio control.
Then, I enabled HITL, again rebooted the vehicle and selected 'HIL-Star' as my airframe.
Note: At this point, the servos should still be active and controlled by radio.

I launched XPlane-11 and started a flight
In QGC, I opened the HIL widget to connect with X-Plane 11
Then, I armed the vehicle
After arming; the real servos and the X-Plane aircraft wings should both be working and you should also be able to fly the plane using the radio.

I hope this helps someone.