PX4/PX4-Autopilot #5681: Test worthy scenario for vtol back transition smoothening

sanderux, 2016-10-17:
@AndreasAntener @dagar @tumbili 

After testing some vibration dampening foam I inadvertently changed the level of the FC by a few degrees  forward. (so it considered level to be more pitch down).

The back transition was VERY smooth that way.

What happened was that the amount of braking force it could apply was much less because it reached it's maximum pitch back degrees, this resulted in very little altitude increase and a much smoother back transition. It did overshoot a few meters more.

Perhaps we could limit this programmatically after a back-transition command based on velocity?

sanderux, 2016-10-18:
BAD: (Without level difference) http://logs.uaventure.com/view/Wk24h4djr8EdgVdbexwrt5
GOOD: (With level difference) http://logs.uaventure.com/view/S4Ko4U6LsZkZWBe4ufYZum

tubeme, 2016-10-18:
I think that the braking is very specific to every frame, either it is delta wing, regular plane, wight, wing profile, wing span etc. So making the braking only autonomous will be very hard task given the unknowns. If we have some control over the braking and release will be much simpler task. It will provide this specific setup depending on the frame performance that we can tune.

Your proposed limiting based on velocity  sounds good to me, but isn't it going to introduce another set of problems as it usually happens.

Playing with FC angle changes the performance of the front transition as well, so it is not viable solution.

dagar, 2017-04-11:
Resolved by adding FW_PSP_OFFSET?

sanderux, 2017-04-11:
No, by increasing back transition duration now that we have adaptive acceptance radius

closed 2017-04-11 by sanderux