PX4/PX4-Autopilot #5695: Navio2: still very basic IO support

mantelt, 2016-10-19:
[Release info for version 1.4.1 states:](https://github.com/PX4/Firmware/releases/tag/v1.4.1)
> Full support for RPI2 + Navio 2 (was present for a while, now integrated / flight tested)

Full with enthusiasm, I wanted to get that combination (Navio2 bought in the beginning of the year) finally up in the air on a fixed-wing platform. The enthusiasm was gone quite fast... the input/output part needs still quite some work IMHO. (I don't blame anyone that this is not yet implemented, it's just not what I would call _full support_).

As a could not find a lot of information regarding PX4 on RPi/Navio (and the little information I found was spread all around), I try to summarize here:

At the moment, the IO Co-processor of the NAVIO2 board only decodes PPM (and S.Bus) signals and generates PWM signals. Both is accessed via sysfs (implemented with a kernel module from emlid). 

The existing Input/Output support for PX4 is based on #4938. For output, [it uses a predefined mixer ](https://github.com/PX4/Firmware/blob/master/src/drivers/navio_sysfs_pwm_out/navio_sysfs_pwm_out.cpp#L164) for a quad.

Out of curiosity, I already extended @HidenoriKobayashi and @mayanez work to support any mixer file ([sneak a peek](https://github.com/mantelt/Firmware/tree/feature/improved_navio_pwm_out)). As there are still some flaws (e.g. it only works with control group 3 at the moment) and before I continue I would like to know, if there are any efforts ongoing in this direction already.

Especially, the [comment](https://github.com/PX4/Firmware/pull/4938#issuecomment-229504513) made by @staroselskii in #4938 implies that there is other work going on, by emlid itself?

I anyway would greatly appreciate if it could somehow be managed to run some variant of the PX4IO firmware on the Navio2 IO, as this would introduce the same level of safety with respect to "FMU" (RPi/px4) crashes (by having a manual override only using the IO). I myself have no experience of flashing microcontrollers besides using the nice pipeline offered with Pixhawk...

LorenzMeier, 2016-10-19:
Emlid @staroselskii has not released their IO firmware to date. I think the best shot you have is to fix the hardcoding to a multi rotor mixer (its a small bug, not a big deal) and work from there.

The WIP change you had looks good. Can you extend that to include all four output groups? They can be done in parallel which avoids hardcoding the wrong thing.

mantelt, 2016-10-19:
Thanks @LorenzMeier for your input. As this is the first time I'm working on the IO part of PX4 code, I'm sort of learning by doing. But I also just figured out that I would need to iterate over all actuator output groups.
I will continue on my changes if you think it could be useful and make a PR as soon as it's in a decent state.

In the meantime, I got some info from EMLID, although its unfortunately still quite unspecific: https://community.emlid.com/t/io-co-processor/4114/5?u=thomasm

nicolaerosia, 2017-05-29:
can this be closed?

paan2097, 2017-06-02:
@mantelt @LorenzMeier @nicolaerosia I'm still running into multiple issues that I've posted on both discuss.px4 and community.emlid forums just getting the initial install running without errors. Right now I'm getting a bunch of GPIO erros and the servos are over extending using the predefined mixer file. Here is my topic: http://discuss.px4.io/t/failure-of-px4-plane-code-running/3436

nicolaerosia, 2017-06-05:
@paan2097 There are some GPIO errors that are showing up only the first time PX4 is started, this PR should fix that, https://github.com/PX4/Firmware/pull/7329
I can't tell about the PWM thing, have you found a fix for that?

On a related note, are you calibrating using QGroundControl? Are you using the voltage sensor? Are you applying rotations to the sensors?

nicolaerosia, 2017-06-05:
@paan2097 you need to disable the navio-rgb overlay in config.txt from the boot partition, I will create a PR for the User guide
```
#dtoverlay=navio-rgb
```

paan2097, 2017-06-06:
@nicolaerosia I've tried to get my plane up and running with both your master branch and navio-optimization branch from your fork of the PX4 github. With the master, the px4_fw.config file is missing the navio_adc start command, but it also doesn't recognize the command anyway. With this, I am no longer getting servo over extension but we also are not able to get passed the radio calibration because it doesn't see the rc input at all. Using the navio-optimization branch, we still get the servo over extension and same errors as previously that I posted here: http://discuss.px4.io/t/failure-of-px4-plane-code-running/3436.

We are simply trying to get the basic install/build/running PX4 Firmware on the NAVIO2 for planes. We followed documentation for RPi native build and run with the px4_fw.config file but we can't seem to avoid these problems. @LorenzMeier @nicolaerosia Have you proven support for running the latest (or any version) or PX4 firmware configured for a basic Plane (we are using Calypso), running on the NAVIO2?

nicolaerosia, 2017-06-06:
@paan2097 have you commented out the navio-rgb overlay as I said up?
navio_adc start was added to px4_fw.config in 
```
commit 32498009a36553a1f50601e205ebd609e894a8df
Author: Nicolae Rosia <nicolae.rosia@gmail.com>
Date:   Sun May 14 09:54:15 2017 +0300

    navio_adc: add driver for Navio2 ADC
    
    Signed-off-by: Nicolae Rosia <nicolae.rosia@gmail.com>
```

To be honest, I've never used native build.
navio-optimization was merged. Could you retry and post the error log?

paan2097, 2017-06-06:
I'm not sure where the error log is, but there is my PX4 output:
pi@navio:~/src/Firmware $ sudo ./build_posix_rpi_native/src/firmware/posix/px4 ./posix-configs/rpi/px4_fw.config 
commands file: ./posix-configs/rpi/px4_fw.config
creating new parameters file
creating new dataman file

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

ERROR [param] importing from 'rootfs/eeprom/parameters' failed (-1)
Command 'param' failed, returned 1
  MAV_BROADCAST: curr: 0 -> new: 1
  MAV_TYPE: curr: 2 -> new: 1
  BAT_CNT_V_VOLT: curr: -1.0000 -> new: 0.0010
  BAT_V_DIV: curr: -1.0000 -> new: 10.9176
  BAT_CNT_V_CURR: curr: -1.0000 -> new: 0.0010
  BAT_A_PER_V: curr: -1.0000 -> new: 15.3910
INFO  [dataman] Unkown restart, data manager file 'rootfs/fs/microsd/dataman' size is 11797680 bytes
ERROR [drivers__linux_gpio] export 4: write: Device or resource busy (16)
ERROR [navio_rgbled] red led: failed to export
ERROR [drivers__linux_gpio] unexport 4: write: Invalid argument (22)
ERROR [drivers__linux_gpio] unexport 27: write: Invalid argument (22)
ERROR [drivers__linux_gpio] unexport 6: write: Invalid argument (22)
Command 'navio_rgbled' failed, returned -1
WARN  [commander] Buzzer: px4_open fail

WARN  [commander] Buzzer init failed
.
INFO  [mavlink] mode: Normal, data rate: 1000000 B/s on udp port 14556 remote port 14550
Sleeping for 1 s; (1000000 us).
INFO  [logger] logger started (mode=all)
INFO  [logger] log root dir created: rootfs/fs/microsd/log
pxh> INFO  [mavlink] using network interface eth0, IP: 10.0.18.10
INFO  [mavlink] with broadcast IP: 10.0.18.255
INFO  [navio_sysfs_pwm_out] Loaded mixer from file ROMFS/px4fmu_common/mixers/AERT.main.mix
INFO  [mavlink] partner IP: 10.0.18.173
INFO  [lib__ecl] EKF aligned, (pressure height, IMU buf: 15, OBS buf: 14)
WARN  [commander] Not ready to fly: Sensors not set up correctly

Exiting...
�Shutting down
Restoring terminal

nicolaerosia, 2017-06-07:
As I said in the last two comments, are you sure you've commented navio-rgb overlay?
I already posted the instructions here, but here's a PR for that: https://github.com/PX4/Devguide/pull/177/commits/aa37e64b30545bf0eb1f3d684534f6b238e6fb37

paan2097, 2017-06-07:
@nicolaerosia Yeah we had commented out the navio-rgb overlay before testing yesterday and still had the same problems.

nicolaerosia, 2017-06-07:
@paan2097 please share the output of the following:
```
ls -l /sys/class/gpio
```

```
cat /boot/config.txt |grep navio
```

```
cat /sys/kernel/debug/gpio
```

paan2097, 2017-06-07:
@nicolaerosia I'm testing on the current PX4/Firmware master branch. Thanks for the help, I'm trying my best to get PX4 running on the NAVIO2 configured for planes. Unfortunately, I'm still running into the servo over extension problem although I ran through all the calibration steps in QGroundControl. All passed except it stated there was no airspeed sensor attached but there is one connected via I2C. Here is the output: 
pi@navio:~ $ ls -l /sys/class/gpio/
total 0
-rwxrwx--- 1 root gpio 4096 Jun  7 20:17 export
lrwxrwxrwx 1 root gpio    0 Jun  7 20:17 gpiochip0 -> ../../devices/platform/soc/3f200000.gpio/gpio/gpiochip0
lrwxrwxrwx 1 root gpio    0 Jun  7 20:17 gpiochip100 -> ../../devices/platform/soc/soc:virtgpio/gpio/gpiochip100
-rwxrwx--- 1 root gpio 4096 Jun  7 20:17 unexport
pi@navio:~ $ cat /boot/config.txt |grep navio
#dtoverlay=navio-rgb
#dtoverlay=navio-rgb
pi@navio:~ $ cat /sys/kernel/debug/gpio
cat: /sys/kernel/debug/gpio: Permission denied
pi@navio:~ $ sudo cat /sys/kernel/debug/gpio
GPIOs 0-53, platform/3f200000.gpio, pinctrl-bcm2835:
 gpio-4   (                    |?                   ) out hi
 gpio-6   (                    |?                   ) out hi
 gpio-27  (                    |?                   ) out hi

GPIOs 100-101, platform/soc:virtgpio, brcmvirt-gpio, can sleep:
 gpio-100 (                    |?                   ) out hi
pi@navio:~ $

nicolaerosia, 2017-06-08:
@paan2097 you're running as user, you need to run as root

paan2097, 2017-06-08:
@nicolaerosia I'm running px4 with the sudo command, do I need to actually login as root? We are using the NAVIO2 with the latest emlid image on a Raspberry Pi 3 Model B. My only connections are the AERT servos and motor, PPM converter to rc in, and a USB wifi dongle. I'm using the latest PX4/Firmware master branch

paan2097, 2017-07-19:
@nicolaerosia The NAVIO2 is up and running PX4 with the exception of the airspeed sensor. I'm using a digital airspeed sensor connected via I2C, but during calibration it states that there is no airspeed sensor plugged in. Advice please?

nicolaerosia, 2017-07-19:
what were your issues?
what airspeed sensor? which driver? Maybe the i2c path is not right.

paan2097, 2017-07-19:
I'm using the digital airspeed sensor recommended in sensor selection. My issue is that it is telling me there is no airspeed sensor attached when I'm trying to calibrate, it doesn't seem to recognize it is even connected. I'm not sure where to look for the correct driver, how can I check/set my I2C path? Thanks!

closed 2018-02-21 by PX4BuildBot