PX4/PX4-Autopilot #5701: Unable to start board / unable to establish connection with MP or QGC

arthursn138, 2016-10-19:
Hi guys,

First of all, yes, I've read all the related topics I could find that might have solved my problem. None of them did.

So after a "soft" crash with my quadcopter, damaging one of my ESCs, my board won't start.
Referring to a lot of threads, I assumed it was a bad SD issue. I tried 4 different cards, one of them brand new) and still got the same beeps and mixer error on the log.

Next step was to reset the board. As recommended on [this Ardupilot thread](http://discuss.ardupilot.org/t/how-to-reset-pixhawk/1584/3) I've loaded a plane firmware, then loaded a quadcopter firmware in order to clear all the flight code and parameters.

After that, I keep getting a solid red light on the main LED, solid green on FMU PWR and solid green on IO PWR. All other lights are off.
When I tried to connect with the Mission Planner (I got the last version on), I kept getting a **"No Heartbeat Packet Received"** error. As mentioned on [this DIYDrones thread](http://diydrones.com/forum/topics/mavlink-not-connecting-no-heartbeat-packet-received-error), I disconnected the GPS and now I simply can not connect through Mission Planner (**Timeout** error). When trying to connect to QGC (v3.0.0), I keep getting the **"make sure you have an SD Card inserted"** error.
This was also with different SD cards.

@LorenzMeier, I saw you were very active on the SD mixer errors, is this familiar to you?

I'm running out of Ideas guys.

=============================
**Last Log Files**

**_bootlog_**
ERROR [init] PX4IO update failed 
ERROR [init] PX4IO not found 
ERROR:[init] Could not load mixer: /etc/mixers/quad_x.main.mix 

**_msgs_2016_10_18_21_07_16_**
2016_10_18_21_07_16: GPS fix lost
2016_10_18_21_07_18: [lpe] baro init 752 m std 19 cm
2016_10_18_21_07_18: [lpe] z resume 

[bootlog.txt](https://github.com/PX4/Firmware/files/540170/bootlog.txt)
[msgs_2016_10_18_21_07_16.txt](https://github.com/PX4/Firmware/files/540171/msgs_2016_10_18_21_07_16.txt)

LorenzMeier, 2016-10-20:
I'm afraid this looks more like a real HW issue. You are offering really good information for debugging, so based on that I'm rather confident you left no step out trying to overcome a software issue. You might be able to get some support from the manufacturer of the board, but I wouldn't know how we can help you with broken hardware.

closed 2016-10-20 by LorenzMeier

arthursn138, 2016-10-20:
Thanks man. That's kinda reliving actually!

davids5, 2016-10-20:
Are there reset switches on the board? Is so ensure the none of them are jammed.

arthursn138, 2016-10-20:
@davids5 All the components look perfect, no jamming, obstructions or overpower signals

davids5, 2016-10-20:
Shucks! I was hoping for a stuck reset switch on the IO processor.