PX4/PX4-Autopilot #5702: PX4FMUV2 Crash on current master

sanderux, 2016-10-19:
Current master crashes my pixhawk fmuv2 after a few minutes with now clear reason.
Sometimes after uploading a mission, sometimes after mode switch, sometimes with no visible action.

The LED stays where it was solid at the time of crashing. Telemetry connection is lost.

The branch this happened on is https://github.com/PX4/Firmware/pull/5579
Locally rebased to https://github.com/PX4/Firmware/commit/78c52be08c1d00e5bacea846b6e979718fd7922b

Can anyone confirm this?
@LorenzMeier @julianoes

LorenzMeier, 2016-10-19:
Can you rebase to a commit before David's dataman changes here? https://github.com/PX4/Firmware/commits/master?after=eyiY6u7Kut%2BxEJv7mJa%2FHePrcpQrNjk%3D

LorenzMeier, 2016-10-19:
@bkueng Can you try to repro first thing in the morning?

sanderux, 2016-10-19:
I have no pixhawk available right now, trying to reproduce on dropix but so far not getting the same result.
I'll try bisecting tomorrow

sanderux, 2016-10-20:
@bkueng have you been able to reproduce this?

bkueng, 2016-10-21:
No unfortunately not. I tried running it with a VTOL airframe, no other HW attached, for several hours, but it did not crash. The usual suspects (low on RAM, stack sizes) looked ok to me.

sanderux, 2016-10-21:
I have also failed to reproduce on a dropix. I will try again on a pixhawk, perhaps sbus or radio telem related?

LorenzMeier, 2016-10-29:
What is your status with this?

sanderux, 2016-10-29:
I have not been able to reproduce this on a different pixhawk. Perhaps hardware failure

closed 2017-01-20 by sanderux