PX4/PX4-Autopilot #5703: Switch to user-space TSC clock access for SITL

LorenzMeier, 2016-10-19:
@bkueng We should switch to accessing the TSC clock directly from user space when we will eventually wrap all system calls to time.

Here is an excellent writeup why we want that:
http://oliveryang.net/2015/09/pitfalls-of-TSC-usage/#why-using-tsc

LorenzMeier, 2016-10-19:
@vilhjalmur89 This would allow us to use TSC even if the kernel is refusing to use it.

bkueng, 2016-10-20:
Did you read the conclusion?

> There is no reliable TSC sync mechanism for user application especially under a Virtualization environment

> Use it for debugging, but never use rdtsc in functional area.

vilhjalmur89, 2016-10-20:
I already tried forcing TSC which resulted in time moving backwards. I would think that all the TSC-calls would need to be made on the same core for this to work.

LorenzMeier, 2016-10-20:
Yep, but we still need a way to be able to work without TSC from the Kernel. If that means user space and locking to one core it might be what we have to do.

closed 2018-02-21 by PX4BuildBot