PX4/PX4-Autopilot #5710: PWM LED would benfit from better layering on io_timers

davids5, 2016-10-20:
see review notes https://github.com/PX4/Firmware/pull/5709

davids5, 2016-10-27:
@julianoes 

I think the 

```
led_pwm_servo_init(void)
{
	/* do basic timer initialisation first */
	for (unsigned i = 0; i < arraySize(led_pwm_timers); i++) {
		led_pwm_timer_init(i);
	}
```
to 
```
led_pwm_servo_init(void)
{
	/* do basic timer initialisation first */
	for (unsigned i = 0; i < arraySize(led_pwm_timers); i++) {
		io_timer_init_timer(i);
	}

```

may have broken other users of the driver.

io_timer_init_timer(i); would be used if a timer spans multiple pwms 
TimerN
[r]
[g]
[b]
[landing]
 
but breaks

TimerX
[r]
[g]
[b]

TimerY
[landing]
 
I think the fix is simple enough, we need to pass a pointer to the timer table in play.
I will have a better look when I am fresh

iZhangHui, 2016-10-29:
@davids5 @LorenzMeier 
 `led_pwm_servo_init(void)
 {
    /* do basic timer initialisation first */
    for (unsigned i = 0; i < arraySize(led_pwm_timers); i++) {
        io_timer_init_timer(i);
    }
`
This change indeed broken the **old revision** MindPX  RGB LED. But, for the latest hardware, we used I2C PWM  IC to drive RGB LED.

There will be  a lot of restriction when other users use PWM LED driver. Your fix proposal looks good. :thumbsup:

closed 2016-12-19 by LorenzMeier