PX4/PX4-Autopilot #5712: The release v1.4.4 has inconsistent commit with the branch v1.4.4

gaochangyu, 2016-10-21:
Hi, all
Today I found that the release v1.4.4 is on commit 60c2aa1 while the brach v1.4.4 has come to commit e73218c. I'm using the code at release v1.4.4 and because of the change in SITL, I don't want to migrate my code to work with higher version.
So, Can somebody explain why the brach v1.4.4 has moved so much forward?

dagar, 2016-10-21:
We don't usually have branches for each release, I believe it was an error. Please use the v1.4.4 tag instead. 60c2aa165feee1cb421a546670fca1344366839c

closed 2016-10-21 by dagar

kd0aij, 2016-10-25:
This confused me too; I was working on a local branch based off v1.4.4 and
suddenly the diff got very large. I don't see how I could accidentally
create a branch in PX4/Firmware though, since I was working in my own fork.