PX4/PX4-Autopilot #5713: Snapdragon flight IMU driver in ROS

nickhuan, 2016-10-21:
Hi all,

I just got a new Qualcomm snapdragon flight from Intrinsyc. I installed ROS and PX4/Firmware on the board. (not sure whether it's the correct way to do) May I know whether there is already some rosnodes publishing rostopic of IMU data (e.g. /sensor_msgs/imu messages) in this package? If yes, what is the simplest way to launch it in ROS. There are so many things in thus package, but I just want to read the IMU data in ROS.

Hope I found the right place to ask the question. Thanks.

julianoes, 2016-10-21:
Hi, it's probably better to ask questions in the forum here: http://discuss.px4.io/

However, to answer your question, I think it's easiest to use mavros: http://dev.px4.io/simulation-ros-interface.html. In order for the PX4 code on Snappy to talk to mavros, you have to start another instance of mavlink like it's done here for SITL https://github.com/PX4/Firmware/blob/78b0d1a01f2f3ebfd33d005d9da5908c72e4843b/posix-configs/SITL/init/lpe/iris#L65 in https://github.com/PX4/Firmware/blob/78b0d1a01f2f3ebfd33d005d9da5908c72e4843b/posix-configs/eagle/flight/mainapp.config.

nickhuan, 2016-10-26:
Hi Julian,

Thanks for the prompt response. I added one line "mavlink start -u 14557 -r 4000000 -m onboard -o 14540" in "Firmware/posix-configs/eagle/flight/mainapp.config". And I ran "roslaunch mavros px4.launch fcu_url:="udp://:14540@127.0.0.1:14557"" from the board. I could rostopic /mavros/imu/data is published but there is not messages running on it. Am I doing it corretly?

julianoes, 2016-10-26:
Seems correct to me. Is PX4 actually correctly run and do you see output in QGroundControl?

nickhuan, 2016-10-26:
I didn't use QGroundControl (cos my QGroundControl couldn't detect the device no matter I use USB-serial or WiFi). I compile PX4 in ROS catkin workspace natively on Snappy. I'm trying to run everything onboard. Is there another way to monitor PX4 besides QGroundControl?

julianoes, 2016-10-26:
> I didn't use QGroundControl (cos my QGroundControl couldn't detect the device no matter I use USB-serial or WiFi)
QGroundControl only works if you are connected over wifi, USB won't work.

> I compile PX4 in ROS catkin workspace natively on Snappy. I'm trying to run everything onboard. Is there another way to monitor PX4 besides QGroundControl?
This has not been done before and I'm pretty sure it won't work properly. The reason is that the cross compilation does not just comiple for the Linux (POSIX) side that you probably compiled for natively but there is also the part which compiles for the QURT on the DSP using the Hexagon toolchain and SDK. Most of the PX4 firmware actually runs on the DSP side, so if that part is missing, nothing will work, and neither QGroundControl nor mavros will be able to see much.

nickhuan, 2016-11-01:
Thanks Julian for the info. I'm trying to follow the tutorial http://dev.px4.io/tutorial-hello-sky.html to write a simple app. I assume I need to use "make eagle_default" to replace "make px4fmu-v4_default" for snappy. Instead of changing "Firmware/cmake/configs/posix_sitl_default.cmake", I assume we should change "Firmware/cmake/configs/posix_sdflight_default.cmake". So I added "examples/px4_simple_app" in "config_module_list" of posix_sdflight_default.cmake, but the px4_simple_app doesn't seem to compile. Even though this is a line printed "[ 88%] Built target examples__px4_simple_app", I could not find the executable in any folder. Neither does "make eagle_default upload" upload any px4_simple_app related stuff onto snappy. (px4 and mainapp.config are uploaded successfully) May I know why. Thanks a lot.

julianoes, 2016-11-01:
No "executable" will be made. But you should be able to start it in `pxh>` manually or add it to [mainapp.config](https://github.com/PX4/Firmware/blob/master/posix-configs/eagle/flight/mainapp.config).

If you build it for qurt, you would have to add it to the [px4.config](https://github.com/PX4/Firmware/blob/master/posix-configs/eagle/flight/px4.config).

nickhuan, 2016-11-11:
Hi Julian, I successfully get the IMU data in rostopic by running mavros as you suggested. Thanks a lot. 
I think we can close this issue now. :) :+1: 

But I got another problem while running the px4 without using a SD card:  no space left on device and cannot delete the log files generated by XP4. I've asked it in the forum http://discuss.px4.io/t/snapdragon-flight-delete-log-files-generated-by-px4-in-read-only-file-system/1721. Maybe you could help me on that.

closed 2016-11-11 by nickhuan