PX4/PX4-Autopilot #5714: Problem with vision_pose/pose and LPE

NCharabaruk, 2016-10-21:
Hello,

I am using a Pixhawk with PX4 firmware on a hexacopter for indoor flight. The drone is being controlled by ROS. I am using laser scan matching to find the xy and yaw position, sonar for z height, and an IMU for pitch and roll. This data is being merged into a single message and published to the /mavros/vision_pose/pose topic. When looking at /mavros/local_position/pose however, the z value is way off. Looking around online, apparently this is because the PX4 firmware is still trying to merge the vision data with its own information. Is there a way to tell the PX4 to only use the data I am sending it? I've looked online and have seen solutions for this when using INAV, but after switching estimators several of the variables that need changing don't show up on the parameter list in QGC. Any help with getting this working would be greatly appreciated.

art-mx, 2016-10-26:
Have you considered decreasing vision standard deviation parameters (LPE_VIS_XY and LPE_VIS_Z) for LPE?

NCharabaruk, 2016-10-26:
I figured it out. There is a parameter to turn off IMU integration for pose estimates. The param was removed but there is still a boolean in the code for it. I turned off IMU integration and now it only looks at what I want it to.

closed 2016-10-26 by NCharabaruk

hmchung, 2017-01-11:
Hi NCharabaruk, can you please kindly explain which part of the code did you change? Apparently, I have also just got into the same problem. Thank you in advance.

SubMishMar, 2018-02-24:
@NCharabaruk could you please tell which parameter you are talking about?