PX4/PX4-Autopilot #5725: Pixracer: Not ready to fly without further info

julianoes, 2016-10-22:
I'm getting a `Not ready to fly` on my Pixracer and can't arm. Unfortunately, I get no other message about why the checks fail. This airframe has been flying just fine previously, so it seems like a regression.

I need to find the root cause of the failed check, as well as why it is not reported.

kd0aij, 2016-10-22:
I find that "commander check" on the console almost always identifies the problem.

julianoes, 2016-10-22:
@kd0aij thanks that's helpful:
```
nsh> commander check
WARN  [commander_tests] PREFLIGHT FAIL: MAG #1 UNCALIBRATED
WARN  [commander_tests] PREFLIGHT FAIL: ACCEL #0 UNCALIBRATED
WARN  [commander_tests] Warning: Primary accelerometer not found
WARN  [commander_tests] PREFLIGHT FAIL: GYRO #0 UNCALIBRATED
WARN  [commander_tests] Warning: Primary gyro not found
WARN  [commander] Preflight check: FAILED
WARN  [commander_tests] PREFLIGHT FAIL: MAG #1 UNCALIBRATED
WARN  [commander_tests] PREFLIGHT FAIL: ACCEL #0 UNCALIBRATED
WARN  [commander_tests] Warning: Primary accelerometer not found
WARN  [commander_tests] PREFLIGHT FAIL: GYRO #0 UNCALIBRATED
WARN  [commander_tests] Warning: Primary gyro not found
WARN  [commander] Prearm check: FAILED
```

The result is strange though because QGC shows that the sensors are "green" and also the params are set as far as I can tell.

julianoes, 2016-10-22:
It looks like all the device ids have changed since I calibrated last beginning of August.

julianoes, 2016-10-22:
Well, I just recalibrated and it's all good again. I'll just drop this.

mhkabir, 2016-10-22:
@julianoes In this case, the Firmware should still report invalid calibrations and force a recalibration with a report to the user. No way to know that something's going wrong otherwise, since there is no visible error message in QGC.

julianoes, 2016-10-22:
The messages did arrive over the 3DR radio but not over USB. They must have got lost somewhere, you're right I should check where.

closed 2017-04-30 by LorenzMeier