PX4/PX4-Autopilot #5726: pre-commit hook for "check_format" failed a new mixer (text) file

kd0aij, 2016-10-22:
and it's a very bad idea to run astyle on a mixer definition file.
Ask me how I know :)

davids5, 2016-10-22:
All this new magic....

Is it astyle alone on the commit hooks? 

BTW: Once I am convinced it should not muck with a file I just use 

```
git commit --no-verify
```

kd0aij, 2016-10-22:
I persist in using Eclipse's git integration, and that just makes for more magic.
BTW @dagar 's Eclipse project works great, but the one generated by cmake for Eclipse-Ninja is now useless for anything but posix builds. It was OK too up to a few months ago, but I was unable to fathom the changes which broke it.

dagar, 2016-10-22:
Can you walk me through exactly what happened, where the file was, etc? `make check_format` is *supposed* to be a warning only with no files touched.

@davids5 Yes, it's just astyle, but we limit it to the src/ folder and still exclude some things.https://github.com/PX4/Firmware/blob/master/Tools/files_to_check_code_style.sh

davids5, 2016-10-22:
@dagar 

It does not look like astyle - with blank EOF warnings on initial commits some files with vim tags.

Next time I get one I will capture it here.

dagar, 2016-10-22:
Oh, right it also does `git diff --check` 

--check
           Warn if changes introduce whitespace errors. What are considered whitespace errors is controlled by core.whitespace configuration. By default, trailing whitespaces (including lines that
           solely consist of whitespaces) and a space character that is immediately followed by a tab character inside the initial indent of the line are considered whitespace errors. Exits with
           non-zero status if problems are found. Not compatible with --exit-code.

davids5, 2016-10-22:
@dagar 

Don't get me wrong I love ALL the new magic - Thank you for it! 

It is just sometimes, usually at the end of a long day, it becomes a rabbit hole I have to go down and in the true fashion of a "rabbit hole"  I never know - how deep it is or where it will come out. But you are super good at helping to teach this old dog new tricks. 

ug...did I really just use that many metaphors :)

dagar, 2016-11-08:
@kd0aij I haven't been able to reproduce this. The check *should* only be certain file extensions and files in src/. Could you explain?

kd0aij, 2016-11-08:
I hit this again yesterday trying to do a commit using Eclipse. I had added new sourcefiles (see https://github.com/PX4/Firmware/pull/5821). Eclipse displayed a large popup with what looked like a binary file displayed as text when I attempted to commit using the git staging tab. Sorry I didn't think to take a screenshot, but the banner must have said the commit hook failed. I shifted to a bash shell, deleted the pre-commit hook and committed manually. It may be that new files are not handled properly by the format checker.

davids5, 2016-11-08:
@kd0aij 

It look like the hooks run a diff test for EOL, and EOL on an  empty line at the end of a file when a new file first committed.  

So it should be simple to reproduce the error on a test branch.

dagar, 2016-11-08:
Oh, okay so it's just the pre-commit hook checking whitespace errors on all
files here - https://github.com/PX4/Firmware/blob/master/Tools/
pre-commit#L50
It should only be running astyle (without modifying) on appropriate files
in src/.

dagar, 2016-11-08:
I understand it was a bit confusing to have it pop up in eclipse, but is `git diff --check` ever wrong? It's supposed to only error if you introduce whitespace errors.

On the command line you can always do `git commit --no-verify`.

kd0aij, 2016-11-08:
I didn't now about `git diff --check` but it doesn't complain now.

dagar, 2016-11-08:
There would have to be a current diff (something not yet committed or staged).

kd0aij, 2016-11-08:
I had a bunch of unstaged changes when I ran the diff.

dagar, 2016-11-08:
Right, but previously if you had stumbled across a particular problematic case that change is no longer around to test. 

Do you still think there's a problem here?

kd0aij, 2016-11-08:
I can't reproduce it at the command line. No complaints from `git diff --check` or `make check_format`

dagar, 2016-11-08:
Alright, ping me on gitter or skype if you hit a problem again and we'll take a closer look.

closed 2016-11-08 by dagar