PX4/PX4-Autopilot #5728: mavlink_shell.py doesn't work over 3DR radio

julianoes, 2016-10-22:
I couldn't get the mavlink shell to work over a 3DR radio:

```
Tools/mavlink_shell.py
Using port /dev/serial/by-id/usb-FTDI_FT231X_USB_UART_DN007AAX-if00-port0
Connecting to MAVLINK...
Traceback (most recent call last):
  File "Tools/mavlink_shell.py", line 205, in <module>
    main()
  File "Tools/mavlink_shell.py", line 118, in main
    mav_serialport = MavlinkSerialPort(args.port, args.baudrate, devnum=10)
  File "Tools/mavlink_shell.py", line 34, in __init__
    self.mav.wait_heartbeat()
  File "/home/julianoes/.local/lib/python2.7/site-packages/pymavlink/mavutil.py", line 382, in wait_heartbeat
    return self.recv_match(type='HEARTBEAT', blocking=blocking)
  File "/home/julianoes/.local/lib/python2.7/site-packages/pymavlink/mavutil.py", line 343, in recv_match
    m = self.recv_msg()
  File "/home/julianoes/.local/lib/python2.7/site-packages/pymavlink/mavutil.py", line 306, in recv_msg
    s = self.recv(n)
  File "/home/julianoes/.local/lib/python2.7/site-packages/pymavlink/mavutil.py", line 802, in recv
    ret = self.port.read(n)
  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 495, in read
    raise SerialException('device reports readiness to read but returned no data (device disconnected or multiple access on port?)')
serial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
```

tumbili, 2016-10-22:
@julianoes Works for me, this is how I use it:
```
python mavlink_shell.py /dev/ttyUSBx --baud 57600
```

julianoes, 2016-10-22:
Oh thanks! You're right. Does it not default to 57600 for anything `USBx`?

julianoes, 2016-10-22:
It doesn't but I'm trying to change it :smile:.

PR #5729 (2016-10-23): mavlink_shell.py: default to 57600 baudrate
I think it makes more sense to use 57600 by default because it works straight out of the box with 3DR radios.

@bkueng objections?

Fixes #5728.
closed 2016-10-23 by LorenzMeier