PX4/PX4-Autopilot #5730: Support NED waypoints within minimal maintenance

jgoppert, 2016-10-22:
I think we can support local ned waypoints  with minimal maintenance by using the local reference point to transform all of the NED waypoints to lat, lon, alt waypoints using the functions in geo.c right when it comes into the system via mavlink_receiver. I think that internally it is best to only have one frame for waypoints (lat, lon, alt).

Let me know your thoughts.

mhkabir, 2016-10-22:
In the longer term, I think the other way around would make more sense. i.e transform the lat/lon/alt to NED when they come in, and navigate in the local frame. More work short-term, and would need a bit of testing to check for bugs, but it would be better.
In an GPS-denied world, a local estimate is guaranteed, but not a global one unless we have atleast locked onto GPS once. I bring this up because I'm doing indoor-outdoor transitions and switching from local to global frames, but this is all done offboard currently. Bringing in a generic way to do it in Firmware would be great.

jgoppert, 2016-10-23:
@mhkabir I agree, right now I'm running into the issue that the waypoint resolution is only 0.6m (1e-7*6371000) so I can't create a precise indoor mission with the current waypoints.

jgoppert, 2016-10-23:
@LorenzMeier thoughts?

LorenzMeier, 2016-10-23:
You need to use MISSION_ITEM_INT (which PX4 already supports).

jgoppert, 2016-10-23:
@LorenzMeier I'm using mavros, do we need to  modify mavros to use MISSION_ITEM_INT?

jgoppert, 2016-10-23:
Actually, in QGC they show correctly after load, then I send them to the vehicle and then load them from the vehicle and they have shifted. So not mavros specific.

jgoppert, 2016-10-23:
I created a QGC issue here documenting the exact waypoint changes: https://github.com/mavlink/qgroundcontrol/issues/4172

dagar, 2017-04-11:
Would it make sense to have the position controllers only use local position and make navigator responsible for mapping global -> local?

closed 2018-02-21 by PX4BuildBot