PX4/PX4-Autopilot #5732: Bebop : support sonar over IIO

mhkabir, 2016-10-22:
We should support the Bebop's sonar unit. A bit convoluted, because we need to actually trigger the sonar pings via SPI and then read back via the ADC.

I already have the software part of optical flow sorted more or less, but need the AGL estimate to scale the readings.

@eyeam3 Are you working on this by any chance? Would you mind syncing over email/Skype on what's planned/ongoing for Bebop?

eyeam3, 2016-10-24:
Yes, I am working on the height sensor. In fact, it is almost finished (80%), but needs a few optimizations and further testing. Unfortunately, this has to wait until next week as I'm really busy with my master's thesis at the moment.

I also created a [GitHub-Project](https://github.com/PX4/Firmware/projects/8) to publicly track ongoing and planned tasks. And to sync, my email is [schaeuble.michael@gmail.com](schaeuble.michael@gmail.com).

mhkabir, 2016-10-24:
@eyeam3 Thanks for the info. Let me know if the sonar stuff is on a branch somewhere. I have time this week to fix things and test it. Good luck with the thesis!

The project should be useful. Great!

@LorenzMeier I have made progress on the camera part of things, and am also in talks with Parrot on how to get the software stabilized forward video pipeline up. However, I'm unsure as to how to structure it architecturally : 
* How should e.g the camera process and optical flow process communicate? (in terms of image data transfer). I don't know if it is wise to pack that sort of data into uORB, or even possible to?
* Should something like the camera driver even be a part of the PX4 codebase? The driver needs a lot of low-level interaction (setting up sensor registers, etc.) plus high-level interaction (images)

LorenzMeier, 2016-10-24:
UORB is the most resource efficient system you can think of. It's perfectly suitable to pipe the images to a couple of threads. So please go ahead and just structure them like the rest of the processes. We can then do one flow module which works on bebop and Snapdragon

mhkabir, 2016-10-24:
Awesome! Thanks :)

LorenzMeier, 2016-12-26:
PR is in.

closed 2016-12-26 by LorenzMeier