PX4/PX4-Autopilot #5733: sdlog2_dump.py CSV generated file missing a lot of data.

doc2142, 2016-10-22:
When I run sdlog2_dump.py on my data file and compare the plots I generated from the CSV data to mission planner plots I notice that there is a lot of mission data. On mission planner data seem to have frequency of 50Hz while the CSV file that was generated has a max of 4Hz data. Any idea how to fix this?

LorenzMeier, 2016-10-22:
This file does not support the mission planner format. It is designed for PX4 (if you want to use PX4, use QGroundControl).

closed 2016-10-22 by LorenzMeier

doc2142, 2016-10-22:
But the bin file I have is from a pixhawk. I got it from the sd card that was recorded by pixhawk.

LorenzMeier, 2016-10-22:
PX4 doesn't record .bin files. You are not running PX4 and hence this is not a PX4 issue.

doc2142, 2016-10-22:
I am sorry for being a complete newbie on this I am totally new on this stuff. Are you saying the pixhawk can run mission planner or px4. Since I am getting bin files I am running mission planner?

kd0aij, 2016-10-22:
What instructions did you follow to flash firmware onto your Pixhawk?

doc2142, 2016-10-22:
I didn't, I am using a pixhawk that was preflashed.

kd0aij, 2016-10-22:
It sounds like you must have some version of ArduPilot on it. If you don't want to reflash it with PX4, you can ask on the ArduPilot forums about analyzing the logfile. PX4 logs have the extension ".px4log" and are placed in the SD card in a directory named "log ".
For PX4 documentation look at px4.io, dev.px4.io and discuss.px4.io