PX4/PX4-Autopilot #5739: px4 flies to HOME_ALT+HOME_ALT and lands earlier than expected

marketingandmoney, 2016-10-23:
even thought all the mission were set to 50m ,px4 flew to ~2x50m which is also close to HOME_ALT+HOME_ALT (112m+112m) ,then decelerates to mission alt by setting throttle to 0 with sharp pitch,and lands 50m earlier than expected

dagar, 2016-10-23:
It looks like flying to twice home altitude is just a coincidence.

[image]

TECS is in underspeed mode because the airspeed is < 90% of FW_AIRSPD_MIN (15 m/s), but why is TECS pitch responding so slowly?

dagar, 2016-10-23:
@AndreasAntener @tumbili - correct me if I'm wrong here, but it looks like it's basically stuck in TECS underspeed mode. Full throttle is forced causing the climb, but pitch changes slowly because the airspeed error is minimal.

LorenzMeier, 2016-10-23:
@dagar I don't see a bug here. The pitch tuning is bad (see the constant pitch rate offset) and consequently it doesn't pitch down as it should (see here: http://logs.uaventure.com/view/xqicBTPWS5LYtM4mXdtxSA#A_PitchRate_PLOT).

I also think that min airspeed and trim airspeed are pretty high given the plane can't reach them at full throttle. To fix this, two things are required:

  * Retuning the pitch control loop until it tracks pitch correctly
  * Reducing airspeed min and trim or bolting a better engine on the system (or leave as is, but make sure the system is perfectly tuned, as it has not a lot of margin)

closed 2016-10-23 by LorenzMeier