PX4/PX4-Autopilot #5740: How to interface with gazebo_SITL using MAVROS?

zhahaoyu, 2016-10-24:
So I found the documentation really hard to understand. I followed instructions from these two sources, 
[http://dev.px4.io/simulation-ros-interface.html](url) and [http://dev.px4.io/ros-mavros-offboard.html](url). However, the quadrotor never rises, and the console constantly outputs error message.

I first ran the gazebo_sitl
```
cd <Firmware_clone>
make posix_sitl_default
source ~/catkin_ws/devel/setup.bash
source Tools/setup_gazebo.bash $(pwd) build_posix_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)/Tools/sitl_gazebo
roslaunch px4 mavros_posix_sitl.launch
```
 and the output is
```
SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: ROLL_180
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fcu_url: udp://:14540@loca...
 * /mavros/gcs_url: 
 * /mavros/global_position/frame_id: fcu
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: fcu_utm
 * /mavros/global_position/tf/frame_id: local_origin
 * /mavros/global_position/tf/send: False
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.000349065850399
 * /mavros/imu/frame_id: fcu
 * /mavros/imu/linear_acceleration_stdev: 0.0003
 * /mavros/imu/magnetic_stdev: 0.0
 * /mavros/imu/orientation_stdev: 1.0
 * /mavros/local_position/frame_id: fcu
 * /mavros/local_position/tf/child_frame_id: fcu
 * /mavros/local_position/tf/frame_id: local_origin
 * /mavros/local_position/tf/send: False
 * /mavros/local_position/tf/send_fcu: False
 * /mavros/mission/pull_after_gcs: True
 * /mavros/mocap/use_pose: True
 * /mavros/mocap/use_tf: False
[... 217 lines cut ...]
[ERROR] [1477288950.918057774, 197.768000000]: MODE: Unsupported FCU
[ERROR] [1477288955.971626596, 202.818000000]: MODE: Unsupported FCU
[ERROR] [1477288961.025242546, 207.852000000]: MODE: Unsupported FCU
[ERROR] [1477288966.078283087, 212.918000000]: MODE: Unsupported FCU
[ERROR] [1477288971.133451537, 217.968000000]: MODE: Unsupported FCU
[ERROR] [1477288976.194463373, 223.018000000]: MODE: Unsupported FCU
[ERROR] [1477288981.248755791, 228.070000000]: MODE: Unsupported FCU
[ERROR] [1477288986.301387819, 233.118000000]: MODE: Unsupported FCU
[ERROR] [1477288991.354523686, 238.168000000]: MODE: Unsupported FCU
[ERROR] [1477288996.407573064, 243.218000000]: MODE: Unsupported FCU
[ERROR] [1477289001.460926055, 248.270000000]: MODE: Unsupported FCU
[ERROR] [1477289006.513461784, 253.318000000]: MODE: Unsupported FCU
[ERROR] [1477289011.566814832, 258.368000000]: MODE: Unsupported FCU
[ERROR] [1477289016.620408010, 263.418000000]: MODE: Unsupported FCU
[ERROR] [1477289021.675254041, 268.468000000]: MODE: Unsupported FCU
[ERROR] [1477289026.729104645, 273.518000000]: MODE: Unsupported FCU
[ERROR] [1477289031.783777507, 278.568000000]: MODE: Unsupported FCU
[ERROR] [1477289036.838509354, 283.618000000]: MODE: Unsupported FCU
[ERROR] [1477289041.892448828, 288.668000000]: MODE: Unsupported FCU
[ERROR] [1477289046.947357703, 293.718000000]: MODE: Unsupported FCU
[ERROR] [1477289052.000189320, 298.768000000]: MODE: Unsupported FCU
[ERROR] [1477289057.054740632, 303.818000000]: MODE: Unsupported FCU
[ERROR] [1477289062.108534809, 308.868000000]: MODE: Unsupported FCU
[ERROR] [1477289067.161737856, 313.918000000]: MODE: Unsupported FCU
```

L4ncelot, 2016-10-24:
Hi, following the same documents as @zhahaoyu I've also encountered the same problem. There is also error while making project using `make posix_sitl_default gazebo`. I don't know if this error has to do with something but I've posted it just in case.

**Error message:**
'Built target platforms__posix__drivers__tonealrmsim
[ 89%] Built target platforms__posix__px4_layer
[ 89%] Built target systemcmds__esc_calib
[ 93%] Built target platforms__posix__work_queue
[ 93%] Built target systemcmds__param
[ 94%] Built target systemcmds__mixer
[ 94%] Built target systemcmds__perf
[ 94%] Built target systemcmds__reboot
[ 95%] Built target systemcmds__sd_bench
[ 96%] Built target systemcmds__topic_listener
[ 96%] [ 96%] Built target modules__fw_pos_control_l1
Built target modules__fw_att_control
[ 97%] Built target modules__logger
[ 97%] Built target px4
/home/l4ncelot/src/Firmware/Tools/sitl_gazebo/src/gazebo_gimbal_controller_plugin.cpp: In member function ‘void gazebo::GimbalControllerPlugin::OnUpdate()’:
/home/l4ncelot/src/Firmware/Tools/sitl_gazebo/src/gazebo_gimbal_controller_plugin.cpp:312:43: error: no matching function for call to ‘gazebo::GimbalControllerPlugin::QtoZXY(ignition::math::Vector3d&)’
       this->QtoZXY(currentAngleYPRVariable));
                                           ^
/home/l4ncelot/src/Firmware/Tools/sitl_gazebo/src/gazebo_gimbal_controller_plugin.cpp:312:43: note: candidate is:
/home/l4ncelot/src/Firmware/Tools/sitl_gazebo/src/gazebo_gimbal_controller_plugin.cpp:251:26: note: ignition::math::Vector3d gazebo::GimbalControllerPlugin::QtoZXY(const Quaterniond&)
 ignition::math::Vector3d GimbalControllerPlugin::QtoZXY(
                          ^
/home/l4ncelot/src/Firmware/Tools/sitl_gazebo/src/gazebo_gimbal_controller_plugin.cpp:251:26: note:   no known conversion for argument 1 from ‘ignition::math::Vector3d {aka ignition::math::Vector3<double>}’ to ‘const Quaterniond& {aka const ignition::math::Quaternion<double>&}’
make[7]: *** [CMakeFiles/rotors_gazebo_gimbal_controller_plugin.dir/src/gazebo_gimbal_controller_plugin.cpp.o] Error 1
make[6]: *** [CMakeFiles/rotors_gazebo_gimbal_controller_plugin.dir/all] Error 2
make[5]: *** [all] Error 2
make[4]: *** [external/Stamp/sitl_gazebo/sitl_gazebo-build] Error 2
make[3]: *** [src/firmware/posix/CMakeFiles/sitl_gazebo.dir/all] Error 2
make[2]: *** [src/firmware/posix/CMakeFiles/gazebo.dir/rule] Error 2
make[1]: *** [gazebo] Error 2
make: *** [posix_sitl_default] Error 2'

zhahaoyu, 2016-10-26:
Hi @L4ncelot, I figured it out. Instead of running the ROS wrapper, I ran the regular posix_sitl_default and opened up another Mavros node. This allows me to create my own node and communicate with the mavros node.

closed 2016-10-26 by zhahaoyu