PX4/PX4-Autopilot #5752: Baro jump with Pixracer

CarlOlsson, 2016-10-25:
Code: cbbee30e483f89350a9207a5684803abd8312fd3
[Logfile](http://logs.uaventure.com/view/Nsdwvw6YVTa2bFPYnWb3AD)

Pixracer with case and mandatory baro foam
[image]

I am not sure if this kind of noise is expected and will be filtered out in a later stage, or if it is something wrong.

julianoes, 2016-10-25:
@CarlOlsson thanks for reporting, this seems like a bug somewhere! Does it happen frequently? Is it easy reproduce? Do you see it on multiple Pixracers, or just one piece of hardware?

CarlOlsson, 2016-10-25:
The airframe is pretty new so I don't know yet if it happens frequently but it happened in two flights out of three. Here is the [logfile](http://logs.uaventure.com/view/9oDo8aE4KuJcZebrZqhcRH) from the second flight. As seen below the baro temp also has a jump at the same time in both flights:
[image]

No I haven't seen it on any other pixracer, but this one is the only one we fly with this code

julianoes, 2016-10-26:
At the same time, that's interesting. I'll try if I can repdouce it.

julianoes, 2016-10-26:
I did a 10min log and I can't see a spike. Can you maybe gather more data in what setups or circumstances this happens?

davids5, 2016-10-26:
@julianoes - I think there is a mutex on the baro because it is on the same bus as the FRAM. I wonder if param operations are the cause.

LorenzMeier, 2016-10-26:
Yeah, it could throw off the baro timing. It could lead to an inversion of temp and pressure data.

julianoes, 2016-10-26:
I was just wondering that as well. And I did some `param set` and `param save` and I did get one spike out of the 4 param operations, so that might be a possibility.

julianoes, 2016-10-26:
So it does happen right when doing a `param set` but it doesn't have to be the first param set and I can only trigger it once.

davids5, 2016-10-26:
Any FRAM write should lock the bus. A save operation on a changed value. But that will be asynchronous to the BARO access. So as the frequency of the write goes up. You should see more locks.

CarlOlsson, 2016-11-13:
[image]
[image]
On friday this happened 2 out of 16 flights on another pixracer

dagar, 2017-04-12:
Was this resolved?

closed 2018-02-21 by PX4BuildBot