PX4/PX4-Autopilot #5753: Error using "make posix gazebo"

L4ncelot, 2016-10-25:
Hi,

I've been trying to run gazebo with quadcopter model with MAVROS. I've been following documentation on [http://dev.px4.io/](url) but after trying to make target using `make posix gazebo` or `make posix_sitl_default gazebo` compiler gave me the following error message.

```
[ 40%] Building CXX object src/modules/uORB/CMakeFiles/modules__uORB.dir/uORBMain.cpp.o
[ 41%] Building CXX object src/modules/uORB/CMakeFiles/modules__uORB.dir/Publication.cpp.o
[ 41%] Building CXX object src/modules/uORB/CMakeFiles/modules__uORB.dir/Subscription.cpp.o
[ 41%] Building CXX object src/modules/uORB/CMakeFiles/modules__uORB.dir/uORBManager.cpp.o
[ 41%] Building CXX object src/modules/uORB/CMakeFiles/modules__uORB.dir/uORBDevices.cpp.o
/home/l4ncelot/src/src/Firmware/Tools/sitl_gazebo/src/gazebo_gimbal_controller_plugin.cpp: In member function ‘void gazebo::GimbalControllerPlugin::OnUpdate()’:
/home/l4ncelot/src/src/Firmware/Tools/sitl_gazebo/src/gazebo_gimbal_controller_plugin.cpp:312:43: error: no matching function for call to ‘gazebo::GimbalControllerPlugin::QtoZXY(ignition::math::Vector3d&)’
       this->QtoZXY(currentAngleYPRVariable));
                                           ^
/home/l4ncelot/src/src/Firmware/Tools/sitl_gazebo/src/gazebo_gimbal_controller_plugin.cpp:312:43: note: candidate is:
/home/l4ncelot/src/src/Firmware/Tools/sitl_gazebo/src/gazebo_gimbal_controller_plugin.cpp:251:26: note: ignition::math::Vector3d gazebo::GimbalControllerPlugin::QtoZXY(const Quaterniond&)
 ignition::math::Vector3d GimbalControllerPlugin::QtoZXY(
                          ^
/home/l4ncelot/src/src/Firmware/Tools/sitl_gazebo/src/gazebo_gimbal_controller_plugin.cpp:251:26: note:   no known conversion for argument 1 from ‘ignition::math::Vector3d {aka ignition::math::Vector3<double>}’ to ‘const Quaterniond& {aka const ignition::math::Quaternion<double>&}’
Linking CXX static library libmodules__uORB.a
[ 41%] Built target modules__uORB
Scanning dependencies of target modules__mavlink
[ 41%] Building C object src/modules/mavlink/CMakeFiles/modules__mavlink.dir/mavlink.c.o
[ 42%] Building CXX object src/modules/mavlink/CMakeFiles/modules__mavlink.dir/mavlink_main.cpp.o
make[7]: *** [CMakeFiles/rotors_gazebo_gimbal_controller_plugin.dir/src/gazebo_gimbal_controller_plugin.cpp.o] Error 1
make[6]: *** [CMakeFiles/rotors_gazebo_gimbal_controller_plugin.dir/all] Error 2
make[6]: *** Waiting for unfinished jobs....
```

I was successfully able to run simulation test with `jmavsim` but not with `Gazebo` and `ROS`. 

Thank you for any advice.

LorenzMeier, 2016-10-25:
What is the output of ```gazebo --version```? And can you run the build again and share the output?

L4ncelot, 2016-10-26:
I have gazebo 7.4 and currently working on Ubuntu 14.04.

```
Gazebo multi-robot simulator, version 7.4.0
Copyright (C) 2012-2016 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org
```
Here is the whole output when running `make posix_sitl_default gazebo` command.

```
-- posix-sitl-default
-- CMAKE_INSTALL_PREFIX: /usr
-- cmake module path: /home/l4ncelot/src/src/Firmware/cmake
-- The CXX compiler identification is GNU 4.8.4
-- The C compiler identification is GNU 4.8.4
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- catkin DISABLED
-- Found PythonInterp: /usr/bin/python2.7 (found suitable version "2.7.6", minimum required is "2.7") 
-- GIT_TAG = v1.5.0-490-gd37a927
-- Configuring done
-- Generating done
-- Build files have been written to: /home/l4ncelot/src/src/Firmware/build_posix_sitl_default
PX4 CONFIG: /home/l4ncelot/src/src/Firmware/build_posix_sitl_default
Scanning dependencies of target prebuild_targets
Scanning dependencies of target git_genmsg
Scanning dependencies of target sitl_gazebo
Scanning dependencies of target git_gencpp
[  0%] [  1%] Built target prebuild_targets
[  1%] [  1%] Generating git_init_Tools_genmsg.stamp
Generating git_init_Tools_gencpp.stamp
Scanning dependencies of target git_mavlink
[  1%] Creating directories for 'sitl_gazebo'
[  2%] [  2%] Built target git_genmsg
Built target git_gencpp
Generating git_init_mavlink_include_mavlink_v1.0.stamp
Scanning dependencies of target git_uavcan
[  2%] Scanning dependencies of target git_matrix
Built target git_mavlink
[  2%] Scanning dependencies of target git_jmavsim
[  2%] Generating git_init_src_modules_uavcan_libuavcan.stamp
[  2%] Generating git_init_src_lib_matrix.stamp
[  2%] Built target git_uavcan
Generating git_init_Tools_jMAVSim.stamp
[  2%] Built target git_matrix
[  2%] Built target git_jmavsim
Scanning dependencies of target df_driver_framework
[  2%] Scanning dependencies of target git_driverframework
No download step for 'sitl_gazebo'
Scanning dependencies of target git_ecl
[  2%] [  3%] [  3%] Generating git_init_src_lib_DriverFramework.stamp
Building CXX object src/lib/DriverFramework/framework/src/CMakeFiles/df_driver_framework.dir/DriverFramework.cpp.o
Generating git_init_src_lib_ecl.stamp
[  4%] [  4%] Built target git_driverframework
[  4%] No patch step for 'sitl_gazebo'
Built target git_ecl
Scanning dependencies of target xml_gen
[... 367 lines cut ...]
[ 47%] Building CXX object src/modules/mavlink/CMakeFiles/modules__mavlink.dir/mavlink_messages.cpp.o
[ 47%] Building CXX object src/modules/mavlink/CMakeFiles/modules__mavlink.dir/mavlink_stream.cpp.o
[ 47%] Building CXX object src/modules/mavlink/CMakeFiles/modules__mavlink.dir/mavlink_rate_limiter.cpp.o
Linking CXX static library libmodules__navigator.a
[ 47%] [ 47%] Building CXX object src/modules/mavlink/CMakeFiles/modules__mavlink.dir/mavlink_receiver.cpp.o
Built target modules__navigator
[ 47%] Building CXX object src/lib/ecl/CMakeFiles/lib__ecl.dir/EKF/vel_pos_fusion.cpp.o
[ 47%] Building CXX object src/lib/ecl/CMakeFiles/lib__ecl.dir/EKF/mag_fusion.cpp.o
[ 47%] Building CXX object src/lib/ecl/CMakeFiles/lib__ecl.dir/EKF/gps_checks.cpp.o
[ 47%] Building CXX object src/lib/ecl/CMakeFiles/lib__ecl.dir/EKF/airspeed_fusion.cpp.o
[ 48%] Building CXX object src/modules/mavlink/CMakeFiles/modules__mavlink.dir/mavlink_ftp.cpp.o
[ 49%] Building CXX object src/lib/ecl/CMakeFiles/lib__ecl.dir/EKF/optflow_fusion.cpp.o
[ 49%] Building CXX object src/modules/mavlink/CMakeFiles/modules__mavlink.dir/mavlink_log_handler.cpp.o
[ 49%] Building CXX object src/modules/mavlink/CMakeFiles/modules__mavlink.dir/mavlink_shell.cpp.o
[ 49%] Building CXX object src/modules/mavlink/CMakeFiles/modules__mavlink.dir/mavlink_ulog.cpp.o
[ 49%] Building CXX object src/lib/ecl/CMakeFiles/lib__ecl.dir/EKF/control.cpp.o
[ 49%] Building CXX object src/lib/ecl/CMakeFiles/lib__ecl.dir/EKF/terrain_estimator.cpp.o
Linking CXX static library liblib__ecl.a
[ 49%] Built target lib__ecl
Linking CXX static library libmodules__mavlink.a
[ 49%] Built target modules__mavlink
make[2]: *** [src/firmware/posix/CMakeFiles/gazebo.dir/rule] Error 2
make[1]: *** [gazebo] Error 2
make: *** [posix_sitl_default] Error 2
```

L4ncelot, 2016-10-26:
I think the problem is related to `sitl_gazebo` package. Because when I was trying to run simulation making the project with `make posix_sitl_default` command and then running
`roslaunch px4 posix_sitl.launch verbose:="true"`

I could see that `plugin librotors_gazebo_multirotor_base_plugin.so` was not found. Therefore I was wondering whether can I compile `sitl_gazebo` package on it's own. Here is output of my build:

```
Scanning dependencies of target sdf
[  8%] [  8%] [ 12%] [ 16%] Running C++ protocol buffer compiler on msgs/Float.proto
Running C++ protocol buffer compiler on msgs/SensorImu.proto
Running C++ protocol buffer compiler on msgs/MotorSpeed.proto
Built target sdf
[ 20%] Running C++ protocol buffer compiler on msgs/opticalFlow.proto
[ 24%] [ 28%] Running C++ protocol buffer compiler on msgs/lidar.proto
Running C++ protocol buffer compiler on msgs/CommandMotorSpeed.proto
Scanning dependencies of target mav_msgs
[ 32%] [ 36%] [ 40%] [ 44%] Building CXX object CMakeFiles/mav_msgs.dir/Float.pb.cc.o
Building CXX object CMakeFiles/mav_msgs.dir/opticalFlow.pb.cc.o
Building CXX object CMakeFiles/mav_msgs.dir/SensorImu.pb.cc.o
Building CXX object CMakeFiles/mav_msgs.dir/lidar.pb.cc.o
[ 48%] Building CXX object CMakeFiles/mav_msgs.dir/CommandMotorSpeed.pb.cc.o
[ 52%] Building CXX object CMakeFiles/mav_msgs.dir/MotorSpeed.pb.cc.o
Linking CXX shared library libmav_msgs.so
[ 52%] Built target mav_msgs
Scanning dependencies of target gazebo_lidar_plugin
Scanning dependencies of target LiftDragPlugin
[ 56%] [ 60%] Building CXX object CMakeFiles/LiftDragPlugin.dir/src/liftdrag_plugin/liftdrag_plugin.cpp.o
Building CXX object CMakeFiles/gazebo_lidar_plugin.dir/src/gazebo_lidar_plugin.cpp.o
Scanning dependencies of target gazebo_opticalFlow_plugin
Scanning dependencies of target rotors_gazebo_controller_interface
[ 64%] Building C object CMakeFiles/gazebo_opticalFlow_plugin.dir/src/flow_settings.c.o
cc1: warning: command line option ‘-std=c++11’ is valid for C++/ObjC++ but not for C [enabled by default]
[ 68%] Building CXX object CMakeFiles/rotors_gazebo_controller_interface.dir/src/gazebo_controller_interface.cpp.o
[ 72%] Building C object CMakeFiles/gazebo_opticalFlow_plugin.dir/src/flow.c.o
cc1: warning: command line option ‘-std=c++11’ is valid for C++/ObjC++ but not for C [enabled by default]
[ 76%] Building CXX object CMakeFiles/gazebo_opticalFlow_plugin.dir/src/gazebo_opticalFlow_plugin.cpp.o
Linking CXX shared library libLiftDragPlugin.so
[ 76%] Built target LiftDragPlugin
Scanning dependencies of target rotors_gazebo_gimbal_controller_plugin
[ 80%] Building CXX object CMakeFiles/rotors_gazebo_gimbal_controller_plugin.dir/src/gazebo_gimbal_controller_plugin.cpp.o
Linking CXX shared library libgazebo_lidar_plugin.so
[ 80%] Built target gazebo_lidar_plugin
Scanning dependencies of target rotors_gazebo_imu_plugin
[ 84%] Building CXX object CMakeFiles/rotors_gazebo_imu_plugin.dir/src/gazebo_imu_plugin.cpp.o
Linking CXX shared library librotors_gazebo_controller_interface.so
[ 84%] Built target rotors_gazebo_controller_interface
Scanning dependencies of target rotors_gazebo_mavlink_interface
[ 88%] Building CXX object CMakeFiles/rotors_gazebo_mavlink_interface.dir/src/gazebo_mavlink_interface.cpp.o
Linking CXX shared library libgazebo_opticalFlow_plugin.so
[ 88%] Built target gazebo_opticalFlow_plugin
Scanning dependencies of target rotors_gazebo_motor_model
[ 92%] Building CXX object CMakeFiles/rotors_gazebo_motor_model.dir/src/gazebo_motor_model.cpp.o
/home/l4ncelot/src/Firmware/Tools/sitl_gazebo/src/gazebo_gimbal_controller_plugin.cpp: In member function ‘void gazebo::GimbalControllerPlugin::OnUpdate()’:
/home/l4ncelot/src/Firmware/Tools/sitl_gazebo/src/gazebo_gimbal_controller_plugin.cpp:312:43: error: no matching function for call to ‘gazebo::GimbalControllerPlugin::QtoZXY(ignition::math::Vector3d&)’
       this->QtoZXY(currentAngleYPRVariable));
                                           ^
/home/l4ncelot/src/Firmware/Tools/sitl_gazebo/src/gazebo_gimbal_controller_plugin.cpp:312:43: note: candidate is:
/home/l4ncelot/src/Firmware/Tools/sitl_gazebo/src/gazebo_gimbal_controller_plugin.cpp:251:26: note: ignition::math::Vector3d gazebo::GimbalControllerPlugin::QtoZXY(const Quaterniond&)
 ignition::math::Vector3d GimbalControllerPlugin::QtoZXY(
                          ^
/home/l4ncelot/src/Firmware/Tools/sitl_gazebo/src/gazebo_gimbal_controller_plugin.cpp:251:26: note:   no known conversion for argument 1 from ‘ignition::math::Vector3d {aka ignition::math::Vector3<double>}’ to ‘const Quaterniond& {aka const ignition::math::Quaternion<double>&}’
make[2]: *** [CMakeFiles/rotors_gazebo_gimbal_controller_plugin.dir/src/gazebo_gimbal_controller_plugin.cpp.o] Error 1
make[1]: *** [CMakeFiles/rotors_gazebo_gimbal_controller_plugin.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 96%] Building CXX object CMakeFiles/rotors_gazebo_mavlink_interface.dir/src/geo_mag_declination.cpp.o
Linking CXX shared library librotors_gazebo_imu_plugin.so
[ 96%] Built target rotors_gazebo_imu_plugin
Linking CXX shared library librotors_gazebo_mavlink_interface.so
[ 96%] Built target rotors_gazebo_mavlink_interface
Linking CXX shared library librotors_gazebo_motor_model.so
[ 96%] Built target rotors_gazebo_motor_model
make: *** [all] Error 2
```

julianoes, 2016-10-26:
This should be fixed, what is the output of `git rev-parse HEAD` in `Firmware` and Firmware/Tools/sitl_gazebo` ?

L4ncelot, 2016-10-26:
in `Firmware` : `1999ed26016b988b616da3ca451b66de30fee2c5`
in `Firmware/Tools/sitl_gazebo` : `380fce8b478361b086cef18597f885002975e701`

I've just cloned repository and updated it's submodules as it's described in documentation.

julianoes, 2016-10-26:
Ok that's good, what is the output of `apt-cache show libignition-common-dev`?
Is there a chance that you have compiled and install libignition yourself?

L4ncelot, 2016-10-26:
You're right. I've installed my version of Gazebo from source which required manual installation of `ignition` library. 
Here is the output:
```
Package: libignition-common-dev
Source: ignition-common
Version: 0.1.0-1~trusty
Architecture: amd64
Maintainer: Jose Luis Rivero <jrivero@osrfoundation.org>
Installed-Size: 316
Depends: libfreeimage-dev, libignition-math2-dev (>> 2.5), libtinyxml2-dev, uuid-dev, libgts-dev, libavformat-dev, libavcodec-dev, libswscale-dev, libignition-common0 (= 0.1.0-1~trusty)
Multi-Arch: same
Homepage: http://ignitionrobotics.org/
Priority: optional
Section: libdevel
Filename: pool/main/i/ignition-common/libignition-common-dev_0.1.0-1~trusty_amd64.deb
Size: 38712
SHA256: 70e262d671629de21473fc0466018a7b3bbee7c9639ea3ee940a83f9ffd18bf2
SHA1: cfb0402e5dec908c18fdbc985b3a57d9183043b0
MD5sum: 30821f351657269f8de35a2433109383
Description: Ignition Common classes and functions for robot apps - Development files
  Ignition Common is a component in the ignition framework, a set of libraries
  designed to rapidly develop robot applications.
Description-md5: ee411d12bf6408eac7b6e2f5120faaf5
```

julianoes, 2016-10-26:
Yer I'm not sure what to do in this case. I would propose to play around with ignition and figure out which version works.

L4ncelot, 2016-10-26:
Ok, I'll try this tomorrow and let you know. For now I just commented building `gazebo_gimbal_controller_plugin` and everything works fine. 

Thank you very much for help.

L4ncelot, 2016-11-26:
Ok, so I finally managed to try reinstalling **Gazebo** with preinstalled binaries instead of building it from source and the problem with  `gazebo_gimbal_controller_plugin` disappeared. Thanks again.

closed 2016-11-26 by L4ncelot