PX4/PX4-Autopilot #5755: Build failure: Race condition in initial clean build leads to failed linking step

LorenzMeier, 2016-10-25:
We can run into an initial race condition on the first build. It's not clear to me yet where it originates, I suspect in the NuttX build (the inbuilt apps do not get built). A subsequent build succeeds. We need to fix the root cause of this.

```
Building px4nucleoF767ZI-v1_default
GIT_SUBMODULES_ARE_EVIL is defined - Skipping All submodule checking!
-- nuttx-px4nucleoF767ZI-v1-default
-- CMAKE_INSTALL_PREFIX: 
-- cmake module path: /home/travis/build/PX4/Firmware/cmake
-- Nuttx build for px4nucleoF767ZI-v1 on m7 hardware, using nsh with ROMFS on px4fmu_common
-- The ASM compiler identification is GNU
-- Found assembler: /usr/lib/ccache/arm-none-eabi-gcc
-- catkin DISABLED
-- Found PythonInterp: /usr/bin/python2.7 (found suitable version "2.7.12", minimum required is "2.7") 
-- NuttX patch: nuttx-patches/Fixed-Shadow-wanings.patch
-- NuttX patch: nuttx-patches/nsh_parse.patch
-- NuttX patch: nuttx-patches/c++11.patch
-- NuttX patch: nuttx-patches/wip_inflight_to_upstream.patch
-- NuttX patch: nuttx-patches/silence-jobserver-warnings.patch
-- NuttX patch: nuttx-patches/static_asset-fix.patch
-- NuttX patch: nuttx-patches/wip_nxphlite_nuttx_cfg.patch
-- NuttX patch: nuttx-patches/ctype_cctype_.patch
-- NuttX patch: nuttx-patches/fixStdint.patch
-- NuttX patch: nuttx-patches/math.h.patch
-- Using C++03
-- Release build type: RelWithDebInfo
-- Adding UAVCAN STM32 platform driver
-- Adding ROMFS on px4nucleoF767ZI-v1
-- GIT_TAG = v1.5.0-1979-g866ec5f
-- Configuring done
-- Generating done
-- Build files have been written to: /home/travis/build/PX4/Firmware/build_px4nucleoF767ZI-v1_default
PX4 CONFIG: /home/travis/build/PX4/Firmware/build_px4nucleoF767ZI-v1_default
ninja: Entering directory `/home/travis/build/PX4/Firmware/build_px4nucleoF767ZI-v1_default'
[543/544] Linking CXX executable src/firmware/nuttx/firmware_nuttx
FAILED: : && /usr/lib/ccache/arm-none-eabi-g++   -g -fno-exceptions -fno-rtti -std=gnu++0x -fno-threadsafe-statics -DCONFIG_WCHAR_BUILTIN -D__CUSTOM_FILE_IO__ -fcheck-new -Wall -Werror -Wextra -Wno-sign-compare -Wshadow -Wfloat-equal -Wpointer-arith -Wmissing-declarations -Wno-unused-parameter -Werror=format-security -Werror=array-bounds -Wfatal-errors -Werror=unused-variable -Werror=reorder -Werror=uninitialized -Werror=init-self -Werror=unused-but-set-variable -Wformat=1 -Wdouble-promotion -Werror=double-promotion -Wno-missing-field-initializers -fvisibility=hidden -include visibility.h -mcpu=cortex-m7 -mthumb -mfpu=fpv5-sp-d16 -mfloat-abi=hard -nodefaultlibs -nostdlib -Os -fno-strict-aliasing -fomit-frame-pointer -funsafe-math-optimizations -ffunction-sections -fdata-sections -fno-strength-reduce -fno-builtin-printf -O2 -g -DNDEBUG  -Wl,--warn-common -Wl,--gc-sections   -T/home/travis/build/PX4/Firmware/build_px4nucleoF767ZI-v1_default/px4nucleoF767ZI-v1/NuttX/nuttx-export/build/ld.script -Wl,-Map=/home/travis/build/PX4/Firmware/build_px4nucleoF767ZI-v1_default/px4nucleoF767ZI-v1/main.map src/firmware/nuttx/CMakeFiles/firmware_nuttx.dir/builtin_commands.c.obj  -o src/firmware/nuttx/firmware_nuttx -L/home/travis/build/PX4/Firmware/build_px4nucleoF767ZI-v1_default/px4nucleoF767ZI-v1/NuttX/nuttx-export/libs  -L/home/travis/build/PX4/Firmware/build_px4nucleoF767ZI-v1_default/external/Install/lib -Wl,--warn-common -Wl,--gc-sections -Wl,--start-group src/drivers/device/libdrivers__device.a src/drivers/stm32/libdrivers__stm32.a src/drivers/stm32/adc/libdrivers__stm32__adc.a src/drivers/stm32/tone_alarm/libdrivers__stm32__tone_alarm.a src/drivers/led/libdrivers__led.a src/drivers/px4fmu/libdrivers__px4fmu.a src/drivers/boards/px4nucleoF767ZI-v1/libdrivers__boards__px4nucleoF767ZI-v1.a src/drivers/rgbled/libdrivers__rgbled.a src/drivers/mpu6000/libdrivers__mpu6000.a src/drivers/mpu9250/libdrivers__mpu9250.a src/drivers/hmc5883/libdrivers__hmc5883.a src/drivers/ms5611/libdrivers__ms5611.a src/drivers/mb12xx/libdrivers__mb12xx.a src/drivers/srf02/libdrivers__srf02.a src/drivers/sf0x/libdrivers__sf0x.a src/drivers/ll40ls/libdrivers__ll40ls.a src/drivers/trone/libdrivers__trone.a src/drivers/gps/libdrivers__gps.a src/drivers/pwm_out_sim/libdrivers__pwm_out_sim.a src/drivers/hott/libdrivers__hott.a src/drivers/hott/hott_telemetry/libdrivers__hott__hott_telemetry.a src/drivers/hott/hott_sensors/libdrivers__hott__hott_sensors.a src/drivers/blinkm/libdrivers__blinkm.a src/drivers/airspeed/libdrivers__airspeed.a src/drivers/ets_airspeed/libdrivers__ets_airspeed.a src/drivers/meas_airspeed/libdrivers__meas_airspeed.a src/drivers/frsky_telemetry/libdrivers__frsky_telemetry.a src/modules/sensors/libmodules__sensors.a src/drivers/mkblctrl/libdrivers__mkblctrl.a src/drivers/px4flow/libdrivers__px4flow.a src/drivers/oreoled/libdrivers__oreoled.a src/drivers/pwm_input/libdrivers__pwm_input.a src/drivers/camera_trigger/libdrivers__camera_trigger.a src/drivers/bst/libdrivers__bst.a src/drivers/snapdragon_rc_pwm/libdrivers__snapdragon_rc_pwm.a src/drivers/lis3mdl/libdrivers__lis3mdl.a src/drivers/bmp280/libdrivers__bmp280.a src/drivers/bma180/libdrivers__bma180.a src/drivers/bmi160/libdrivers__bmi160.a src/drivers/tap_esc/libdrivers__tap_esc.a src/systemcmds/bl_update/libsystemcmds__bl_update.a src/systemcmds/config/libsystemcmds__config.a src/systemcmds/dumpfile/libsystemcmds__dumpfile.a src/systemcmds/esc_calib/libsystemcmds__esc_calib.a src/systemcmds/hardfault_log/libsystemcmds__hardfault_log.a src/systemcmds/mixer/libsystemcmds__mixer.a src/systemcmds/motor_ramp/libsystemcmds__motor_ramp.a src/systemcmds/mtd/libsystemcmds__mtd.a src/systemcmds/nshterm/libsystemcmds__nshterm.a src/systemcmds/param/libsystemcmds__param.a src/systemcmds/perf/libsystemcmds__perf.a src/systemcmds/pwm/libsystemcmds__pwm.a src/systemcmds/reboot/libsystemcmds__reboot.a src/systemcmds/sd_bench/libsystemcmds__sd_bench.a src/systemcmds/top/libsystemcmds__top.a src/systemcmds/topic_listener/libsystemcmds__topic_listener.a src/systemcmds/ver/libsystemcmds__ver.a src/modules/commander/libmodules__commander.a src/modules/load_mon/libmodules__load_mon.a src/modules/navigator/libmodules__navigator.a src/modules/mavlink/libmodules__mavlink.a src/modules/gpio_led/libmodules__gpio_led.a src/modules/uavcan/libmodules__uavcan.a src/modules/land_detector/libmodules__land_detector.a src/modules/attitude_estimator_q/libmodules__attitude_estimator_q.a src/modules/position_estimator_inav/libmodules__position_estimator_inav.a src/modules/local_position_estimator/libmodules__local_position_estimator.a src/modules/ekf2/libmodules__ekf2.a src/modules/fw_pos_control_l1/libmodules__fw_pos_control_l1.a src/modules/fw_att_control/libmodules__fw_att_control.a src/modules/mc_att_control/libmodules__mc_att_control.a src/modules/mc_pos_control/libmodules__mc_pos_control.a src/modules/vtol_att_control/libmodules__vtol_att_control.a src/modules/logger/libmodules__logger.a src/modules/sdlog2/libmodules__sdlog2.a src/modules/param/libmodules__param.a src/modules/systemlib/libmodules__systemlib.a src/modules/systemlib/mixer/libmodules__systemlib__mixer.a src/modules/uORB/libmodules__uORB.a src/modules/dataman/libmodules__dataman.a src/lib/controllib/liblib__controllib.a src/lib/mathlib/liblib__mathlib.a src/lib/mathlib/math/filter/liblib__mathlib__math__filter.a src/lib/rc/liblib__rc.a src/lib/ecl/liblib__ecl.a src/lib/external_lgpl/liblib__external_lgpl.a src/lib/geo/liblib__geo.a src/lib/geo_lookup/liblib__geo_lookup.a src/lib/conversion/liblib__conversion.a src/lib/launchdetection/liblib__launchdetection.a src/lib/terrain_estimation/liblib__terrain_estimation.a src/lib/runway_takeoff/liblib__runway_takeoff.a src/lib/tailsitter_recovery/liblib__tailsitter_recovery.a src/lib/DriverFramework/framework/liblib__DriverFramework__framework.a src/platforms/nuttx/libplatforms__nuttx.a src/platforms/common/libplatforms__common.a src/platforms/nuttx/px4_layer/libplatforms__nuttx__px4_layer.a src/modules/bottle_drop/libmodules__bottle_drop.a src/examples/rover_steering_control/libexamples__rover_steering_control.a src/examples/px4_simple_app/libexamples__px4_simple_app.a src/modules/uavcan/libuavcan/libuavcan/libuavcan.a src/modules/uavcan/libuavcan/libuavcan_drivers/stm32/driver/libuavcan_stm32_driver.a -lapps -lnuttx -lm -lgcc src/firmware/nuttx/libromfs.a -Wl,--end-group libmsg_gen.a src/lib/DriverFramework/framework/src/libdf_driver_framework.a && :
/home/travis/build/PX4/Firmware/build_px4nucleoF767ZI-v1_default/px4nucleoF767ZI-v1/NuttX/nuttx-export/libs/libapps.a(nsh_session.o): In function `nsh_session':
/home/travis/build/PX4/Firmware/build_px4nucleoF767ZI-v1_default/px4nucleoF767ZI-v1/NuttX/apps/nshlib/nsh_session.c:148: undefined reference to `readline'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
Makefile:166: recipe for target 'px4nucleoF767ZI-v1_default' failed
make[1]: *** [px4nucleoF767ZI-v1_default] Error 1
Makefile:279: recipe for target 'check_px4nucleoF767ZI-v1_default' failed
make: *** [check_px4nucleoF767ZI-v1_default] Error 2
```

dagar, 2016-10-25:
This is nuttx_v3 right?

LorenzMeier, 2016-10-25:
Yes. But I have seen it on current NuttX on my laptop 1-2 times in the last year as well.

davids5, 2016-10-26:
@dagar 

Would you have time to look at  conditionally disabling parallel build of nuttx on a branch of nuttx_v3?

The idea is we can set a env var on CI to stop the Nuttx parallel build and validate the hypothesis. 

I know it will slow things down, but statistically as the the number targets goes up in nuttx_v3 - I can not make it through a single CI cycle on a build of  code that builds.  So the net the time lost would be a savings.

dagar, 2016-10-26:
It's `-j${THREADS}` here -https://github.com/PX4/Firmware/blob/nuttx_v3/cmake/nuttx/px4_impl_nuttx.cmake#L310

Which is a cmake variable you can override - https://github.com/PX4/Firmware/blob/nuttx_v3/CMakeLists.txt#L146

LorenzMeier, 2016-10-29:
Yeah, but that's not fixing it locally either. So the real thing is to fix the generation order of the builtin apps header in NuttX so its not generated too late.

davids5, 2016-10-29:
I have my FIRST local failure yesterday. I can not tell you how many builds I have run. So I like what @LorenzMeier is suggesting. If we order the NuttX/apps and builtin build to be dependant on the Nuttx/nuttx builds that should help. @dagar How do we do that?

dagar, 2016-11-28:
@LorenzMeier As discussed with @davids5 I believe this is temporarily resolved. The issue is the way the nuttx build calls ar incrementally (https://www.gnu.org/software/make/manual/html_node/Archive-Pitfalls.html). For now I made the nuttx build single threaded in nuttx_v3, which surprisingly/disappointingly only increases the build time on my laptop from ~20 to 25 seconds.

If we want to fix the fundamental issue I think we should probably take the discussion upstream.

Please keep opening issues if you see any build failures at all.

closed 2016-11-28 by dagar

davids5, 2016-11-28:
@dagar I do not think upstream is option - build changes to suite alternate build system are typically rejected by @nuttx

dagar, 2016-11-28:
Unless I'm wrong about the underlying issue, it seems like it should also be a problem for a standalone Nuttx build. We'd just need to find a way to only call archive once per lib.