PX4/PX4-Autopilot #5756: Snapdragon: linking fails on QURT

julianoes, 2016-10-26:
Unfortunately #5715 leads to "linking" errors on the Snapdragon. The linking on Snapdragon doesn't happen until you start the px4 bin and it loads the part running on the DSP. It then resolves all the symbols it needs.

I verified and it definitely does compile `apps.cpp` in and it even passes the `make qurt_travis_default` build test which supposedly should detect linking errors.

@CarloWood any advice or idea how to debug this?

```
    556734: HAP:12350:undefined PLT symbol _Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE (1194) /libp (symbol.c:303)
    556734: HAP:12350:undefined PLT symbol _Z13list_builtinsRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE (77) /libpx (symbol.c:303)
    556753: HAP:12350:undefined PLT symbol _Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE (1194) /libp (symbol.c:303)
    556753: HAP:12350:undefined PLT symbol _Z13list_builtinsRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE (77) /libpx (symbol.c:303)
    556772: HAP:12350:undefined PLT symbol _Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE (1194) /libp (symbol.c:303)
    556772: HAP:12350:undefined PLT symbol _Z13list_builtinsRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE (77) /libpx (symbol.c:303)
    556792: HAP:12350:undefined PLT symbol _Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE (1194) /libp (symbol.c:303)
    556792: HAP:12350:undefined PLT symbol _Z13list_builtinsRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE (77) /libpx (symbol.c:303)
    556811: HAP:12350:undefined PLT symbol _Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE (1194) /libp (symbol.c:303)
    556811: HAP:12350:undefined PLT symbol _Z13list_builtinsRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE (77) /libpx (symbol.c:303)
    556829: HAP:12350:undefined PLT symbol _Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE (1194) /libp (symbol.c:303)
    556829: HAP:12350:undefined PLT symbol _Z13list_builtinsRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE (77) /libpx (symbol.c:303)
    556848: HAP:12350:undefined PLT symbol _Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE (1194) /libp (symbol.c:303)
    556848: HAP:12350:undefined PLT symbol _Z13list_builtinsRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE (77) /libpx (symbol.c:303)
    556868: HAP:12350:undefined PLT symbol _Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE (1194) /libp (symbol.c:303)
    556868: HAP:12350:undefined PLT symbol _Z13list_builtinsRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE (77) /libpx (symbol.c:303)
    556888: HAP:12350:undefined PLT symbol _Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE (1194) /libp (symbol.c:303)
    556888: HAP:12350:undefined PLT symbol _Z13list_builtinsRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE (77) /libpx (symbol.c:303)
    556907: HAP:12350:undefined PLT symbol _Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE (1194) /libp (symbol.c:303)
    556907: HAP:12350:undefined PLT symbol _Z13list_builtinsRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE (77) /libpx (symbol.c:303)
    557353: HAP:12350:undefined PLT symbol _Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE (1194) /libp (symbol.c:303)
    557353: HAP:12350:undefined PLT symbol _Z13list_builtinsRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE (77) /libpx (symbol.c:303)
    557380: HAP:12350:undefined PLT symbol _Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE (1194) /libp (symbol.c:303)
    557380: HAP:12350:undefined PLT symbol _Z13list_builtinsRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE (77) /libpx (symbol.c:303)
    557716: HAP:12349:undefined PLT symbol _Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE (1194) /libp (symbol.c:303)
    557716: HAP:12349:undefined PLT symbol _Z13list_builtinsRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE (77) /libpx (symbol.c:303)
```

CarloWood, 2016-10-27:
Oh, is there no way I can reproduce this?

CarloWood, 2016-10-27:
Starting work on this now...

CarloWood, 2016-10-27:
I'm confused #5715 says that it is merged = but I don't see it in master... Did you force push master to remove it again??

CarloWood, 2016-10-27:
I do not have that target at all?!
```
sean:/usr/src/px4/Firmware/Firmware.git>make qurt_travis_default
Makefile:286: *** "qurt_travis_default cannot be the first argument. Use '/usr/bin/make help|list_config_targets' to get a list of all possible [configuration] targets.".  Stop.
```

CarloWood, 2016-10-27:
I have 'qurt_eagle_travis' and 'qurt_eagle_default' and 'qurt_excelsior_default'. The first containing the word 'travis' and the latter two causing cmake/configs/qurt_sdflight_default.cmake to be included, where the 'sd' stands for snapdragon... so that's as close as I can test. I'd like to understand how on earth you can compile a qurt_travis_default though, you're probably not using master?

CarloWood, 2016-10-27:
Ok, here's what I did so far:

1. I checkout out the commit one before the 'Clean up of app.h' commit, namely c200ef88edb314b852ec4a5bcfb4042d5dfc6da8
2. I applied my  qurt_target_fixes.diff patch in order to be able to compile qurt_eagle_travis.
3. I ran ```make qurt_eagle_travis; make qurt_eagle_default; make qurt_excelsior_default```
4. I found where that generated and/or linked with ```init_app_map(std::map<std::basic_string<char, std::char_traits<char>, std::allocator<char> >, int (*)(int, char**), std::less<std::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::basic_string<char, std::char_traits<char>, std::allocator<char> > const, int (*)(int, char**)> > >&)``` by running:

```
sean:/usr/src/px4/Firmware/Firmware.git>for f in $(find build_qurt_eagle_travis -type f -exec file {} \; | grep ': ELF' | sed -e 's/:.*//'); do echo "$f"; readelf -W --symbols "$f" | grep '_Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE'; done | grep -B 1 '_Z12init'
build_qurt_eagle_travis/src/firmware/qurt/px4
  1185: 000268a0  1716 FUNC    GLOBAL HIDDEN     8 _Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE
--
build_qurt_eagle_travis/src/platforms/qurt/px4_layer/CMakeFiles/platforms__qurt__px4_layer.dir/main.cpp.o
   112: 00000000  1716 FUNC    GLOBAL HIDDEN    44 _Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE

sean:/usr/src/px4/Firmware/Firmware.git>for f in $(find build_qurt_eagle_default -type f -exec file {} \; | grep ': ELF' | sed -e 's/:.*//'); do echo "$f"; readelf -W --symbols "$f" | grep '_Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE'; done | grep -B 1 '_Z12init'
build_qurt_eagle_default/src/firmware/qurt/libpx4.so
  1980: 000a9c70  2276 FUNC    GLOBAL HIDDEN     8 _Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE
--
build_qurt_eagle_default/src/firmware/qurt/libpx4muorb_skel.so
  1835: 000a5850  2276 FUNC    GLOBAL HIDDEN     8 _Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE
--
build_qurt_eagle_default/src/platforms/qurt/px4_layer/CMakeFiles/platforms__qurt__px4_layer.dir/main.cpp.o
   118: 00000000  2276 FUNC    GLOBAL HIDDEN    44 _Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE
--
build_qurt_eagle_default/src/drivers/qshell/qurt/CMakeFiles/drivers__qshell__qurt.dir/qshell.cpp.o
   112: 00000000     0 NOTYPE  GLOBAL DEFAULT  UND _Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE

sean:/usr/src/px4/Firmware/Firmware.git>for f in $(find build_qurt_excelsior_default -type f -exec file {} \; | grep ': ELF' | sed -e 's/:.*//'); do echo "$f"; readelf -W --symbols "$f" | grep '_Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE'; done | grep -B 1 '_Z12init'
build_qurt_excelsior_default/src/firmware/qurt/libpx4.so
  1980: 000a9c70  2276 FUNC    GLOBAL HIDDEN     8 _Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE
--
build_qurt_excelsior_default/src/firmware/qurt/libpx4muorb_skel.so
  1835: 000a5850  2276 FUNC    GLOBAL HIDDEN     8 _Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE
--
build_qurt_excelsior_default/src/platforms/qurt/px4_layer/CMakeFiles/platforms__qurt__px4_layer.dir/main.cpp.o
   118: 00000000  2276 FUNC    GLOBAL HIDDEN    44 _Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE
--
build_qurt_excelsior_default/src/drivers/qshell/qurt/CMakeFiles/drivers__qshell__qurt.dir/qshell.cpp.o
   112: 00000000     0 NOTYPE  GLOBAL DEFAULT  UND _Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE
```

So, already here we can see what is the problem: it's HIDDEN... The visibility wasn't set correctly after I moved the definitions to apps.cpp

CarloWood, 2016-10-27:
Next I checked out 6fc30c76a6cbc2d3b876c0d001b9751ecfbaf260 (Clean up of app.h),
reran the same commands:

```
sean:/usr/src/px4/Firmware/Firmware.git>for f in $(find build_qurt_eagle_travis -type f -exec file {} \; | grep ': ELF' | sed -e 's/:.*//'); do echo "$f"; readelf -W --symbols "$f" | grep '_Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE'; done | grep -B 1 '_Z12init'
build_qurt_eagle_travis/src/firmware/qurt/CMakeFiles/px4.dir/__/__/__/apps.cpp.o
   185: 00000000  1716 FUNC    GLOBAL HIDDEN   101 _Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE
build_qurt_eagle_travis/src/firmware/qurt/px4
  1257: 00020e2c  1716 FUNC    GLOBAL HIDDEN     8 _Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE
--
build_qurt_eagle_travis/src/platforms/qurt/px4_layer/CMakeFiles/platforms__qurt__px4_layer.dir/main.cpp.o
   101: 00000000     0 NOTYPE  GLOBAL DEFAULT  UND _Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE

sean:/usr/src/px4/Firmware/Firmware.git>for f in $(find build_qurt_eagle_default -type f -exec file {} \; | grep ': ELF' | sed -e 's/:.*//'); do echo "$f"; readelf -W --symbols "$f" | grep '_Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE'; done | grep -B 1 '_Z12init'
build_qurt_eagle_default/src/firmware/qurt/libpx4.so
  2051: 0002e57c  2276 FUNC    GLOBAL HIDDEN     8 _Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE
--
build_qurt_eagle_default/src/firmware/qurt/CMakeFiles/px4.dir/__/__/__/apps.cpp.o
   191: 00000000  2276 FUNC    GLOBAL HIDDEN   101 _Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE
build_qurt_eagle_default/src/firmware/qurt/libpx4muorb_skel.so
  1180: 00000000     0 NOTYPE  GLOBAL DEFAULT  UND _Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE
  3206: 00000000     0 NOTYPE  GLOBAL DEFAULT  UND _Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE
--
build_qurt_eagle_default/src/platforms/qurt/px4_layer/CMakeFiles/platforms__qurt__px4_layer.dir/main.cpp.o
   101: 00000000     0 NOTYPE  GLOBAL DEFAULT  UND _Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE
--
build_qurt_eagle_default/src/drivers/qshell/qurt/CMakeFiles/drivers__qshell__qurt.dir/qshell.cpp.o
   112: 00000000     0 NOTYPE  GLOBAL DEFAULT  UND _Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE

sean:/usr/src/px4/Firmware/Firmware.git>for f in $(find build_qurt_excelsior_default -type f -exec file {} \; | grep ': ELF' | sed -e 's/:.*//'); do echo "$f"; readelf -W --symbols "$f" | grep '_Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE'; done | grep -B 1 '_Z12init'
build_qurt_excelsior_default/src/firmware/qurt/libpx4.so
  2051: 0002e57c  2276 FUNC    GLOBAL HIDDEN     8 _Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE
--
build_qurt_excelsior_default/src/firmware/qurt/CMakeFiles/px4.dir/__/__/__/apps.cpp.o
   191: 00000000  2276 FUNC    GLOBAL HIDDEN   101 _Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE
build_qurt_excelsior_default/src/firmware/qurt/libpx4muorb_skel.so
  1180: 00000000     0 NOTYPE  GLOBAL DEFAULT  UND _Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE
  3206: 00000000     0 NOTYPE  GLOBAL DEFAULT  UND _Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE
--
build_qurt_excelsior_default/src/platforms/qurt/px4_layer/CMakeFiles/platforms__qurt__px4_layer.dir/main.cpp.o
   101: 00000000     0 NOTYPE  GLOBAL DEFAULT  UND _Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE
--
build_qurt_excelsior_default/src/drivers/qshell/qurt/CMakeFiles/drivers__qshell__qurt.dir/qshell.cpp.o
   112: 00000000     0 NOTYPE  GLOBAL DEFAULT  UND _Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE
```

We see a lot more UNDefined references, and the object is defined only in apps.cpp.o (and through that in qurt/libpx4.so, but it is HIDDEN.

So, what we need is a patch that will change that into non HIDDEN.

CarloWood, 2016-10-27:
Next I created a new branch (I don't know how I could add a pull request for this PR, is that possible?),
and removed my qurt_target_fixes.diff because it seems enough to concentrate on just qurt_eagle_default to get this fixed.

I wrote what seemed to be the most logical attempt at fixing what you reported, nl. adding __EXPORT in apps.h.in for those two functions. And then reran the above command(s) again:

```
sean:/usr/src/px4/Firmware/Firmware.git>for f in $(find build_qurt_eagle_default -type f -exec file {} \; | grep ': ELF' | sed -e 's/:.*//'); do echo "$f"; readelf -W --symbols "$f" | grep '_Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE'; done | grep -B 1 '_Z12init'
build_qurt_eagle_default/src/firmware/qurt/libpx4.so
  1641: 0002e63c  2276 FUNC    GLOBAL DEFAULT    8 _Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE
  3888: 0002e63c  2276 FUNC    GLOBAL DEFAULT    8 _Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE
--
build_qurt_eagle_default/src/firmware/qurt/CMakeFiles/px4.dir/__/__/__/apps.cpp.o
   191: 00000000  2276 FUNC    GLOBAL DEFAULT  101 _Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE
build_qurt_eagle_default/src/firmware/qurt/libpx4muorb_skel.so
  1180: 00000000     0 FUNC    GLOBAL DEFAULT  UND _Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE
--
build_qurt_eagle_default/src/platforms/qurt/px4_layer/CMakeFiles/platforms__qurt__px4_layer.dir/main.cpp.o
   101: 00000000     0 NOTYPE  GLOBAL DEFAULT  UND _Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE
--
build_qurt_eagle_default/src/drivers/qshell/qurt/CMakeFiles/drivers__qshell__qurt.dir/qshell.cpp.o
   112: 00000000     0 NOTYPE  GLOBAL DEFAULT  UND _Z12init_app_mapRSt3mapISsPFiiPPcESt4lessISsESaISt4pairIKSsS3_EEE
```

And since the HIDDEN is gone I assume this will have fixed it.
However, there are more functions declared in apps.h.in that might also be used (obviously) and also need to be visible outside of libpx4.so. I'm not sure how to test this, but at least I can detect any ALREADY existing problems by looking for UND for them in other object files.

CarloWood, 2016-10-27:
Looking at apps.h.in I can make a reg.exp. that would catch any function declared there *except* the ones that are generated and part of builtin_apps_decl_string, and that is:
```(shutdown_main|list_tasks_main|list_files_main|list_devices_main|list_topics_main|sleep_main)```

So, we can run:
```
sean:/usr/src/px4/Firmware/Firmware.git>readelf -s -W build_qurt_eagle_default/src/firmware/qurt/CMakeFiles/px4.dir/__/__/__/apps.cpp.o | grep -E '(shutdown_main|list_tasks_main|list_files_main|list_devices_main|list_topics_main|sleep_main)'
    22: 00000000    64 FUNC    LOCAL  DEFAULT  245 shutdown_main.extracted_region
    23: 00000000    32 FUNC    LOCAL  DEFAULT  249 shutdown_main.extracted_region45
    24: 00000000    44 FUNC    LOCAL  DEFAULT  250 shutdown_main.extracted_region46
    25: 00000000    20 FUNC    LOCAL  DEFAULT  266 shutdown_main.extracted_region46.1.2.3
    26: 00000000    80 FUNC    LOCAL  DEFAULT  253 shutdown_main.extracted_region48
    27: 00000000    52 FUNC    LOCAL  DEFAULT  256 shutdown_main.extracted_region49
   272: 00000000    12 FUNC    GLOBAL HIDDEN   111 list_devices_main
   273: 00000000    12 FUNC    GLOBAL HIDDEN   109 list_files_main
   274: 00000000    12 FUNC    GLOBAL HIDDEN   107 list_tasks_main
   275: 00000000    12 FUNC    GLOBAL HIDDEN   113 list_topics_main
   276: 00000000   256 FUNC    GLOBAL HIDDEN   105 shutdown_main
   277: 00000000   360 FUNC    GLOBAL HIDDEN   115 sleep_main
```

showing that also these (as expected) all HIDDEN. The question is now: do they need to be visible?
I think they are only referenced once by putting them into a table which is also generated... So, basically, no, they need not to be visible.

And indeed, running ```for f in $(find build_qurt_eagle_default -type f -exec file {} \; | grep ': ELF' | sed -e 's/:.*//'); do echo "$f"; readelf -W --symbols "$f" | grep -E '(shutdown_main|list_tasks_main|list_files_main|list_devices_main|list_topics_main|sleep_main)'; done``` reveals that there are not external references (UND) to them.

The functions (grepped for with this reg.exp) are only used in init_app_map itself, in apps.cpp.

So, they can be removed from apps.h.in and moved to apps.cpp.in and need not be __EXPORT-ed.

Finally, the builtin_apps_string will contain functions that ARE in other files, but which probably already are visible.

CarloWood, 2016-10-27:
I wrote the patch for the above. And verified of one of generated commands that my last remark is correct:
```
src/modules/commander/commander.cpp:extern "C" __EXPORT int commander_main(int argc, char *argv[]);
```
aka, it is already marked as __EXPORT.

CarloWood, 2016-10-27:
I made a PR / proposal for fix, here: https://github.com/PX4/Firmware/pull/5765
Please let me know if this solved the problem.

mcharleb, 2016-10-28:
I used to have a build target that did a full link with stubs (eagle_travis) to catch unresolved symbols.

LorenzMeier, 2016-10-29:
Closing this for now.

CarloWood, 2016-10-29:
@mcharleb apps.h/cpp use <map>, but so did this before my change. Note however that this is not used by DriverFramework, only by src/platforms/posix/main.cpp, src/platforms/qurt/px4_layer/main.cpp and src/drivers/qshell/qurt/qshell.{h,cpp}.

julianoes, 2016-11-01:
Sorry, I've been busy and wasn't watching here. I've retested this and get:

```
    304775: HAP:4179:undefined PLT symbol _Dtest (1328) /libpx4muorb_skel.so (symbol.c:303)
```

mcharleb, 2016-11-01:
@CarloWood Ah, so this all runs on Apps proc. That should be fine.
@jywilson Is _Dtest a symbol you have seen before? Does it need to be exported?

CarloWood, 2016-11-02:
2:30 AM ... but lemme have a quick attempt...

On current master, with ```make qurt_eagle_default```, I get:
```
sean:/usr/src/px4/Firmware/Firmware.git>for f in $(find build_qurt_eagle_default -type f -exec file {} \; | grep ': ELF' | sed -e 's/:.*//'); do echo "$f"; readelf -W --symbols "$f" | grep -E '(_Dtest)'; done | grep -B 1 _Dtest
build_qurt_eagle_default/src/firmware/qurt/libpx4.so
  1850: 00000000     0 FUNC    GLOBAL DEFAULT  UND _Dtest
--
build_qurt_eagle_default/src/firmware/qurt/libpx4muorb_skel.so
  1328: 00000000     0 FUNC    GLOBAL DEFAULT  UND _Dtest
--
build_qurt_eagle_default/src/modules/mc_pos_control/CMakeFiles/modules__mc_pos_control.dir/mc_pos_control_main.cpp.o
   211: 00000000     0 NOTYPE  GLOBAL DEFAULT  UND _Dtest
--
build_qurt_eagle_default/src/modules/local_position_estimator/CMakeFiles/modules__local_position_estimator.dir/BlockLocalPositionEstimator.cpp.o
   310: 00000000     0 NOTYPE  GLOBAL DEFAULT  UND _Dtest
```
while c200ef8 gives no output - aka, there isn't a symbol _Dtest defined, but also not used.
So, this is a bit like the locale: the question is why is this symbol suddenly needed?

CarloWood, 2016-11-02:
Next I tested 6fc30c7 (Clean up of app.h), and also there there is no _Dtest anywhere.
I then cherry-picked 52b8e75aeadd59e0f12f3de3f9e5759ead7f592b and 0bff3593d3ecc766f7c8981a3259d10a9110833d and still no _Dtest.

So, I'm afraid this is unrelated to the 6fc30c7 (Clean up of app.h) commit.

CarloWood, 2016-11-02:
I did a git bisect and
```
baf89f4398ae9bc8bfdf6c0761aaba0667ba9db5 is the first bad commit
commit baf89f4398ae9bc8bfdf6c0761aaba0667ba9db5
Author: Carlo Wood <carlo@alinoe dot com>
Date:   Mon Oct 24 18:05:04 2016 +0200

    Clean up of px4_defines.h (remove math.h)
```

I'll look into this tomorrow and figure out where that symbol comes from :(

julianoes, 2016-11-02:
@CarloWood that's really odd! Thanks for your help here.

venuannamraju, 2016-11-02:
Hi Julian, Carlo
I get the same undefined PLT symbol _Dtest error on the mini-dm along with "FastRPC initialization failed" on the adb shell.
I used the oct 19 fc addon that helped pass the DSPAL tests. 
Has anyone got the snapdragon flight build working successfully with earlier versions ?
Any help is most appreciated.

venu

CarloWood, 2016-11-02:
Good morning... starting work this now.

CarloWood, 2016-11-02:
The problem is that Qualcomm/HEXAGON_Tools/7.2.12 defines a ::isfinite() as follows:

```
extern "C" {
short _Dtest(double *);
}
inline int fpclassify(double _Left)
 {
 return (_Dtest(&_Left));
 }
template<class _Ty> inline
 bool isfinite(_Ty _Left)
 {
 return (fpclassify(_Left) <= 0);
 }
namespace std {
using :: isfinite;
}
```
But this _Dtest seems to be not defined anywhere (that I can find).

You'd think it should be defined in some libm.so, because the above gets
dragged in from including <cmath>, but of the gazzilion libm.so's that we
have:
```
/usr/src/px4/snapdragon/Qualcomm/Hexagon_SDK/3.0/gcc-linaro-4.9-2014.11-x86_64_arm-linux-gnueabihf_linux/libc/usr/lib/libm.so
/usr/src/px4/snapdragon/Qualcomm/qrlinux_v3.1.1_sysroot/linaro-rootfs/usr/lib/arm-linux-gnueabihf/libm.so
/usr/src/px4/snapdragon/Qualcomm/qrlinux_v3.1.1_sysroot/usr/include/eglibc-locale-internal-cortexa8hf-vfp-neon-linux-gnueabi/usr/lib/libm.so
/usr/src/px4/snapdragon/Qualcomm/qrlinux_v3.1.1_sysroot/usr/lib/libm.so
/usr/src/px4/snapdragon/cross_toolchain/downloads/qrlSDK/sysroots/eagle8074/linaro-rootfs/usr/lib/arm-linux-gnueabihf/libm.so
/usr/src/px4/snapdragon/cross_toolchain/downloads/qrlSDK/sysroots/eagle8074/usr/include/eglibc-locale-internal-cortexa8hf-vfp-neon-linux-gnueabi/usr/lib/libm.so
/usr/src/px4/snapdragon/cross_toolchain/downloads/qrlSDK/sysroots/eagle8074/usr/lib/libm.so
```
None of them has a symbol _Dtest. That is, after ignoring the four broken symbolic links in there, it turns out to be three copies of the same libm-2.17-2013.07-2.so in /usr/src/px4/snapdragon/cross_toolchain/downloads/qrlSDK and /usr/src/px4/snapdragon/Qualcomm/Hexagon_SDK/3.0/gcc-linaro-4.9-2014.11-x86_64_arm-linux-gnueabihf_linux/libc/usr/lib/libm.so where the latter is the best candidate to define _Dtest, because it belongs to the same package as the math.h that we're including (/usr/src/px4/snapdragon/Qualcomm/HEXAGON_Tools/7.2.12/Tools/bin/../target/hexagon/include/cmath --> /usr/src/px4/snapdragon/Qualcomm/HEXAGON_Tools/7.2.12/Tools/bin/../target/hexagon/include/math.h --> /usr/src/px4/snapdragon/Qualcomm/HEXAGON_Tools/7.2.12/Tools/bin/../target/hexagon/include/ymath.h which declares the undefined _Dtest). 

So, I'm just going for a "broken HEXAGON math lib", where probably nothing works that tests anything. 

In fact,
```
>readelf -s -W /usr/src/px4/snapdragon/Qualcomm/Hexagon_SDK/3.0/gcc-linaro-4.9-2014.11-x86_64_arm-linux-gnueabihf_linux/libc/usr/lib/libm.so | grep ' _[^_]'
     4: 00000000     0 NOTYPE  WEAK   DEFAULT  UND _ITM_deregisterTMCloneTable
    13: 00000000     0 NOTYPE  WEAK   DEFAULT  UND _Jv_RegisterClasses
    16: 00000000     0 NOTYPE  WEAK   DEFAULT  UND _ITM_registerTMCloneTable
   227: 0007307c     4 OBJECT  WEAK   DEFAULT   23 _LIB_VERSION@@GLIBC_2.4
  2025: 00073090     0 NOTYPE  LOCAL  DEFAULT   24 _bss_end__
  2028: 00073088     0 NOTYPE  LOCAL  DEFAULT   23 _edata
  2049: 0002e218     0 FUNC    LOCAL  DEFAULT   13 _fini
  2211: 00073090     0 NOTYPE  LOCAL  DEFAULT   24 _end
  2268: 00072ef8     0 OBJECT  LOCAL  DEFAULT  ABS _DYNAMIC
  2319: 00073000     0 OBJECT  LOCAL  DEFAULT  ABS _GLOBAL_OFFSET_TABLE_
  2352: 00003b78     0 FUNC    LOCAL  DEFAULT   10 _init
  2376: 0007307c     4 OBJECT  LOCAL  DEFAULT   23 _LIB_VERSION_INTERNAL
  2405: 00000000     0 NOTYPE  WEAK   DEFAULT  UND _ITM_deregisterTMCloneTable
  2680: 0007307c     4 OBJECT  WEAK   DEFAULT   23 _LIB_VERSION
  2698: 00000000     0 NOTYPE  WEAK   DEFAULT  UND _Jv_RegisterClasses
  2732: 00000000     0 NOTYPE  WEAK   DEFAULT  UND _ITM_registerTMCloneTable
```
(aka, no math symbols are defined that start with a single underscore)
BUT:
```
sean:/usr/src/px4/Firmware/Firmware.git/build_qurt_eagle_default/src/modules/local_position_estimator>cat /usr/src/px4/snapdragon/Qualcomm/HEXAGON_Tools/7.2.12/Tools/target/hexagon/include/{math,ymath}.h | sed -e 's/_CSTD/underscore_CSTD/g;s/_NO_THROW/underscore_NO_THROW/g;s/\([(,=][a-z ]*\) _/\1 underscore_/g' | grep -E '[[:space:]]_[A-Z].*;' | grep -E -v '^[[:space:]]' | sed -e 's/underscore_/_/g'
int _FFpcomp(float, float) _NO_THROW;
int _Fpcomp(double, double) _NO_THROW;
int _LFpcomp(long double, long double) _NO_THROW;
int _FDclass(float) _NO_THROW __attribute__ ((__const__));
int _Dclass(double) _NO_THROW __attribute__ ((__const__));
int _LDclass(long double) _NO_THROW __attribute__ ((__const__));
int _FDsign(float) _NO_THROW __attribute__ ((__const__));
int _Dsign(double) _NO_THROW __attribute__ ((__const__));
int _LDsign(long double) _NO_THROW __attribute__ ((__const__));
using _CSTD _FPCOMP; using _CSTD fpclassify;
void _Feraise(int);
double _Cosh(double, double);
short _Dtest(double *);
short _Exp(double *, double, long);
double _Log(double, int);
double _Sin(double, unsigned int);
double _Sinh(double, double);
float _FCosh(float, float);
short _FDtest(float *);
short _FExp(float *, float, long);
float _FLog(float, int);
float _FSin(float, unsigned int);
float _FSinh(float, float);
extern /* const */ _Dconst _FDenorm, _FInf, _FNan, _FSnan;
long double _LCosh(long double, long double);
short _LDtest(long double *);
short _LExp(long double *, long double, long);
long double _LLog(long double, int);
long double _LSin(long double, unsigned int);
long double _LSinh(long double, long double);
extern /* const */ _Dconst _LDenorm, _LInf, _LNan, _LSnan;
```
which seems to mean we don't have a working exp, log, sin, sinh, cos, cosh, signbit, isfinite, isinf, isnan, isnormal, isgreater, isgreaterequal, isless, islessequal, islessgreater, isunordered ...(?)

CarloWood, 2016-11-02:
That should fix this issue. I'm sorry for having caused this time waste for some people; I just assumed that if something linked (now) that apparently the hack to use __builtin_isfinite wasn't needed anymore, even though perhaps in the past it was. If my above assessment is correct then it is kinda shocking I must say... it's hard to believe that the hexagon SDK would be THIS broken :/. I'm confused.

venuannamraju, 2016-11-02:
@Carlo 
Thanks for the update. Will try it out and let you know.

venuannamraju, 2016-11-02:
@Carlo @julianoes 
I tried your repo and the linking issue is gone. Thanks for that.
I do get mag sensor failure like other have gotten before. The listener sensor-mag command does return non Zero values though. Did anyone see this error recently ? 
Also do ROMFS and posix-config need to be uploaded to the board ? The px4 binary seems to create a sym link in the path and a second init fails due to sym link failure. 

thanks 
venu
====================================
[08500/00]  04:19.118  HAP:61:In dspal_entry  0214  main.cpp
[08500/00]  04:19.126  HAP:61:attempting to open the ADSP command file: /dev/fs/px4.config  0178  main.cpp
[08500/00]  04:19.126  HAP:61:reading commands from /dev/fs/px4.config   0185  main.cpp
[08500/02]  04:19.130  HAP:61:Sleeping for 1, 1000000  0118  apps.h
[08500/00]  04:20.131  configuring UART for 4-wire mode, DAL id: 0x2001001  0852  DalUart.c
[08500/02]  04:20.135  HAP:61:Batch transfer failed: 0x110000c8  0524  i2c.c
[08500/00]  04:20.135  HAP:61:1016415 error: read register reports a read of -1 bytes, but attempted to set 1 bytes   0133  I2CDevObj.cpp
[08500/00]  04:20.135  HAP:61:1016457 error: unable to communicate with the hmc5883 mag sensor   0086  HMC5883.cpp
[08500/00]  04:20.135  HAP:61:1016490 error: mag sensor initialization failed, sensor read thread not started   0155  HMC5883.cpp
[08500/03]  04:20.135  HAP:61:HMC5883 start fail: -5  0165  df_hmc5883_wrapper.cpp
[08500/03]  04:20.135  HAP:61:DfHmc9250Wrapper start failed  0347  df_hmc5883_wrapper.cpp
[08500/00]  04:20.237  HAP:61:1119018 Reset MPU9250   0089  MPU9250.cpp
[08500/00]  04:20.301  HAP:60:1182210 IMU temperature initialized to: 52.418282   0516  MPU9250.cpp
[08500/00]  04:20.303  HAP:61:1184745 additional sensor configuration succeeded   0187  BMP280.cpp
[08500/03]  04:20.305  HAP:61:start  0204  df_trone_wrapper.cpp
[08500/02]  04:20.306  HAP:61:I2C_Write failed: 285212872 num_bytes_written 0  0414  i2c.c
[08500/00]  04:20.306  HAP:61:1187737 Error: i2c write failed. Reported -1 bytes written   0216  I2CDevObj.cpp
[08500/00]  04:20.306  HAP:61:1187773 Nonzero result when writing to TRONE_WHO_AM_I_REG 0x1   0110  TROne.cpp
[08500/00]  04:20.306  HAP:61:1187808 probing not successful   0071  TROne.cpp
[08500/03]  04:20.306  HAP:61:TROne start fail: -1  0134  df_trone_wrapper.cpp
[08500/03]  04:20.306  HAP:61:DfTROneWrapper start failed  0215  df_trone_wrapper.cpp
[08500/02]  04:20.323  HAP:61480:Batch transfer failed: 0x110000c8  0524  i2c.c
[08500/00]  04:20.323  HAP:61480:1204276 error: read register reports a read of -1 bytes, but attempted to set 1 bytes   0133  I2CDevObj.cpp
[08500/00]  04:20.323  HAP:61480:1204321 error: unable to communicate with the hmc5883 mag sensor   0086  HMC5883.cpp
[08500/00]  04:20.323  HAP:61480:1204357 error: mag sensor initialization failed, sensor read thread not started   0155  HMC5883.cpp
[08500/03]  04:20.323  HAP:61480:HMC5883 start fail: -5  0165  df_hmc5883_wrapper.cpp
[08500/02]  04:20.325  HAP:61480:Batch transfer failed: 0x110000c8  0524  i2c.c
[08500/00]  04:20.325  HAP:61480:1206921 error: read register reports a read of -1 bytes, but attempted to set 1 bytes   0133  I2CDevObj.cpp
[08500/00]  04:20.325  HAP:61480:1206962 error: unable to communicate with the hmc5883 mag sensor   0086  HMC5883.cpp
[08500/00]  04:20.325  HAP:61480:1206994 error: mag sensor initialization failed, sensor read thread not started   0155  HMC5883.cpp
[08500/03]  04:20.325  HAP:61480:HMC5883 start fail: -5  0165  df_hmc5883_wrapper.cpp
[08500/02]  04:20.328  HAP:39:LED: getHandle fail   0275  commander_helper.cpp
[08500/02]  04:20.328  HAP:39:LED init failed  1259  commander.cpp
[08500/02]  04:20.328  HAP:39:Buzzer: px4_open fail   0138  commander_helper.cpp
[08500/02]  04:20.328  HAP:39:Buzzer init failed  1263  commander.cpp
[08500/02]  04:20.331  HAP:39:Preflight checks always pass on Snapdragon.  0392  PreflightCheck.cpp
[08500/02]  04:20.397  HAP:61480:Batch transfer failed: 0x110000c8  0524  i2c.c
[08500/00]  04:20.397  HAP:61480:1278452 error: read register reports a read of -1 bytes, but attempted to set 1 bytes   0133  I2CDevObj.cpp
[08500/00]  04:20.397  HAP:61480:1278497 error: unable to communicate with the hmc5883 mag sensor   0086  HMC5883.cpp
[08500/00]  04:20.397  HAP:61480:1278537 error: mag sensor initialization failed, sensor read thread not started   0155  HMC5883.cpp
[08500/03]  04:20.397  HAP:61480:HMC5883 start fail: -5  0165  df_hmc5883_wrapper.cpp
[08500/00]  04:20.400  configuring UART for 4-wire mode, DAL id: 0x2001005  0852  DalUart.c
[08500/00]  04:20.401  HAP:34:Trying to initialize mixer from config file /dev/fs/quad_x.main.mix  0145  pwm_out_rc_in.cpp
[08500/00]  04:20.403  HAP:34:Successfully initialized mixer from config file  0157  pwm_out_rc_in.cpp
[08500/00]  04:21.980  HAP:4133:EKF using pressure height - commencing alignment  0407  ekf.cpp
[08500/00]  04:26.600  HAP:4133:EKF alignment complete  0064  control.cpp

[08500/03]  06:05.795  HAP:59:[get_data_bulk] Error: Semaphore is up when there is no data on the control/data queues  0455  uORBFastRpcChannel.cpp
[08500/02]  07:35.390  HAP:110:cannot find /oemconfig.so  0141  load.c
[08500/03]  07:35.390  HAP:111::error: -1: 0 == dynconfig_init(&conf, "security")   0696  sigverify.c
[08500/02]  07:35.395  mmpm_register: MMPM client for USM ADSP core 12  0117  UltrasoundStreamMgr_Mmpm.cpp
[... cut ...]
[08500/00]  07:35.400  loading the blsp config file failed, loading the default config  0262  main.c
[08500/02]  07:35.400  Loaded default UART-BAM mapping  0036  blsp_config.c
[08500/02]  07:35.400  UART tty-1: BAM-9  0044  blsp_config.c
[08500/02]  07:35.400  UART tty-2: BAM-6  0044  blsp_config.c
[08500/02]  07:35.400  UART tty-3: BAM-8  0044  blsp_config.c
[08500/02]  07:35.400  UART tty-4: BAM-2  0044  blsp_config.c
[08500/02]  07:35.400  UART tty-5: BAM N/A  0049  blsp_config.c
[08500/02]  07:35.400  UART tty-6: BAM N/A  0049  blsp_config.c

julianoes, 2016-11-02:
@venuannamraju I tested it and for me it now just crashes straightaway:

```
2010884: HAP:61:In dspal_entry (main.cpp:220)
   2012094: AvsCoreSvc: StartSvcHandler Enter (AdspCoreSvc.cpp:518)
   2012094: AdspCoreSvc: Started successfully (AdspCoreSvc.cpp:534)
   2012097: loading BLSP configuration (blsp_config.c:217)
   2012292: HAP:159:unable to open the specified file path (file.c:167)
   2012292: failed to open /usr/share/data/adsp/blsp.config (blsp_config.c:233)
   2012292: loading the blsp config file failed, loading the default config (main.c:261)
   2012292: Loaded default UART-BAM mapping (blsp_config.c:36)
   2012292: HAP:110:cannot find /oemconfig.so (load.c:141)
   2012292: HAP:111::error: -1: 0 == dynconfig_init(&conf, "security") (sigverify.c:696)
   2012293: HAP:80:cannot find /voiceproc_tx.so (load.c:141)
   2012293: HAP:80:cannot find /voiceproc_rx.so (load.c:141)
```

Need to digg further.

julianoes, 2016-11-02:
@CarloWood but good catch, the linking issue seems resolved!

venuannamraju, 2016-11-03:
@julianoes 
I also got the same errors but in a slightly different order.  I assumed it was OK since listener sensor_gyro worked and no errors were seen on the pxh>.  Is there a release version that I can use before I decide to pick up latest commits ?

thanks 
venu

julianoes, 2016-11-03:
Use the commit before #5715.

LorenzMeier, 2016-11-03:
@bharath374 Could you please help out with some of your resources here?

venuannamraju, 2016-11-03:
@julianoes @bharath374 
I used the commit d57e9f1 (before #5714), below is what I see in mini-dm. 
I am trying to get a working version on the board so I can explore other aspects. Any help is much appreciated.

[08500/00]  29:22.105  HAP:61:In dspal_entry  0222  main.cpp
[08500/00]  29:22.126  HAP:61:attempting to open the ADSP command file: /dev/fs/px4.config  0186  main.cpp
[08500/00]  29:22.127  HAP:61:reading commands from /dev/fs/px4.config   0193  main.cpp
[08500/02]  29:22.131  HAP:61:Sleeping for 1, 1000000  0118  apps.h
[08500/00]  29:23.132  configuring UART for 4-wire mode, DAL id: 0x2001001  0852  DalUart.c
[08500/02]  29:23.138  HAP:61:Batch transfer failed: 0x110000c8  0524  i2c.c
[08500/00]  29:23.138  HAP:61:1033719 error: read register reports a read of -1 bytes, but attempted to set 1 bytes   0133  I2CDevObj.cpp
[08500/00]  29:23.138  HAP:61:1033761 error: unable to communicate with the hmc5883 mag sensor   0086  HMC5883.cpp
[08500/00]  29:23.138  HAP:61:1033794 error: mag sensor initialization failed, sensor read thread not started   0155  HMC5883.cpp
[08500/03]  29:23.138  HAP:61:HMC5883 start fail: -5  0165  df_hmc5883_wrapper.cpp
[08500/03]  29:23.138  HAP:61:DfHmc9250Wrapper start failed  0347  df_hmc5883_wrapper.cpp
[08500/00]  29:23.240  HAP:61:1136193 Reset MPU9250   0089  MPU9250.cpp
[08500/00]  29:24.341  HAP:61:2237171 I2C transfer to mag timed out   0464  MPU9250_mag.cpp
[08500/00]  29:24.341  HAP:61:2237212 MPU9250 soft reset failed.   0082  MPU9250_mag.cpp
[08500/00]  29:24.341  HAP:61:2237408 mag initialization failed 1 tries   0202  MPU9250_mag.cpp
[08500/00]  29:24.406  HAP:60:2301728 IMU temperature initialized to: 54.191135   0516  MPU9250.cpp
[08500/00]  29:24.408  HAP:61:2303720 additional sensor configuration succeeded   0187  BMP280.cpp
[08500/03]  29:24.410  HAP:61:start  0204  df_trone_wrapper.cpp
[08500/02]  29:24.411  HAP:61:I2C_Write failed: 285212872 num_bytes_written 0  0414  i2c.c
[08500/00]  29:24.411  HAP:61:2306663 Error: i2c write failed. Reported -1 bytes written   0216  I2CDevObj.cpp
[08500/00]  29:24.411  HAP:61:2306700 Nonzero result when writing to TRONE_WHO_AM_I_REG 0x1   0110  TROne.cpp
[08500/00]  29:24.411  HAP:61:2306735 probing not successful   0071  TROne.cpp
[08500/03]  29:24.411  HAP:61:TROne start fail: -1  0134  df_trone_wrapper.cpp
[08500/03]  29:24.411  HAP:61:DfTROneWrapper start failed  0215  df_trone_wrapper.cpp
[08500/02]  29:24.418  HAP:28712:Batch transfer failed: 0x110000c8  0524  i2c.c
[08500/00]  29:24.418  HAP:28712:2314126 error: read register reports a read of -1 bytes, but attempted to set 1 bytes   0133  I2CDevObj.cpp
[08500/00]  29:24.418  HAP:28712:2314170 error: unable to communicate with the hmc5883 mag sensor   0086  HMC5883.cpp
[08500/00]  29:24.418  HAP:28712:2314205 error: mag sensor initialization failed, sensor read thread not started   0155  HMC5883.cpp
[08500/03]  29:24.418  HAP:28712:HMC5883 start fail: -5  0165  df_hmc5883_wrapper.cpp
[08500/02]  29:24.420  HAP:28712:Batch transfer failed: 0x110000c8  0524  i2c.c
[08500/00]  29:24.420  HAP:28712:2315875 error: read register reports a read of -1 bytes, but attempted to set 1 bytes   0133  I2CDevObj.cpp
[08500/00]  29:24.420  HAP:28712:2315917 error: unable to communicate with the hmc5883 mag sensor   0086  HMC5883.cpp
[08500/00]  29:24.420  HAP:28712:2315949 error: mag sensor initialization failed, sensor read thread not started   0155  HMC5883.cpp
[08500/03]  29:24.420  HAP:28712:HMC5883 start fail: -5  0165  df_hmc5883_wrapper.cpp
[08500/02]  29:24.421  HAP:12327:LED: getHandle fail   0269  commander_helper.cpp
[08500/02]  29:24.421  HAP:12327:LED init failed  1298  commander.cpp
[08500/02]  29:24.421  HAP:12327:Buzzer: px4_open fail   0132  commander_helper.cpp
[08500/02]  29:24.421  HAP:12327:Buzzer init failed  1302  commander.cpp
[08500/02]  29:24.423  HAP:12327:Preflight checks always pass on Snapdragon.  0392  PreflightCheck.cpp
[08500/00]  29:24.488  configuring UART for 4-wire mode, DAL id: 0x2001005  0852  DalUart.c
[08500/00]  29:24.488  HAP:34:Trying to initialize mixer from config file /dev/fs/quad_x.main.mix  0145  pwm_out_rc_in.cpp
[08500/00]  29:24.490  HAP:34:Successfully initialized mixer from config file  0157  pwm_out_rc_in.cpp
[08500/02]  29:24.490  HAP:28712:Batch transfer failed: 0x110000c8  0524  i2c.c
[08500/00]  29:24.490  HAP:28712:2385605 error: read register reports a read of -1 bytes, but attempted to set 1 bytes   0133  I2CDevObj.cpp
[08500/00]  29:24.490  HAP:28712:2385649 error: unable to communicate with the hmc5883 mag sensor   0086  HMC5883.cpp
[08500/00]  29:24.490  HAP:28712:2385683 error: mag sensor initialization failed, sensor read thread not started   0155  HMC5883.cpp
[08500/03]  29:24.490  HAP:28712:HMC5883 start fail: -5  0165  df_hmc5883_wrapper.cpp
[08500/00]  29:26.061  HAP:37:EKF using pressure height - commencing alignment  0409  ekf.cpp
[08500/00]  29:30.801  HAP:37:EKF alignment complete  0064  control.cpp

julianoes, 2016-11-03:
@venuannamraju This looks mostly "fine". Is anything not working in particular? And if yes, could you please open a new issue. This is really about the linking not working (or whatever error it is now).

venuannamraju, 2016-11-03:
@julianoes 
I saw the errors were similar to your logs where you reported the crash, so I pasted my output. If you think the output seems fine, we can close it. 
thanks
venu

CarloWood, 2016-11-03:
@julianoes Can you test which commit is causing your current/remaining issue?
I'd suggest to
```
git show baf89f4398ae9bc8bfdf6c0761aaba0667ba9db5 > apph-1.diff
git show 52b8e75aeadd59e0f12f3de3f9e5759ead7f592b > apph-2.diff
git show 0bff3593d3ecc766f7c8981a3259d10a9110833d > apph-3.diff

git show 90f3e3b5d322a09a8dfbed6a3c7ce48da271bfd7 > decls-1.diff
git show e19d2f80b05b30b73e500e6c3d4718a2ab86c5d4 > decls-2.diff
```
where the last one is really https://github.com/PX4/Firmware/pull/5796/commits/e19d2f80b05b30b73e500e6c3d4718a2ab86c5d4

Then checkout latest master, and remove all five commits (decls-2.diff is not yet in there):
```
patch -R -p1 < decls-1.diff
patch -R -p1 < apph-3.diff
patch -R -p1 < apph-2.diff
patch -R -p1 < apph-1.diff
```
Test if the problem exists [1]. If not,
```
patch -p1 < apph-1.diff
patch -p1 < apph-2.diff
patch -p1 < apph-3.diff
```
Test if problem exists. If not,
```
patch -p1 < decls-1.diff
patch -p1 < decls-2.diff
```
Test if problem exists (it should cause you already said so).

If at [1] the problem already exists then it is caused by another commit than mine.
In that case I'd add decls-2.diff (as you already did before) and then make a new
branch and rebase it:
```
git checkout master
git checkout -b testbranch
patch -p1 < decls-2.diff
git commit -a -m 'decls-2'
git rebase -i --onto e29b9b5d39 e29b9b5d39 testbranch
```
where you squash 52b8e75ae and 0bff3593d3 into baf89f4398ae and
the 'decls-2' commit into 90f3e3b5d322.
Then do a git bisect:
```
git bisect start testbranch e29b9b5d39
```
etc

CarloWood, 2016-11-03:
PS I found _Dtest back: it's defined in their libc.so... Apparently we're not linking with -lc or with the wrong one thus...

jywilson, 2016-11-03:
Hey guys, I would like to help with this problem.  Is there an easy way that I can repro your results?  Do you have a branch that I can sync to, just to save time?

I have a JTAG that I can use to get a call stack at the point of the crash.  For crash debugging mini-dm is problematic because it drops messages that were in the buffer at the time of the crash.

CarloWood, 2016-11-04:
@jywilson I'm afraid @julianoes has to answer that.

jywilson, 2016-11-04:
Okay.  

@julianoes If you could help with the above, I would appreciate it.

julianoes, 2016-11-05:
@jywilson It should be easy. Get latest master, do `make eagle_default upload`, run it and watch it crash. That's at least what happens for me.

It didn't crash before the refactoring which started with #5715 which lead to linking errors. The linking errors were resolved in later PR's by @CarloWood, however, it's seems now to crash on QURT.

julianoes, 2016-11-05:
@jywilson I tried to manually get a "stack trace" by printfing to find where it crashes. I found the that it crashes in `param set` on QURT as follows:

https://github.com/PX4/Firmware/blob/master/src/systemcmds/param/param.c#L146
https://github.com/PX4/Firmware/blob/master/src/systemcmds/param/param.c#L480
https://github.com/PX4/Firmware/blob/master/src/modules/systemlib/param/param_shmem.c#L508
https://github.com/PX4/Firmware/blob/master/src/modules/systemlib/param/param_shmem.c#L520
https://github.com/PX4/Firmware/blob/master/src/platforms/qurt/px4_layer/shmem_qurt.c#L337
https://github.com/PX4/Firmware/blob/master/src/platforms/qurt/px4_layer/shmem_qurt.c#L260

At that point `shmem_info_p` is NULL and it segfaults. However, `shmem_info_p` is previously initialized to some value here: https://github.com/PX4/Firmware/blob/master/src/platforms/qurt/px4_layer/shmem_qurt.c#L149

I don't understand why `shmem_info_p` ends up being NULL.

jywilson, 2016-11-06:
Got it.  Thanks.  I will give it a try in Monday and let you know what I learn.

CarloWood, 2016-11-06:
@jywilson Have you done a git bisect? If not, lemme update my post now that 09d36a6 was merged into master:
```
git show baf89f4398ae9bc8bfdf6c0761aaba0667ba9db5 > apph-1.diff
git show 52b8e75aeadd59e0f12f3de3f9e5759ead7f592b > apph-2.diff
git show 0bff3593d3ecc766f7c8981a3259d10a9110833d > apph-3.diff
git show 09d36a63ef38f0aa4501998f01ae8faf118f6bd6 > apph-4.diff

git show 90f3e3b5d322a09a8dfbed6a3c7ce48da271bfd7 > decls-1.diff
```
Then checkout latest master, and remove all five commits:
```
patch -R -p1 < decls-1.diff
patch -R -p1 < apph-4.diff
patch -R -p1 < apph-3.diff
patch -R -p1 < apph-2.diff
patch -R -p1 < apph-1.diff
```
Test if the problem exists [1]. If not,
```
patch -p1 < apph-1.diff
patch -p1 < apph-2.diff
patch -p1 < apph-3.diff
patch -p1 < apph-4.diff
```
Test if problem exists. If not,
```
patch -p1 < decls-1.diff
```
Test if problem exists (it should cause you already said so).

If at [1] the problem already exists then it is caused by another commit than mine.
In that case I'd make a new branch and rebase it:
```
git checkout master
git reset --hard
git checkout -b testbranch
git rebase -i --onto e29b9b5d39 e29b9b5d39 testbranch
```
where you squash 52b8e75ae, 0bff3593d3 and 09d36a6 into baf89f4398ae
(move those three commits/lines directly below the top line (baf89f4)
and change the first word from pick into squash).

Then do a git bisect:
```
git bisect start testbranch e29b9b5d39
```
which marks testbranch as 'bad' and `e29b9b5d39`as good and then
checks out the next commit to test. 

So just 'make clean; make submodulesclean; make eagle_default upload'
and test. When it crashes type 'git bisect bad', when it doesn't crash 'git bisect good'
and go back again to 'make clean; ...' etc. until the bisect ends.

julianoes, 2016-11-06:
@CarloWood thanks for the recipe. I've actually done similar checks by going back to the commit before your changes and then cherry-picking them one-by-one. However, I didn't always clean build there, so I can't be certain about the results.

I've followed the steps and it crashes after flashing the state after:
```
patch -p1 < apph-4.diff
```

CarloWood, 2016-11-06:
Did a few hours session on mumble with Julian and found out the following:

- The pointer being set (shmem_info_p) and the one being used later on where actually different variables (had a different) address.
- The reason for that is because shmem_qurt.c is linked in twice: this file ends up in ```build_qurt_eagle_default/src/platforms/qurt/px4_layer/libplatforms__qurt__px4_layer.a``` which is added to ```build_qurt_eagle_default/src/firmware/qurt/libpx4.so``` but *also* erroneously to ```build_qurt_eagle_default/src/firmware/qurt/libpx4muorb_skel.so```!

The actual issue is WAY WAY worse, not only is shmem_info_p duplicated, all static and global variables in all 'modules' that are added to libpx4.so are duplicated:
```
"drivers__device;modules__sensors;platforms__posix__drivers__df_mpu9250_wrapper;platforms__posix__drivers__df_bmp280_wrapper;platforms__posix__drivers__df_hmc5883_wrapper;platforms__posix__drivers__df_trone_wrapper;platforms__posix__drivers__df_isl29501_wrapper;systemcmds__param;modules__attitude_estimator_q;modules__position_estimator_inav;modules__local_position_estimator;modules__ekf2;modules__mc_att_control;modules__mc_pos_control;modules__param;modules__systemlib;modules__systemlib__mixer;modules__uORB;modules__commander;modules__land_detector;drivers__gps;drivers__pwm_out_rc_in;drivers__qshell__qurt;lib__controllib;lib__mathlib;lib__mathlib__math__filter;lib__geo;lib__ecl;lib__geo_lookup;lib__conversion;lib__terrain_estimation;lib__runway_takeoff;lib__tailsitter_recovery;lib__DriverFramework__framework;platforms__common;platforms__qurt__px4_layer;platforms__posix__work_queue;modules__muorb__adsp;df_mpu9250;df_bmp280;df_hmc5883;df_trone;df_isl29501"
```
being the list of object archives (.a) that are added to both, libpx4.so and libpx4muorb_skel.so

We found back the underlaying problem in a question on stackoverflow:

http://stackoverflow.com/questions/31503597/cmake-link-a-shared-library-to-static-libraries

I think this is the real problem: you cannot add STATIC libraries to SHARED libraries the way it is currently done in the project.

julianoes, 2016-11-06:
As a note: the debug `shmem_info_p` can be debugged like this: https://github.com/julianoes/Firmware/commit/cdaf3866cc06222a14fde54c707329bca8020bf7

CarloWood, 2016-11-06:
Hi, I created a small example that shows how I think this should work, here:
https://github.com/CarloWood/cmaketest

CarloWood, 2016-11-06:
What you think, shouldn't we start using this method for Firmware too?

julianoes, 2016-11-10:
@CarloWood @mcharleb @jywilson the fix/workaround has gone in, thanks everyone!

@CarloWood Feel free to make a PR to improve the overall linking scheme. I'm afraid, I don't have time to look into it at the moment.

PR #5822 (2016-11-10): cmake_hexagon: rpcmem and linking fix
This updates cmake_hexagon and cmake_hexagon inside DriverFramework
which fixes somethine about rpcmem.a and the double linking issue.

Finally closes #5756.
closed 2016-11-10 by julianoes