PX4/PX4-Autopilot #5757: macosx dependency on Xcode 8

Ole2mail, 2016-10-26:
hello all.
just managed to compile default make for Firmware having Xcode dependency fixed to 6.

any ideas why do we strict to Xcode 8 (the one involving to have AppStore etc?
[xcode6.patch.txt](https://github.com/PX4/Firmware/files/553622/xcode6.patch.txt)

LorenzMeier, 2016-10-29:
Can you share the output of ```make posix jmavsim```?

micheda, 2016-12-22:
Hi Lorenz,

same situation here and the patch worked, output of `make posix jmavsim`:
[typescript.txt](https://github.com/PX4/Firmware/files/669130/typescript.txt)

julianoes, 2017-04-18:
The tests are based on XCode 8 and we don't have the bandwidth to support anything else. Thanks for the patch in case someone else runs in the same issue.

closed 2017-04-18 by julianoes