PX4/PX4-Autopilot #5759: Enable use of dual GPS with EKF2

priseborough, 2016-10-26:
Replaces https://github.com/PX4/ecl/issues/173

Support from dual GPS is required for vehicle that experience large orientation changes (eg tail-sitters) or are large enough such that masking requires use of dual GPS units.

This will need to be implemented in the EKF2 module, preferably in a function called from Ekf2::task_main(), which then provides consolidated GPS data to be used by the ecl library.

It will require access to a complete set of GPS receiver solution status information including horizontal accuracy, vertical accuracy, speed accuracy, number of satellites. The offset between receiver position data will need to be tracked an compensated for to enable a switchover without significant position jumps.

dagar, 2018-01-17:
We could run GPS through sensors. Each unit could publish a new sensor_gps message (with offsets), the sensors module (or gps library) could consolidate and publish an overall vehicle_gps_position (or just vehicle_gps?).

dagar, 2019-01-30:
Implemented.

closed 2019-01-30 by dagar