PX4/PX4-Autopilot #5760: Mavros offboard example mistake?

zhahaoyu, 2016-10-26:
I was learning how to write ROS nodes to interact with mavros. On the example [http://dev.px4.io/ros-mavros-offboard.html](url), it says

```
// wait for FCU connection
  while(ros::ok() && current_state.connected){
        ros::spinOnce();
        rate.sleep();
    }
```

Is it `!current_state.connected` instead of `current_state.connected`?

In addition, I am not sure why we want to send a few instructions before the vehicle is armed. Is it because there can be a lag between sending the signal and receiving the signal on the FCU?

```
    //send a few setpoints before starting
    for(int i = 100; ros::ok() && i > 0; --i){
        local_pos_pub.publish(pose);
        ros::spinOnce();
        rate.sleep();
    }
```

huanglilong, 2016-10-28:
please read it carefully, [mavros-offboard](http://dev.px4.io/ros-mavros-offboard.html), 

The px4 flight stack has a timeout of 500ms between two offboard commands. If this timeout is exceeded, the commander will fall back to the last mode the vehicle was in before entering offboard mode. This is why the publishing rate must be faster than 2 Hz to also account for possible latencies. This is also the same reason why it is recommended to enter offboard mode from POSCTL mode, this way if the vehicle drops out of offboard mode it will stop in its tracks and hover.

zhahaoyu, 2016-10-29:
Thanks @huanglilong. So my understanding is since the px4 flight stack has a timeout of 500ms, we would like to prepare a bunch of commands in its task queue, otherwise, if the the FCU times out, it will f
 the FCU times out, it will leave the OFFBOARD mode.

huanglilong, 2016-10-29:
"In addition, I am not sure why we want to send a few instructions before the vehicle is armed. Is it because there can be a lag between sending the signal and receiving the signal on the FCU?"
----
before switch to OFFBOARD mode, you should send a bunch of commands to FCU, if not, FCU(pixhawk) will reject switching to OFFBOARD mode. it has nothing to do with delay.

huanglilong, 2016-10-29:
[offnode](https://github.com/huanglilong/mavros/blob/master/mavros/src/offb_node.cpp), i have tested it with gazebo and pixhawk

zhahaoyu, 2016-10-29:
@huanglilong In the cpp file you provided, I don't think you ever send the service requests.

huanglilong, 2016-10-30:
you can try this [package](https://github.com/huanglilong/mission)