PX4/PX4-Autopilot #5766: logger problems on fmu-v4 with latest master 5b52cd0fe6

kd0aij, 2016-10-28:
This is the console output for 2 successive arm/disarm sequences (via RC):
```
nsh> 
nsh> INFO  [logger] Start file log
INFO  [logger] Opened log file: /fs/microsd/log/sess001/log001.ulg
INFO  [logger] closed logfile, bytes written: 372848
INFO  [logger] Start file log
ERROR [logger] Can't open log file /fs/microsd/log/sess001/log002.ulg
```
Result is that only the first arm/disarm segment is logged. Must reboot in order to recover.
logger on/off works properly, creating sequentially numbered files within the same session folder.

sdlog2 never opens a logfile when armed via RC, or QGC or commander.
sdlog2 on/off works correctly also.

LorenzMeier, 2016-10-28:
@bkueng Can you look into that?

bkueng, 2016-10-28:
@kd0aij I cannot reproduce this. When arming/disarming via RC several times, logger always successfully starts/stops. The same with sdlog2 then started with `-a`.

I know the symptoms don't look like it, but did you try another SD card or reformatting?

kd0aij, 2016-10-28:
Yes, I tried a second SD card. And both cards have been fine in previous flights.
I first encountered the problem flight testing the pr-mixerEnhancements branch, and reproduced it on the bench in that branch. Then I fetched upstream, checked out/pulled master, "make clean" then "make px4fmu-v4 upload" to generate the results filed in this issue.

Did I forget something? There must be something wrong with my builds.

kd0aij, 2016-10-28:
@bkueng I'm also having trouble with mixing on the FMU in https://github.com/PX4/Firmware/pull/5720
I'd appreciate it if you could try that PR on fmu-v2 to see if the new AUX_TRIM parameters have any effect on the aux pwm outputs. In my testing, although the new MAIN_TRIM parameters work as expected for the main pwm outputs, on the IO coprocessor, the aux outputs do not. Perhaps the two problems are somehow related.

kd0aij, 2016-10-28:
@bkueng I just used QGC to flash master onto the same pixracer, and again reproduced the problem on the bench with a third SD card. That rules out my local build (gcc 4.9.3) as the cause.

bkueng, 2016-10-28:
Sure, I'll look into it when I find the time. Can you try to find out why logger fails to open the file (print the errno)? Maybe it runs out of file descriptors? Do you use a custom topics list?

kd0aij, 2016-10-28:
not using a custom file list. errno=24 which is "too many open files" according to nuttx/include/errno.h
I would have though regression testing is done using the default (all?) topics.

LorenzMeier, 2016-10-28:
It is totally acceptable to not be able to log all topics on a resource constrained system. So I would like to see this fixed by closing stdin right before opening the log file. The logger should also output a MAVLink message that it was not able to log all topics.

kd0aij, 2016-10-28:
@LorenzMeier I don't understand why @bkueng couldn't reproduce the problem on fmu-v2 though.
Is there a particular reference platform (e.g. fmu-v2) that default configs are tested against?

LorenzMeier, 2016-10-28:
Just to clarify: closing stdin will free the file descriptor needed for the log file

kd0aij, 2016-10-29:
@bkueng I just verified by bumping CONFIG_NFILE_DESCRIPTORS up to 52, that the new logger is failing to release the descriptor used for writing the logfile. With that change, it succeeds in opening and writing two logs, then fails as before:
```
INFO  [logger] Start file log
INFO  [commander] Takeoff detected
INFO  [logger] Opened log file: /fs/microsd/log/sess001/log001.ulg
INFO  [logger] closed logfile, bytes written: 253952
INFO  [commander] Landing detected
INFO  [logger] Start file log
INFO  [logger] Opened log file: /fs/microsd/log/sess001/log002.ulg
INFO  [logger] closed logfile, bytes written: 287199
INFO  [logger] Start file log
ERROR [logger] Can't open log file /fs/microsd/log/sess001/log003.ulg, errno: 24
```
I can't see any obvious differences between the way sdlog2 closes the file and the way the new logger does it, but obviously something is different. Is there any relationship between these files and "virtual" files? Logging fewer topics is also a usable workaround...

bkueng, 2016-10-31:
I was able to reproduce, w/o any HW attached. We were leaking a file descriptor, which is why it worked the first N times before it failed.
@kd0aij can you test #5783?

closed 2016-10-31 by LorenzMeier

kd0aij, 2016-10-31:
verified in latest master on fmu-v4

PR #5783 (2016-10-31): fix logger: avoid leaking file descriptor in get_log_time()
orb_subscribe can succeed, but if there is no publisher, orb_copy will fail.
We still need to unsubscribe in that case.

fixes #5766