PX4/PX4-Autopilot #5768: Refuse connection to MissionPlanner as long as it has no PX4 support

LorenzMeier, 2016-10-28:
Mission planner implements the parameter protocol not according to the MAVLink reference implementation (it was never cross-tested to QGC, which was the reference at the time) which can mess up tuning gains when people try to use it.

We should actively refuse the connection to ensure that PX4 users don't end up with messed up params and a really bad user experience.

The refusal message should be kind though, just hinting that its not supported yet. In an optimal scenario Mission Planner would add support to PX4 and we could add it as GCS option.

squilter, 2016-10-28:
I think the only proper way to implement this is to have Mission Planner reject the connection. Otherwise, if Mission Planner does add proper support, PX4 would still be rejecting the connection. Even if PX4 is then updated, it still means that older versions of PX4 won't work with Mission Planner because of an artificial restriction.

LorenzMeier, 2016-10-28:
Good point. I don't want to close the door on that option. I'll talk to Michael to see if he can either add for the param interface (AFAIK the only major problem) or refuse the connection as a stopgap.

LorenzMeier, 2016-10-28:
Closing this accordingly.

closed 2016-10-28 by LorenzMeier