PX4/PX4-Autopilot #5771: get() doesn't fetch a new param value if previous value was 0 unless rebooted

mswingtra, 2016-10-28:
I have spotted it in the navigator, but I guess it will apply to all the places where `get()` function is used to fetch parameter values.

If param value is non-zero and it is changed via QGC, then it works fine. However, once it is set to 0, and then it is changed, the new value is no longer fetched until the system is rebooted.

LorenzMeier, 2016-10-29:
Quick code inspection suggests no issues - @bkueng Can you look into this?

bkueng, 2016-10-31:
I cannot reproduce, looks fine to me, except for an excessive param update in [0], which is called regularily. I'm not sure why this was added, but I don't think it's needed.

[0] https://github.com/PX4/Firmware/blob/master/src/modules/navigator/geofence.cpp#L104

mswingtra, 2016-10-31:
@bkueng As an example, I am printing out the value of `_geofence.getGeofenceAction()` to the console with PX4_WARN on the current master (93fb02bfa38b6a337e0ec3984e42485e0c00fc82) in SITL - it shows the value set in QGC (v.3.0.2) until I set it to 0. After that, if I change it to non-zero value, it does not update on the px4 side anymore.

dagar, 2016-10-31:
@mswingtra ah that's actually a geofence bug. It's only updating geofence params in _geofence.inside(), which is no longer called after you set it to 0 (none).
https://github.com/PX4/Firmware/blob/master/src/modules/navigator/navigator_main.cpp#L503

mswingtra, 2016-10-31:
@dagar No, no - I am printing out directly the result of the function `_geofence.getGeofenceAction()` and I'm calling it in the main loop. This function only [gets the value of the set parameter](https://github.com/PX4/Firmware/blob/master/src/modules/navigator/geofence.h#L112) and this is not working for me.

dagar, 2016-10-31:
Once you set that particular geofence param to 0 the geofence class no longer updates param values. https://github.com/PX4/Firmware/blob/master/src/modules/navigator/geofence.cpp#L104

mswingtra, 2016-10-31:
Ah, I see now -  thanks for pointing this out.

bkueng, 2016-11-01:
@dagar If we initialize the Superblock base of Geofence in [0] with navigator instead of null, the `updateParams` call becomes unnecessary, and it will automatically be called from the navigator.

[0] https://github.com/PX4/Firmware/blob/master/src/modules/navigator/geofence.cpp#L61

mswingtra, 2016-11-02:
@bkueng I looked into the block library, and I am not sure if I understand it fully, but if we do the change you proposed, wouldn't we have to call `updateChildParams` in navigator as navigator would be regarded as a parent to geofence? Then that would update params for all the navigation modules which is not desired here.

bkueng, 2016-11-02:
No, as the `updateParams` call in Navigator calls `SuperBlock::updateParams`, which in turn calls `updateChildParams` if there are children. From what I see, this is the intended way how to used it.

LorenzMeier, 2016-11-02:
If you do that the child will also inherit the parameter prefix of the parent unless paeans are initialized with the full name constructor. Watch for that.

LorenzMeier, 2016-11-06:
@mswingtra I just fixed a param system issue in another PR. I recommend to re-test on master.

PR #5812 (2016-11-06): Geofence: Param update fix
@LorenzMeier This fixes #5771.
closed 2016-11-06 by dagar