PX4/PX4-Autopilot #5773: Error with make posix_sitl_default gazebo with gazebo7

Jaeyoung-Lim, 2016-10-28:
I have been using gazebo6 with ROS indigo(on Ubuntu 14.04) to run the PX4 plugin and now changing to gazebo7 with ROS kinetic (on Ubuntu 16.04)

I have followed the instructions and the
```
make posix_sitl_default jmavsim
```
works just fine. However, I cannot make the plugin run with gazebo7.
Below is the error when I am trying to run the px4 plugin with gazebo

The [document](http://dev.px4.io/simulation-ros-interface.html) states that the SITL does not work in Gazebo7 but I see some posts that it does. (and the documentation is now mostly based on gazebo7)
I am not sure how I should understand the support for gazebo7. 
Moreover, I cannot find the packages through apt-get

```
PX4 CONFIG: /home/cosmonaut/src/Firmware/build_posix_sitl_default
Scanning dependencies of target sitl_gazebo
[  0%] Built target prebuild_targets
[  0%] Creating directories for 'sitl_gazebo'
[  0%] Built target git_gencpp
[  0%] Built target git_genmsg
[  0%] Built target git_mavlink
[  0%] Built target git_uavcan
[  1%] Built target git_matrix
[  1%] Built target git_jmavsim
[  1%] Built target git_driverframework
[  3%] Built target df_driver_framework
[  4%] No download step for 'sitl_gazebo'
[  5%] Built target git_gazebo
[  5%] Built target git_ecl
[  6%] Built target xml_gen
[  6%] Built target mixer_gen
[  6%] No patch step for 'sitl_gazebo'
[  6%] No update step for 'sitl_gazebo'
[  6%] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr
[ 26%] Built target msg_gen
[ 26%] Built target platforms__common
[ 26%] Built target generate_topic_listener
-- The C compiler identification is GNU 5.4.0
[ 26%] Built target drivers__boards__sitl
[ 28%] Built target drivers__device
[ 29%] Built target drivers__pwm_out_sim
[ 32%] Built target drivers__gps
[ 37%] Built target systemcmds__tests
[ 38%] Built target platforms__posix__drivers__adcsim
[ 38%] Built target platforms__posix__drivers__accelsim
[ 40%] Built target drivers__vmount
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
[ 40%] Built target platforms__posix__drivers__gpssim
[ 41%] Built target platforms__posix__drivers__barosim
[ 41%] Built target platforms__posix__drivers__airspeedsim
[ 41%] Built target platforms__posix__drivers__rgbledsim
[ 42%] Built target platforms__posix__drivers__ledsim
[ 42%] Built target platforms__posix__drivers__gyrosim
[ 43%] Built target platforms__posix__px4_layer
[ 44%] Built target platforms__posix__drivers__tonealrmsim
[ 47%] Built target platforms__posix__work_queue
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
[ 47%] Built target systemcmds__esc_calib
[ 47%] Built target systemcmds__mixer
[ 47%] Built target systemcmds__param
[ 48%] Built target systemcmds__perf
[ 48%] Built target systemcmds__reboot
[ 48%] Built target systemcmds__sd_bench
[ 49%] Built target systemcmds__topic_listener
[ 49%] Built target systemcmds__top
[ 49%] Built target systemcmds__ver
[ 50%] Built target systemcmds__motor_ramp
[... 186 lines cut ...]
                 from /home/cosmonaut/src/Firmware/Tools/sitl_gazebo/src/gazebo_controller_interface.cpp:22:
/usr/include/gazebo-7/gazebo/msgs/msgs.hh:24:37: fatal error: ignition/math/Inertial.hh: No such file or directory
compilation terminated.
CMakeFiles/rotors_gazebo_mavlink_interface.dir/build.make:62: recipe for target 'CMakeFiles/rotors_gazebo_mavlink_interface.dir/src/gazebo_mavlink_interface.cpp.o' failed
CMakeFiles/rotors_gazebo_controller_interface.dir/build.make:62: recipe for target 'CMakeFiles/rotors_gazebo_controller_interface.dir/src/gazebo_controller_interface.cpp.o' failed
make[7]: *** [CMakeFiles/rotors_gazebo_mavlink_interface.dir/src/gazebo_mavlink_interface.cpp.o] Error 1
make[7]: *** [CMakeFiles/rotors_gazebo_controller_interface.dir/src/gazebo_controller_interface.cpp.o] Error 1
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/rotors_gazebo_controller_interface.dir/all' failed
make[6]: *** [CMakeFiles/rotors_gazebo_controller_interface.dir/all] Error 2
CMakeFiles/Makefile2:141: recipe for target 'CMakeFiles/rotors_gazebo_mavlink_interface.dir/all' failed
make[6]: *** [CMakeFiles/rotors_gazebo_mavlink_interface.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make[5]: *** [all] Error 2
src/firmware/posix/CMakeFiles/sitl_gazebo.dir/build.make:110: recipe for target 'external/Stamp/sitl_gazebo/sitl_gazebo-build' failed
make[4]: *** [external/Stamp/sitl_gazebo/sitl_gazebo-build] Error 2
CMakeFiles/Makefile2:9137: recipe for target 'src/firmware/posix/CMakeFiles/sitl_gazebo.dir/all' failed
make[3]: *** [src/firmware/posix/CMakeFiles/sitl_gazebo.dir/all] Error 2
CMakeFiles/Makefile2:10011: recipe for target 'src/firmware/posix/CMakeFiles/gazebo.dir/rule' failed
make[2]: *** [src/firmware/posix/CMakeFiles/gazebo.dir/rule] Error 2
Makefile:3510: recipe for target 'gazebo' failed
make[1]: *** [gazebo] Error 2
Makefile:160: recipe for target 'posix_sitl_default' failed
make: *** [posix_sitl_default] Error 2

```

eric1221bday, 2016-10-28:
I know that gazebo 7 can build in Ubuntu 14.04 with ROS Indigo, so I suspect, as it looks like it's a package dependency issue, ignition, a package gazebo depends seems to have been changed in 16.04 resulting in the error

Jaeyoung-Lim, 2016-10-28:
@eric1221bday 

But it is weird because gazebo7 runs just fine on its own. The error occurs only when trying to run SITL with gazebo

zhahaoyu, 2016-10-29:
@Jaeyoung-Lim I suspect there is some dependency issue on these packages. I run ROS kinetic, Gazebo 7, everything works fine. It might be worth reinstalling/updating everything, and give a try then.

julianoes, 2016-11-01:
This looks like something with the dependency to ignition is wrong. Check what you have installed there.

nikkk91, 2016-11-24:
Do you know the solution to solve the problem of ignition/math/Inertial.hh? i still have the same problem.

julianoes, 2016-11-24:
What ignition stuff do you have installed? It's some version mismatch there.

rikba, 2016-12-01:
Did you try ```sudo apt-get install libignition-math2-dev```?

hamid-m, 2016-12-21:
Same issue here with Ubuntu 16.04.  @rikba I tried that command and the package is already installed:
> libignition-math2-dev is already the newest version (2.2.3+dfsg1-4)
@Jaeyoung-Lim , did you find a resolution?

camciefe, 2017-02-15:
@Jaeyoung-Lim @nikkk91 @hamid-m 

When you get the error message, just type:
     killall gzserver
Then, build your 'sitl_gazebo package' once again from the beginning.

closed 2017-04-03 by mhkabir

zubair123khan, 2018-01-19:
hi , 
i am getting the same error , can you please tell specifically whats going on wrong ?? i have ros - kinetic

julianoes, 2018-01-22:
Please make a new issue and add compile output, installed versions, etc.

mrabbah, 2018-02-25:
I got the same error here, ubuntu 16.04, ros kinetic desktop full, gazebo 7, I followed this link : https://github.com/disaster-robotics-proalertas/usv_sim_lsa and when I run : catkin_make_isolated --install <br/>
I got : 
In file included from /usr/include/gazebo-7/gazebo/physics/Entity.hh:30:0,
                 from /usr/include/gazebo-7/gazebo/physics/Model.hh:34,
                 from /usr/include/gazebo-7/gazebo/physics/Actor.hh:24,
                 from /usr/include/gazebo-7/gazebo/physics/physics.hh:2,
                 from /home/sanji/usv_sim_lsa/src/keel_plugin/src/keel_plugin.cpp:5:
/usr/include/gazebo-7/gazebo/msgs/msgs.hh:24:37: fatal error: ignition/math/Inertial.hh: No such file or directory
compilation terminated.
CMakeFiles/keel_plugin.dir/build.make:62: recipe for target 'CMakeFiles/keel_plugin.dir/src/keel_plugin.cpp.o' failed
make[2]: *** [CMakeFiles/keel_plugin.dir/src/keel_plugin.cpp.o] Error 1
CMakeFiles/Makefile2:867: recipe for target 'CMakeFiles/keel_plugin.dir/all' failed
make[1]: *** [CMakeFiles/keel_plugin.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
<== Failed to process package 'keel_plugin': 
  Command '['make', '-j8', '-l8']' returned non-zero exit status 2

Reproduce this error by running:
==> cd /home/sanji/usv_sim_lsa/build_isolated/keel_plugin && make -j8 -l8

Command failed, exiting.

julianoes, 2018-02-26:
@mrabbah this is an old thread, please open a new one where appropriate, e.g. in https://github.com/disaster-robotics-proalertas/usv_sim_lsa if that's where you have the instructions from.

RealityVoid, 2018-04-09:
Just as an FYI, I hit the same error today, SITL worked with Gazebo9 but when installing Gazebo7 it croaked when building. 

The fix was installing  libignition math 2
sudo apt-get install libignition-math2-dev

That made SITL work again with Gazebo7.

HajimeK, 2018-10-09:
Solved with
sudo apt-get install libignition-math2-dev