PX4/PX4-Autopilot #5776: Not able to control copter using setpoint_velocity/cmd_vel

L4ncelot, 2016-10-29:
Hi,

I've been trying to control quadcopter with **px4** firmware with **sitl**, **gazebo** and **ROS**. I've successfully started simulation and established right connection with **mavros**. I can control quadcopter either by using **pxh command line** or sending poses to **mavros/setpoint_position/local** topic. This works fine.

However when I wanted to try controlling quadcopter with **cmd_vel** message types, I couldn't do such a thing. For this I'm using my **joystick** and **teleop** from **mavros_extras** pkg. When looking at topics and messages I can see that my joystick commands transfer into **cmd_vel** messages which are then published to **mavros/setpoint_velocity/cmd_vel** topic which is fine. But then in Gazebo the model is doing nothing and I don't know where the problem is.

I've also tried to set other flight modes (e.g. **OFFBOARD**, **MANUAL** etc.) but this also doesn't seem to work. 

Am I missing something? Or is there some velocity control functionality at all?

Thank you for any advice.

L4ncelot, 2016-11-01:
Ok, so after some struggle I've figured it out. I didn't know I had to actually **arm** the vehicle using `rosrun mavros mavsafety arm` and operate in `OFFBOARD` mode. Now the `teleop` works fine.

closed 2016-11-01 by L4ncelot

vinhk, 2017-10-19:
@L4ncelot did this work on actual quadcopter?

L4ncelot, 2017-10-20:
@vinhk I haven't tested this on actual copter. I just wanted it to work in simulation. But I think it should work on actual one as well since it's just velocity_cmd publisher and nothing else.