PX4/PX4-Autopilot #5787: Throttle scaled differently in SIL and HIL

evanmw, 2016-10-31:
Throttle is treated like all other outputs when scaled for HIL_ACTUATOR_CONTROLS messages in SIL (see Simulator::pack_actuator_message() in src/modules/simulator/simulator_mavlink.cpp).

Throttle is scaled 0 to 1 instead of -1 to 1 when publishing messages in HIL mode (see MavlinkStreamHILActuatorControls::send() in src/mavlink/mavlink_messages.cpp).

A fix would be to remove the special scaling for throttle when in HIL mode.  Ensuring scaling is -1 to 1 should happen by clamping the output in the mixer.  This ensures flexibility in which output index is the throttle.

julianoes, 2016-11-01:
This should be fixed with #5716 already.

evanmw, 2016-11-01:
@julianoes  Oh, nope, you're right.  Although it's the opposite fix :)  (made SIL mimic HIL as opposed to changing HIL to mimic SIL)

Is there a reason that throttle should be forced to be output 3 and scaled 0 to 1 as opposed to having that in the mixer?  It just seems more configurable/flexible if it's up to the mixer to keep things positive.  The only downside is you lose half your available resolution since you're only mapping 0 to 1 from pwm outputs 1500 to 2000 instead of mapping from 1000 to 2000.

julianoes, 2016-11-01:
With the simulation we want to mimic the real system and most real system have ESCs that go from ~1000 to ~2000. Therefore this makes more sense in my opinion.

However, my change from -1..1 still leaves the possibilty to add ESCs that can reverse, or variable pitch propellors, both resulting in a negative thrust.

closed 2016-12-28 by LorenzMeier