PX4/PX4-Autopilot #5794: AUX1 output pass through from RC channel not functioning

Aperture-Laboratories, 2016-11-01:
Issue: The AUX1 and AUX2 outputs are unable to output anything that is mapped to the RC channel pass through. They output a 50-60% duty cycle upon boot when examined with a scope.

Hardware: Pixhawk FMUv2 running 1.4.4 BETA
Hexactoper, fully configured and calibrated
Taranis X9D, with 16 channel receiver
Channel 7 (a dial/pot on the X9D) mapped to AUX1
Servo connected to AUX1 and 5v BEC on the servo rails

The system console reports the mixer for the aux outputs loads:
``` INFO  [init] Mixer: /etc/mixers/pass.aux.mix on /dev/pwm_output1 ```

The ```fmu test``` command in the console also changes all the servo outputs to fluctuate from 0% duty cycle to 100% as observed by the scope, so it's not a hardware issue. The servo also works on a servo tester.

diegostf, 2016-11-13:
Same problem happens here, with plane configuration and passthrough channel 5 to AUX1, for flaps.

tops4u, 2016-12-05:
Works in my Hexacopter Config (1.5.1RC3) but also earlier. But I'm using a Spektrum Receiver with PWM Outputs which are then connected to a PPM Encoder.

LorenzMeier, 2016-12-05:
@Aperture-Laboratories @diegostf Please re-test with 1.5.1RC3 or later. We fixed a few things there. Please also make sure you have AUX1 mapped on your remote control (otherwise nothing will happen).

Aperture-Laboratories, 2016-12-05:
Tested and confirmed working for me on my hexacopter.

LorenzMeier, 2016-12-05:
Nice, closing.

closed 2016-12-05 by LorenzMeier