PX4/PX4-Autopilot #5801: VTOL: FW flight very strange lock down in Loiter.

tubeme, 2016-11-03:
Hi guys,

We've had a very strange case in witch once we've transitioned, we put the plane in Loiter manually, and from then on the VTOL just stopped responding. It disconnected from QGC and started Loiter but with Cruise speed, and not min airspeed. We switched off the RC to make it failsafe to no avail. It just continued circling. We tried everithing possible but nothing happened. At some point the batteries just started to die and the plane just landed very nice in Loiter. I must say the TECS just did a brilliant work. 

We could not find the reason, because the LOG file just stopped recording very soon in to the flight. And the other logs are very ordinary.

http://logs.uaventure.com/view/dAEJ87t6rCy5qw7MqmAKkC

This is from the messages log:

2016_11_02_14_45_12: ARMED by RC
2016_11_02_14_45_13: [blackbox] /fs/microsd/log/2016-11-02
2016_11_02_14_45_13: [blackbox] recording: 14_45_11.px4log
2016_11_02_14_45_18: Takeoff detected

nothing more.

And here is the PreFlight log again normal log.

[preflight_perf14_45_11.txt](https://github.com/PX4/Firmware/files/569289/preflight_perf14_45_11.txt)

It was on PixFalcon with the digital airspeed.

The plane was holding the loiter brilliantly all the time, so the GPS must have been working. The TECS were brilliant. Just it did not respond to our RC nor the our Telemetry connection. The telemetry was having a green light which mean there was a connection between the telemetry modules, but there was no obvious traffic.

kd0aij, 2016-11-03:
This is a situation in which it would be nice to have MANUAL mode available.

sanderux, 2016-12-20:
@tubeme have you seen this behavior again?

tubeme, 2017-01-11:
@sanderux No never again. Sorry for the delay.

dagar, 2017-04-12:
Unfortunately, this was probably a hard fault. Newer PX4 has hard fault logging if it happens again.

closed 2017-04-12 by dagar