PX4/PX4-Autopilot #5802: Crash when flying below the Home Point

s3erjaeh, 2016-11-03:
When I fly with my Pixhawk in state_HIL simulation with the current master build below the home point altitude, the Pixhawk do strange things:

In **Stabilize Mode** the Pixhawk is unable to rotate around the YAW axis. After flying above the relative altitude of zero, the yaw rotation ability comes back.

In **Altitude- and Position-Hold Mode** the Pixhawk suddently crashes to the ground when I fly below a relative altitiude of zero (rotor rotation stops). After switching to stabilize mode and fly above the home point I can reproduce all this behaviour.

Can someone confirm/reproduce this?

dagar, 2016-11-03:
Thanks for the bug report, flying below home isn't a well tested scenario. Which simulator are you using?

s3erjaeh, 2016-11-03:
I using the AeroSim RC Simulator with a self programmed plugin. I wrote this for my company, so I'm not allowed to publish it for testing. But i hope this behaviour can be also reproduced with a other Simulator.

dagar, 2016-11-29:
@s3erjaeh I was never able to reproduce this using HIL + jmavsim. Could you try the same firmware with jmavsim and compare to your own plugin?

dagar, 2017-04-12:
This works for me. Please reopen if you'd like to discuss debugging your setup.

closed 2017-04-12 by dagar