PX4/PX4-Autopilot #5803: [-Werror=missing-braces]

buaayly, 2016-11-03:
When I compiled the firmware, I got the error as following, 

/src/modules/uavcan/uavcan_main.cpp:1263:3: error: missing braces around initializer for 'const char* [4]' [-Werror=missing-braces]
compilation terminated due to -Wfatal-errors.
cc1plus.exe: all warnings being treated as errors
make[3]: *** [src/modules/uavcan/CMakeFiles/modules__uavcan.dir/uavcan_main.cpp.obj] Error 1
make[2]: *** [src/modules/uavcan/CMakeFiles/modules__uavcan.dir/all] Error 2

How can I solve this problem?

bkueng, 2016-11-04:
Can you test  #5805?

s3erjaeh, 2016-11-04:
I had the same problem. You must update the gcc compiler, this is a bug in gcc. On Windows you simply download the gcc here: https://launchpadlibrarian.net/287101671/gcc-arm-none-eabi-5_4-2016q3-20160926-win32.exe 
After installation you copy the folders bin,lib and share to the px4 Toolchain folder (px4\toolchain) and override the existing gcc compilers. This will fix the issue.

closed 2016-11-04 by LorenzMeier

buaayly, 2016-11-04:
Thanks s3erjaeh! It works!

buaayly, 2016-11-04:
bkueng.
I updated my gcc, and the problem was solved. So I can not test the #5805 now, sorry.

PR #5805 (2016-11-04): uavcan: fix initialization of std::array
in C++11, double braces are needed for std::array aggregate initialization, or assignment with =.
see: http://en.cppreference.com/w/cpp/container/array

Fixes  #5803