PX4/PX4-Autopilot #5806: Noise on servos when connecting to QGC

CarlOlsson, 2016-11-04:
Master fw 25be7aa7cf30b633afd1abeb30f15547719f1471
When I connect Pixracer to QGC on my Iphone the flaps start to have random movements. 
Log is [here](http://logs.uaventure.com/view/VE3gawVUitdowphE86WWkj)
[image]
[image]

LorenzMeier, 2016-11-07:
Can you share your post flight perf file?

LorenzMeier, 2016-11-07:
And did you have manual input enabled on your phone (thumb sticks?)

CarlOlsson, 2016-11-11:
No I don't think I have any manual inputs on my phone, and I don't have the perf file, but it is very reproducible so will upload one.

CarlOlsson, 2016-11-24:
It is definitly the thumb sticks on the iphone, as soon as I disable them the jittering goes away. [Here](http://logs.px4.io/plot_app?log=395944a7-b8e3-41d8-af7e-00f091fbb4fc) is a logfile where I activate the thumb sticks and then deactivate them again.

LorenzMeier, 2016-11-25:
@DonLakeFlyer I think you need to fix the user model. We keep running into this and users just don't get that QGC is sending RC controls to the vehicle. They think that unless they use the joystick or thumb sticks nothing is sent even if enabled.

dagar, 2016-11-25:
@LorenzMeier what if we also changed the mavlink handling of MANUAL_CONTROL and COM_RC_IN_MODE a bit? Is there any reason we should allow actual RC and MANUAL_CONTROL at the same time?

https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_receiver.cpp#L1397

COM_RC_IN_MODE
 0 - RC only, don't process MANUAL_CONTROL
 1 - MANUAL_CONTROL -> manual_control_setpoint (completely ignore RC?)
 2 - MANUAL_CONTROL -> rc_input_values (completely ignore RC?)

DonLakeFlyer, 2016-11-26:
Open to suggestions on how to change the user model. If you turn on joystick there needs to be continuous sending of outputs just like an RC control. Not sure how to get around that.

LorenzMeier, 2016-12-26:
@dagar Can you send a PR?

dagar, 2016-12-27:
I can do the quick fix described above. Using priorities as discussed in #5942 can still happen later.

What are you thoughts on Mavlink MANUAL_CONTROL -> manual_control_setpoint and rc_input_values going forward? Keep both? The entire thing is really quite confusing, even for developers.

dagar, 2017-02-16:
Should be resolved. PX4 will now only take the first source (RC or Mavlink).

closed 2017-02-16 by dagar